betaflight/betaflight #10868: FOXEER F722 V2 and DSHOT 600 not working BF 4.1+

sebass-85, 2021-07-28:
### Describe the bug
After upgrade to BF 4.2.9 there is no chance to set up motor protocol dshot, only oneshot.  If i choose e.g. dshot600, after restart, can't connect to BF, USB is not recognize. Only can flash firmware from beginning. 
Compared custom defaults(for FOXEER F722 V2) with defaults and found setting which crash. 
in custom defaults **adc_device = 3**
in default **adc_device = 1**
if i change it to 1, then is possible to set dshot but there is no information about voltage in BF(and OSD).

### To Reproduce
Flash any BF since 4.1.0 to FOXEER F722 V2 and apply custom defaults, then change motor protocol to any dshot.

### Expected behavior
Configurator can connect to FC with any dshot and can see voltage. 

### Flight controller configuration
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.9 Apr 27 2021 / 19:34:29 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: FOXE, board_name: FOXEERF722V2, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

board_name FOXEERF722V2
manufacturer_id FOXE

# master
set acc_calibration = 42,101,-82,1
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# 
# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: BEEPER
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: GYRO_CS 1
B03: CAMERA_CONTROL
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: FREE
C02: ADC_CURR
C03: OSD_CS
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: FREE
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: ADC
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
BF 4.1.0 - 4.2.9
 - Flight controller: 
FOXEER F722 V2
 - Other components:
ESC Racerstar Air50
VTX RushTank Ultimate 2
Camera Runcam Phoenix 2
CRSF Nano
GPS BN 220 
 - How are the different components wired up:

butterSnake, 2021-07-30:
This is happing with 4.3 builds too

DusKing1, 2021-07-30:
no motor resource was found in your `resource show all`

fpvdoc, 2021-07-30:
I experience issues that might be related to this. Freshly flashed device can connect but if I only set the motor protocol, hit save and reboot, I can‘t connect anymore. USB isn‘t recognized anymore. If I set motor protocol + bidirectional dshot everything works. Same goes for a diff, if only motor protocol is set in the diff I run into the issue. If bidir is being enabled as well, it‘s working.

sebass-85, 2021-07-30:
Guys, solution(i hope) it's on #10870.
**adc_device = 3**
**dma ADC 3 0**
works for me.

Squarefpv, 2021-07-30:
Have you tried using zadg at all

fpvdoc, 2021-07-30:
> Have you tried using zadg at all

Doesnt has anything to do with Zadig if the com port disappear for no reason after setting the motor protocol and as soon as you reflashed its recognized again.

haslinghuis, 2021-07-30:
@Johnsnow-foxeer Can you please have a look at it?
`FOXE-FOXEERF722V2.config` line 70 should be `dma ADC 3 0` according to this issue.

haslinghuis, 2021-07-30:
Transferred issue to https://github.com/betaflight/unified-targets/issues/480

Johnsnow-foxeer, 2021-08-02:
Hi Sir

Has the problem fixed?

Best regards

Harvey

------------------------------------------------------------------
Sender:haslinghuis ***@***.***>
Sent At:2021 Jul. 31 (Sat.) 07:14
Recipient:betaflight/betaflight ***@***.***>
Cc:Foxeer ***@***.***>; Mention ***@***.***>
Subject:Re: [betaflight/betaflight] FOXEER F722 V2 and DSHOT 600 not working BF 4.1+ (#10868)

Closed #10868.

fpvdoc, 2021-08-02:
Hi Harvey,

no it hasn‘t. I use the latest nightly build #2480 and if I disable bidirectional dshot or paste a diff with just the motor protocol being set but no bidirectional, the com port is gone after saving.

best regards
Marco

Johnsnow-foxeer, 2021-08-03:
Please change to 
dma ADC 3 0

haslinghuis, 2021-08-03:
It's changed. Just approve the linked PR https://betaflight/unified-targets#482.

Johnsnow-foxeer, 2021-08-03:
OK, thank you, testing now

haslinghuis, 2021-08-03:
Closing now as this is transferred to https://betaflight/unified-targets#482

closed 2021-08-03 by haslinghuis