betaflight/betaflight #10870: FOXE-FOXEERF722V2.config config file error

Johnsnow-foxeer, 2021-07-29:
### Describe the bug
Can not find the Uart, cannot setup parameters

### To Reproduce
Flash any BF since 4.1.0 to FOXEER F722 V2 and apply custom defaults, then change motor protocol to any dshot.

### Expected behavior
Configurator can connect to FC with any dshot and can see voltage.

### Flight controller configuration

# dma

Correct
dma ADC 3 0

ERROR 
dma ADC 3 1

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

DusKing1, 2021-07-29:
Wrong repo, unified-target

Karl33to, 2021-07-29:
I'm having exactly the same issue as described with this board

Johnsnow-foxeer, 2021-07-30:
Hi 

Can you upgrade this? we have already submitted how to fix this

Best regards

Harvey 

------------------------------------------------------------------
Sender:Karl Payne ***@***.***>
Sent At:2021 Jul. 30 (Fri.) 01:09
Recipient:betaflight/betaflight ***@***.***>
Cc:Foxeer ***@***.***>; Author ***@***.***>
Subject:Re: [betaflight/betaflight] FOXE-FOXEERF722V2.config config file error (#10870)

I'm having exactly the same issue as described with this board