betaflight/betaflight #10881: Automatic ARM + max RPM

NiverMK, 2021-08-02:
### Describe the bug

My quad was on my kneels. RC was turned off. I have connected the battery - there were default 3 + 2 beeps. After 1-2 seconds there was another beep and right after that engine #1 started to spin at max rpm. Now I have 10 surgical stitches on my left hand.
Also there were problems with ESC reinitialization, when inited ESC beeped without any command from my side, or when engines are turned off in the air. It seems that all these problems are the same and started to happen after I have updated to v4.2.8.

### To Reproduce

Happened once.
1) Keep RC turned off;
2) Connect battery.
3) Wait for ESC initialization;
4) Engines can start to rotate.

### Expected behavior

FC shouldn't arm itself and turn on engines without my command.

### Flight controller configuration

```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.8 Feb 15 2021 / 12:10:35 (101738d8e) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722HD, version: a684eb8f, date: 2020-05-26T06:40:03Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722HD
manufacturer_id MTKS
mcu_id 004700463338510839393832
signature 

# feature
feature -AIRMODE

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 65536 115200 57600 0 115200

# aux
aux 0 0 0 1400 2100 0 0
aux 1 1 0 1800 2100 0 0
aux 2 13 1 1650 2100 0 0

# rxfail
rxfail 3 h

# master
set acc_calibration = 18,-10,364,1
set mag_hardware = NONE
set min_check = 1000
set max_check = 2000
set serialrx_provider = CRSF
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 20
set failsafe_throttle = 1200
set failsafe_throttle_low_delay = 300
set failsafe_procedure = AUTO-LAND
set align_board_pitch = 180
set ibata_scale = 180
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set osd_rssi_alarm = 40
set osd_cap_alarm = 200
set osd_alt_alarm = 300
set osd_vbat_pos = 2520
set osd_rssi_pos = 2107
set osd_link_quality_pos = 2104
set osd_rssi_dbm_pos = 121
set osd_tim_1_pos = 2081
set osd_current_pos = 2464
set osd_mah_drawn_pos = 2496
set osd_altitude_pos = 2476
set osd_warnings_pos = 12745
set osd_disarmed_pos = 2283
set gyro_1_sensor_align = DEFAULT

profile 0

# profile 0
set p_pitch = 52
set i_pitch = 50
set d_pitch = 30
set p_roll = 44
set i_roll = 40
set d_roll = 28
set p_yaw = 70
set i_yaw = 45

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 30
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 80

rateprofile 1

rateprofile 2

# rateprofile 2
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 30
set roll_srate = 65
set pitch_srate = 65

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: PINIO 1
B00: FREE
B01: FREE
B02: GYRO_CS 1
B03: PINIO 2
B04: MOTOR 3
B05: MOTOR 4
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: CAMERA_CONTROL
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: MOTOR 1
C09: MOTOR 2
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12:
    CH2: CAMERA_CONTROL
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: Matek F722-PX
 
 - Other components:
	ESC: DALRC 4IN1 ESC 40A (BLHeli_32 v32.6)
	Reciever: TBS TRACER NANO RX (v5.07)
 
 - How are the different components wired up:
	FC and ESC are connected through JST 8 pin
	Receiver is soldered to FC

I found a similar problem:
https://github.com/betaflight/betaflight/issues/9403
I know, that it's almost impossible to reproduce this bug, but it's very critical.
I think, that it would be nice to create a feature, that will allow to do post init of ESC, not with battery connection like now. I mean, FC shouldn't send any kind of PWM/digital signal to ESC, until pilot switch the tumbler.
Should I request something like that?

McGiverGim, 2021-08-03:
This is a very bad bug :( but without being able to reproduce it, is very difficult for the developers to find the cause and fix it.

As a suggestions, I always plug the battery with the quad in a surface and my hand on the top of the battery (quad with top battery). This must not happen, but if the quad starts to spin like in your case is a safer way because I have the quad controlled with the hand and far from the props.

About the feature that you are proposing, I don't know if it will serve, but you can open the feature request to see what others with more knowledge think about it.

NiverMK, 2021-08-04:
> This is a very bad bug :( but without being able to reproduce it, is very difficult for the developers to find the cause and fix it.
>
> As a suggestions, I always plug the battery with the quad in a [...]

Your suggestions are correct, but now I think I will not touch my quad for months.
I have crteated feature request (https://github.com/betaflight/betaflight/issues/10887) and hope devs will pay attention to it.

frizzle101101, 2021-08-11:
Did u have usb plugged in at all during this?

NiverMK, 2021-08-12:
> Did u have usb plugged in at all during this?

nope, only quad and battery

closed 2021-09-20 by github-actions[bot]