betaflight/betaflight #10887: Post initialization of ESC

NiverMK, 2021-08-04:
### Is your feature request related to a problem? Please describe.

According to these issues (https://github.com/betaflight/betaflight/issues/10881, https://github.com/betaflight/betaflight/issues/9403) it seems that there is very rare but safety critical bug when FC arms itself and turns on engines.

### Describe the solution you'd like

Prearm mode is just a protection from accidental tumbler switch, but there is no real protection from possible bugs in Betaflight.
I propose to create a mode that would allow to do post initinitialization of ESC, not with battery connection like now. I mean, FC shouldn't send any kind of PWM/digital signal to ESC, until pilot switches the tumbler. This mode should be the first and main flag to start ESC initialization. There should be nothing at all, that can avoid this flag and send something to ESC.
Also with this mode it would be easier to find possible bugs in receiver and ESC firmwares.

### Describe alternatives you've considered

Automatic ESC initialization with configurable timer also would be nice feature in addition to mode above.

frizzle101101, 2021-08-11:
These are really bad issues both of which resulted in stitches, this shld not be taken lightly, I am definitely for doing things in a safer way with more preventative protection in the logic of the code hopefully more seasoned devs dont just shrug this off cuz clearly its happend more than once, sorry for ur sacrifice

NiverMK, 2021-08-12:
> These are really bad issues both of which resulted in stitches, this shld not be taken lightly, I am definitely for doing things in a safer way with more preventative protection in the logic of the [...]

Thanks for your reply. Hope my sacrifice won't be in vain :)

limonspb, 2021-09-01:
I've seen motors spinning by themselves, but that was a bad/damaged ESC that decided to spin them regardless of the FC commands.

NiverMK, 2021-09-13:
> I've seen motors spinning by themselves, but that was a bad/damaged ESC that decided to spin them regardless of the FC commands.

I don't think, that the problem is with a bad/damaged ESC. I have tried many times to reproduce this bug or to get undefined behavior, but it was unsuccessfull. After that day quad is flying very well without any kind of problems.
But I do not exclude that this could be a bug in BLHeli firmware. So that's why Post initialization mode is necessary. It can help to understand, where the problem is and maybe save somebody's arms, because in my case engine started to rotate only after ESC was initilized by FC.

frizzle101101, 2021-09-13:
Was thinking about it...my question tht position me FOR doing this are; why
do we initilize the escs first thing and then do gyro why would there be
any disadvantages where the gyro initilize first then escs initilize or if
the user wants it to initilize after arm they should have the option
too...this is a very good safety consideration where the default would be
the after arm and to the users caution can change it in the cli to one of
the 3 options... Like: esc_init=PRE_GYRO,POST_GYRO,POST_ARM

closed 2021-10-21 by github-actions[bot]