betaflight/betaflight #10891: feedforward goes crazy !?

TheIsotopes, 2021-08-05:
### Describe the bug
since #10805 is merged one of my quads have problems with feedforward that's going crazy.
with other quads i can not determine the problem. 
going back to the build before this was merged i do not have this problem and all is good.

@ctzsnooze is there a bug, or can you tell me what's wrong. 

build with merged #10805:
[image]

build before this was merged:
[image]

### To Reproduce
usually occurs with flips, cannot describe a procedure to trigger this, maybee seems to be triggered by increased noise  

### Flight controller configuration

```
# Betaflight / STM32F405 (S405) 4.3.0 Aug  4 2021 / 05:27:00 (034904cd1) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF405MINI, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF405MINI
manufacturer_id MTKS

# name: SKINNY

# feature
feature -SOFTSERIAL
feature -OSD
feature LED_STRIP

# serial
serial 1 0 115200 57600 0 115200
serial 3 64 115200 57600 0 115200

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# led
led 0 6,5::AW:0
led 1 7,5::AW:0
led 2 8,5::AW:0
led 3 9,5::AW:0
led 4 9,2::AW:0
led 5 8,2::AW:0
led 6 7,2::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 1 900 1100 0 0
aux 2 26 2 900 1100 0 0
aux 3 36 1 1900 2100 0 0

# master
set gyro_lowpass2_hz = 500
set dyn_notch_count = 2
set dyn_notch_q = 450
set dyn_notch_min_hz = 200
set dyn_notch_max_hz = 500
set acc_hardware = NONE
set rc_smoothing_auto_factor = 45
set serialrx_provider = GHST
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set bat_capacity = 1550
set use_cbat_alerts = ON
set cbat_alert_percent = 20
set vbat_duration_for_warning = 50
set vbat_duration_for_critical = 20
set pid_process_denom = 2
set gyro_rpm_notch_q = 350
set gyro_rpm_notch_min = 80
set name = SKINNY

# profile 0
set iterm_relax_cutoff = 20
set throttle_boost = 7
set throttle_boost_cutoff = 25
set p_pitch = 41
set i_pitch = 81
set d_pitch = 34
set f_pitch = 85
set p_roll = 37
set i_roll = 76
set d_roll = 31
set f_roll = 81
set p_yaw = 40
set i_yaw = 81
set f_yaw = 81
set d_min_roll = 20
set d_min_pitch = 22
set feedforward_smoothing = 30
set feedforward_jitter_reduction = 10
set feedforward_max_rate_limit = 90
set feedforward_boost = 14
set simplified_pids_mode = RPY
set simplified_master_multiplier = 90
set simplified_dterm_filter = ON

# rateprofile 0
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 15
set roll_srate = 108
set pitch_srate = 108
set yaw_srate = 99

# end the command batch
batch end
```

### Setup / Versions
 
 - Flight controller: ...
 
 - Other components: ...
 
 - How are the different components wired up: ...

ctzsnooze, 2021-08-06:
Thanks for reporting this, I haven't seen it happen so far like that.

I could much better assist if you could possibly share the logfile and, even better, make the log with debug mode set to `FEEDFORWARD` or `FF_INTERPOLATED` (whichever name appears in the list).

It looks like the ADC in the hardware settings of your OpenTx firmware is set on, and this is causing duplicates.  The previous log showed spiking in feedforward, as well, just not as bad.  The RC data steps are variable in width (timing).  Sometimes they appear to be at normal speed, other times at half normal speed, and other times widely spaced.  This is typical of an ADC being active.

It seems that the PR exaggerates problems greatly in the region where there is alternating duplicate packets.  It will be the result of the interpolation algorithm that tries to predict the next value, which works really well for occasional duplicates.  It looks like it is leading to feedback when there are long trains of duplicates.  I'll have to revert something to stop it.

In the meantime, un-check your ADC checkbox and feedforward will work properly again.

TheIsotopes, 2021-08-06:
You are right ADC is on, gonna disable this now. 

I also noticed that this only occurs on my quads with Ghost RX (Mode: Race250), on the quads with FrSky / FPort everything is fine. I noticed this because yesterday i forgot to mention that this also affects a second quad, this also has Ghost RX in Mode: Race250. This seems to be the crucial point. 

Unfortunately I deleted the blackbox because no further debugs were activated. :-(
I will test this again today with the ADC deactivated and the current build with setting debug mode.

ctzsnooze, 2021-08-06:
Thanks.  Even though it will look a lot better with the ADC off, I must fix the behaviour of the code in this setting.

TheIsotopes, 2021-08-06:
ok, with deactivated ADC everything works as expected again. Thanks for the tip. 👍 
Nevertheless, I have attached the last bb log with four flights. It is from the last master with the feedforward debug enabled. 

[BTFL_BLACKBOX_LOG_SKINNY_20210806.txt](https://github.com/betaflight/betaflight/files/6945899/BTFL_BLACKBOX_LOG_SKINNY_20210806.txt)

daleckystepan, 2021-08-06:
@TheIsotopes Good catch. I would like to reuse this issue a little bit. As you have IRC Ghost I would be happy if you are willing to test some of improvements to IRC Ghost I am working on. Eg 11/12 bit resolution support and some other changes. Contact me on Slack or here. Thank you.

ctzsnooze, 2021-08-07:
I've raised the above PR, which prevents the feedback part and greatly improves the overall situation when ADC is on.

Even with the bug fix, ADC remains a real nuisance, causing stepping in setpoint and spikes in feedforward that cannot be 'coded out'.  

It would be fantastic if ADC off the default in OpenTx.  Lots of people don't realise how essential it is to turn it off.

TheIsotopes, 2021-08-09:
ok, everything has been said and the bugfix has been merged. Test flights were successful.

closed 2021-08-09 by TheIsotopes