betaflight/betaflight #10899: FW 4.2.x on F4 PCB does not FUNCTION with Genuine Spektrum Rx

dribbe, 2021-08-12:
### Describe the bug

Basically, a genuine Spektrum Remote Rx (Sat) does not work with Betaflight 4.x.x on and F4 board (a lemon does!)
(Works on same hardware with FW 3.5.7!)

### To Reproduce

-F4 PCB such as OMNIBUSF4 xx or REVO or similar F4 PCB able to use UART1 as serial rx/satellite
-Genuine Spektrum Sat (SPM9645 is the classic DSMX one with MANY out there)
   https://www.horizonhobby.com/product/dsmx-remote-receiver/SPM9645.html
   (now replaced by 9745)
-Lemon Sat good to have for Cross check
-Spektrum Tx (I am primarily using IX12 and DX18 currently)
-hex file (4.2.9 or 4.2.6 for example for OMNIBUSF4 like as set in configurator FW page) or maybe uses this: 
    Binary: betaflight_4.2.9_STM32F405.hex
    Unified Target: AIRB-OMNIBUSF4.config
  (any FW 4.2.x that supports the PCB above)
-betaflight_3.5.7_OMNIBUSF4.hex (FW for Cross-Check)
-Configurator... 10.7 (or 10.6 or 10.8)
Flash the F4 board
Connect Sat Rx
Set serial rx on UART1, and Spektrum2048 (1024 and SRXL also tested/related) (alternates also below)
Bind DSMX 11ms (DSMX 22ms and DSM2 11 / 22ms behave same)   Bound Via - spektrum_sat_bind=9
Set Telemetry OFF (ON Tests same, see alternate cases below)
Check Tx/Rx function on BF Receiver screen

### Expected behavior

CASE1
Lemon Satellite via Sat Connector/3.3v/TX1/A09 on chip
Bound Via - spektrum_sat_bind=9
Works fine

CASE2
Spektrum 9645 Sat via Sat Connector/3.3v/TX1/A09 on chip
**No Function** (Link light is Solid on Sat, Bound Via - spektrum_sat_bind=9 (tried also 3, 5 and 7)
(Works fine with FW 3.5.7 flashed on same setup. Just re-flash and bingo)

CASE3 (also) SRXL No Telem
Spektrum 4649T Serial Data Rx via 'servo' Connector/5v/**TX1/A09** on chip
Set to SPEKTRUM2048/SRXL
Set to Telemetry OFF (also tried on)
**No Function** (Link light is Solid on Rx)
(Works with FW 3.5.7 flashed on same setup) (Also, I am sure I have used this in the past on 4.2.6 on F3 boards OK)

CASE4 (also) SRXL use Telem
Spektrum 4649T Serial Data Rx via 'servo' Connector/5v/**RX1/A10** on chip
Set to SPEKTRUM2048/SRXL
Set to Telemetry ON (also tried on)
**WORKS**

CASES 1&2 are the main case SRXL CASES 3&4 added as further issue and as a clue!

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller: Omnibus F4 and various other similar F4 boards act same
 
 - Other components: Genuine Spektrum Satelite 9645. Others helpful as above.
 
 - How are the different components wired up: Plug sattelite into 3 pin 3.3V port (TX1 signal)  Power with USB (or 5V BEC same)

Odd that the SRXL works on the UART1 Telem Pin, but not on the non-telem.

Basically the SPM9645 Sat and 9645 serial data rx in the basic TX1 non-telemetry config  always worked... for  example in FW 3.5.x, but is BROKEN on all the 4.2.x FW I have tried... when running on and F4 chip/PCB.

Pretty confident all of this worked on 4.2.6 FW when running F3 hardware.

dribbe, 2021-08-12:
FWIW.... 
Also tried binding sat while connected to a conventional spektrum Rx.
No difference

Usually the Arming Disable Flags will say RX_FAILSAFE

For Sat DSMX 2048/11ms
The Receiver screen shows the 12 channels of bars in the default/failsafe positions... no movement.

Have tried many many combinations of 1024, 2048, 11ms, 22ms, telemetry on/off.... all acts same as reported above.

Tried the new 9745 Spektrum Remote Rx.... Same
Swap Sat out for the Lemon Sat... works fine... EVERY TIME.

David

dribbe, 2021-08-12:
Here are the Resources from an OmnibusF4 flashed with the generic 4.2.9 from the configurator
calibrate
set to Serial RX UART 1
Use Serial / SPEKTRUM2048
(No Telem)
set ONESHOT125
set map TAER1234

Works Fine with Lemon Sat
Plug in Spektrum Sat, Nothing

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FLASH_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: PREINIT 3
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#

dribbe, 2021-08-12:
Full dump from the generic OMNIBUSF4  FW4.2.9

dump

# version
# Betaflight / STM32F405 (S405) 4.2.9 Apr 27 2021 / 19:33:01 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name OMNIBUSF4
manufacturer_id AIRB

# name: -

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
[... 1044 lines cut ...]
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

#

closed 2021-09-20 by github-actions[bot]

durzanoob, 2026-09-21:
## Fix: genuine Spektrum satellite on UART1 (OMNIBUSF4SD / SuperS F4) — Betaflight 4.x

**Hardware:** Happymodel Mantis85, OMNIBUSF4SD target (SuperS F4 board, STM32F405),
genuine Spektrum DSMX satellite on UART1 (RX = PA10).

**Symptom:** Works perfectly on 3.5.7. On every 4.x (4.0–4.5) the satellite binds
(solid orange LED) but Betaflight stays in failsafe / no channel movement (RXLOSS),
with an identical serial-RX config. Clone (Lemon) sats work; an external pull-up on
the signal line also "fixes" it.

**Root cause:** Older Betaflight initialised unused GPIOs to input pull-up (a defined
level). 4.x removed that default, leaving the board's PPM/ESCSERIAL pad (PB8) floating.
On this board a floating PB8 breaks reception of the (low-drive) genuine Spektrum sat
on UART1. I can't confirm the exact electrical path without the board schematic, but
empirically driving PB8 to a defined level restores reception 100% reliably — which is
consistent with the removed pull-up init, and with why an external pull-up or a clone
sat also work while 3.5.7 works.

**Fix — paste in CLI, then reboot:**

resource ESCSERIAL 1 none
resource PINIO 1 B08
set pinio_config = 1,1,1,1
save

This frees PB8 from ESCSERIAL and drives it as a push-pull PINIO output, so it no
longer floats.

**Notes:**
- Side effect: PB8 can no longer be used for PPM input or ESCSERIAL — irrelevant if you
  use serial RX. Fully reversible.
- A full-chip-erase reflash wipes this; reapply afterwards.
- Suggestion for maintainers: set a default state for PB8 in the OMNIBUSF4SD config
  (e.g. via the DEFAULT_GPIO_ARRAY mechanism from #14145), if safe across all boards
  sharing this config.

This has been open ~5 years across #10898 / #10899 / #9648 / #8987 / #8021 with no
posted solution — hope it helps others.