betaflight/betaflight #10906: Add failsafe stage 3 (sanity checks failed, GPS rescue off)

SyncroIT, 2021-08-17:
### Is your feature request related to a problem? Please describe.
When GPS rescue mode is enabled as a failsafe stage 2, in case the sanity checks fail the drone enters "Fall Mode". This happens very frequently, imagine you're flying the drone very close to yourself (< 50 meters) and it suddenly loses the RX signal because of interference,  the drone just falls when it could land! 

### Describe the solution you'd like
I would like to add a failsafe in case of sanity checks failure, so I can determine the behaviour of the drone in case of failsafe and failure of the sanity checks. This would be very useful because I could configure the drone to Land in case of Failsafe with Failed Sanity Checks instead of dropping to the ground and potentially damaging itself.

### Describe alternatives you've considered
The only alternative is to ignore the sanity checks and let the drone gain altitude in GPS rescue, but that is absolutely not safe and nobody should do that. In case you fail to recover the signal (example: battery dead in the radio, or HW issue in the RX) your drone would be gone and it could also harm other people or damage objects in the surroundings.

[image]

[image]

etracer65, 2021-08-17:
Betaflight doesn't actually have the ability to "land". The poorly named "land" mode is really a completely uncontrolled, run the motors at a fixed speed for a fixed amount of time, chop up anything in its path mode. It should probably be renamed to "blender" mode. We don't really consider a quad with motors spinning descending on your head to be "safer" then just disarming.

SyncroIT, 2021-08-17:
> Betaflight doesn't actually have the ability to "land". The poorly named "land" mode is really a completely uncontrolled, run the motors at a fixed speed for a fixed amount of time, chop up anything [...]

That's not the point. Whether Land mode is good or bad is not the point here.
The point is giving the ability to choose what the drone should do in case GPS Rescue fails, instead of disarming it. 
Right now, if GPS rescue fails because of Sanity Checks, the drone disarms, and that's an hardcoded behaviour, you cannot chose to do otherwise.

It would be much better if you could chose one of the modes (Land/Drop) available at Stage 2, unfortunately, as it is right now, you can't. If you select "GPS Rescue" you can't choose another mode to apply in case of GPS Rescue failure.

If you don't like "Land" mode, you will still be able to select Drop.

In the future there could be improvements to Land mode, or maybe even in the present somebody (example: me) would like to use that mode instead of disarming the drone at 20 meters of height and watching the drone fall down because the sanity checks haven't passed.

etracer65, 2021-08-17:
> That's not the point. Whether Land mode is good or bad is not the point here.
> The point is giving the ability to choose what the drone should do in case GPS Rescue fails, instead of disarming it. [...]

That is the point. We decided that disarming was the safer behavior in these cases.

> It would be much better if you could chose one of the modes (Land/Drop) available at Stage 2, unfortunately, as it is right now, you can't. If you select "GPS Rescue" you can't choose another mode [...]

No need, that's the current behavior.

> In the future there could be improvements to Land mode, or maybe even in the present somebody (example: me) would like to use that mode instead of disarming the drone at 20 meters of height and watching the drone fall down because the sanity checks haven't passed.

It's far more likely that the "land" option will be removed entirely as it's legacy and not considered safe to have an uncontrolled fully-armed quad descending into whatever happens to be in its way.

GPS Rescue was designed to work with the minimum number of sensors available to accomplish its goal - get back in range. It only requires a GPS and thus it is limited by what can be done only with a GPS. It has no ability to hover, and no ability to land. Adding these would require that a compass sensor be installed and properly configured/calibrated. Very few Betaflight quads have a fully functional compass installed because even if they did it's extremely difficult to mount one on a small quad and keep it far enough from the motors and other strong magnetic sources. Adding autonomous flight capabilities is outside the scope of Betaflight's goals so there's no interest in having a true auto-land capability. GPS Rescue was initially not appropriate either and was only accepted after the extremely limited "get back in range" scope was agreed upon along with rigorous safety enforcement and sanity checks.

It's intentionally coded the way it is. Much discussion went into what should happen if GPS Rescue cannot meet the minimum safety checks and it was decided that disarming was the safest option. If you are intentionally enabling GPS Rescue in an unsafe manner, then stop doing that. It's not meant to be a toy and is only designed to present a last-ditch effort to get the quad back in range due to a signal loss. So if it fails and disarms it's no worse then if GPS Rescue never existed. If you are getting real failsafes within the minimum activation distance then you have a fundamental problem with your radio and/or receivers and that's the problem you should fix instead of looking for changes to GPS Rescue.

If you want more autonomous features then INAV is probably a better option for you.

SyncroIT, 2021-08-17:
> > That's not the point. Whether Land mode is good or bad is not the point here.
> > The point is giving the ability to choose what the drone should do in case GPS Rescue fails, instead of disarming it.
> > Right now, if GPS rescue fails because of Sanity Checks, the drone disarms, and that's an hardcoded behaviour, you cannot chose to do otherwise.
> 
> That is the point. We decided that "land" mode was unsafe and disarming was the safer behavior in these cases. So no point to allow a choice.
> 
> > It would be much better if you could chose one of the modes (Land/Drop) available at Stage 2, unfortunately, as it is right now, you can't. If you select "GPS Rescue" you can't choose another mode to apply in case of GPS Rescue failure.
> > If you don't like "Land" mode, you will still be able to select Drop.
> 
> No need, that's the current and intended behavior.
> 
> > In the future there could be improvements to Land mode, or maybe even in the present somebody (example: me) would like to use that mode instead of disarming the drone at 20 meters of height and watching the drone fall down because the sanity checks haven't passed.
> 
> It's far more likely that the "land" option will be removed entirely as it's legacy and not considered safe to have an uncontrolled fully-armed quad descending into whatever happens to be in its way.
> 
> GPS Rescue was designed to work with the minimum number of sensors available to accomplish its goal - get back in range. It only requires a GPS and thus it is limited by what can be done only with a GPS. It has no ability to hover, and no ability to land. Adding these would require that a compass sensor be installed and properly configured/calibrated. Very few Betaflight quads have a fully functional compass installed because even if they did it's extremely difficult to mount one on a small quad and keep it far enough from the motors and other strong magnetic sources. Adding autonomous flight capabilities is outside the scope of Betaflight's goals so there's no interest in having a true auto-land capability. GPS Rescue was initially not appropriate either and was only accepted after the extremely limited "get back in range" scope was agreed upon along with rigorous safety enforcement and sanity checks.
> 
> It's intentionally coded the way it is. Much discussion went into what should happen if GPS Rescue cannot meet the minimum safety checks and it was decided that disarming was the safest option. If you are intentionally enabling GPS Rescue in an unsafe manner, then stop doing that. It's not meant to be a toy and is only designed to present a last-ditch effort to get the quad back in range due to a signal loss. So if it fails and disarms it's no worse then if GPS Rescue never existed. If you are getting real failsafes within the minimum activation distance then you have a fundamental problem with your radio and/or receivers and that's the problem you should fix instead of looking for changes to GPS Rescue.
> 
> If you want more autonomous features then INAV is probably a better option for you.

I understand that Landing mode is not safe and you decided to hardcode the disarming of the drone in case one or more sanity checks fail, but for me this still looks very weird. In my mind, this "scheme" makes much more sense:

- Stage 1: waiting for X milliseconds for signal recovery 
- Stage 2: GPS rescue if enabled and sanity checks are passing, otherwise skip this stage.
- Stage 3: user choice (Land or Drop) 

While, right now, the user choice is an alternative to the GPS Rescue. 
My feature request was not that much about safety (because, anyway, the safety is determined by the user, I could misconfigure everything in Betaflight and make the drone very unsafe, as well as I could configure everything correctly and enjoy a nice flight) but it was about making things more logical.

If you think about it, "Land" and "Drop" are both "features" designed to stop the drone, while GPS rescue is designed to fly the drone closer to you. By thinking like this, it would make much more sense for a user to pick "Land"/"Drop" as a "last thing to do" instead of GPS Rescue. GPS Rescue cannot be on final stage because the sanity checks could fail. 

It's like, there is a condition (sanity checks) that can go wrong and you're not letting the user decide what to do in case that happens. It's a gap in a feature that is designed exactly for that, to determine what the drone should do in case everything goes wrong (failsafe).

mituritsyn, 2021-08-19:
I'm vote for deleting land option) Lets be consistent BF not designet for aby autonomous things idk why GPS rescue was implemented so any dumb now thinking he can use BF as Inav...

haslinghuis, 2021-08-19:
@SyncroIT thanks for the proposal. You simply can regain control of the quad when it's in range. See `failsafe_stick_threshold`.

closed 2021-08-19 by haslinghuis