betaflight/betaflight #10910: feature GPS unavailable

antoniosap, 2021-08-19:
### Describe the bug
```
CLI
# feature list
Available: RX_PPM INFLIGHT_ACC_CAL RX_SERIAL MOTOR_STOP SERVO_TILT SOFTSERIAL GPS RANGEFINDER TELEMETRY 3D RX_PARALLEL_PWM RX_MSP RSSI_ADC LED_STRIP DISPLAY OSD CHANNEL_FORWARDING TRANSPONDER AIRMODE RX_SPI SOFTSPI ESC_SENSOR ANTI_GRAVITY DYNAMIC_FILTER

# feature GPS
unavailable
```

### To Reproduce
select GPS port
[image]
select GPS config
[image]
on save it reset
[image]

### Expected behavior
GPS connected

### Flight controller configuration

### Setup / Versions
Betaflight / SPRACINGF3 (SRF3) 4.0.6 Sep  1 2019 / 01:05:57 (2a64051a2) MSP API: 1.41 / FEATURE CUT LEVEL 6

etracer65, 2021-08-19:
This is correct. GPS is not support on F3 flight controllers.

closed 2021-08-19 by etracer65

jackson-dengyang, 2024-08-08:
I met the same problem. I connected the Benewake TFmini lidar to FC (F722), selected the port, and then typed 'feature RANGEFINDER' in CLI. CLI showed it as unavailable.Does this mean that F722 does not support RangeFinder?