betaflight/betaflight #10911: Enable Barometer DPS310 default I2C address to 0x77

kafie1980, 2021-08-20:
### Is your feature request related to a problem? Please describe.

The Infineon DPS310 barometric pressure sensor has a default I2C address set to 0x77 as per the official datasheet: https://www.infineon.com/dgdl/Infineon-DPS310-DataSheet-v01_02-EN.pdf?fileId=5546d462576f34750157750826c42242

However in Betaflight the default I2C address for DPS310 is set to 0x76 : https://github.com/betaflight/betaflight/blob/master/src/main/drivers/barometer/barometer_dps310.c

Due to this Betaflight does not autodetect the DPS310 barometer.

### Describe the solution you'd like

Perhaps its best to update the DPS310_I2C_ADDR to 0x77

Or if possible enable both 0x76 and 0x77 in the code.

### Describe alternatives you've considered

On most devices such as the Adafruit DPS310 we have additional solder pads at the back of the board to change the default I2C address to 0x76 which is based on the SDO pin pulled-down to GND: https://learn.adafruit.com/adafruit-dps310-precision-barometric-pressure-sensor

Another alternative is to manually set the I2C device address via the CLI to 0x77 (hex) or in decimals to 119:
set baro_i2c_address = 119

swilkens, 2021-08-22:
This previous bug report shows a link to the mateksys F722-SE board (which uses this baro) with an adress set to 118 (0x76).

kafie1980, 2021-08-22:
My main reason for posting this was to ensure users who decide to buy a third party add-on DPS310 barometer module (such as from Adafruit or Grove) can stumble upon this request and know what they can do at the user end to get the barometer detected.

I will let the the Betaflight team decide whats best to avoid user confusion in the future.

haslinghuis, 2021-08-22:
> Or if possible enable both 0x76 and 0x77 in the code.

This would work.

SteveCEvans, 2021-08-22:
Fixed. :)

kafie1980, 2021-08-22:
Awesome. I will close this feature request.

haslinghuis, 2021-08-22:
@kafie1980 please check setting the address in cli like mentioned here: https://github.com/betaflight/betaflight/pull/10914#issuecomment-903347710

closed 2021-08-22 by haslinghuis

kafie1980, 2021-08-23:
Correction in your CLI commands above as this is not related to gyro i2c address.

Yes, manually setting the address via the CLI works using the following command:

set baro_i2c_address = 119 

For those particular FC's like the iFlight or Mateksys boards this could have been done as part of the Custom Defaults for those targets.

This is not an issue I wanted the Betaflight team to spend too much time on and I just wanted to have this documented.

Let us leave it at this. 

I really appreciate the hard work you guys are doing for the community. Thanks.

haslinghuis, 2021-08-23:
Yes this should be added to the unified target config file for the particular board