betaflight/betaflight #10912: strip led don't save configuration

antoniosap, 2021-08-21:
### Describe the bug
strip led don't save configuration,
led and color commands don't exist

### To Reproduce
[image]

### Expected behavior
as manual

### Flight controller configuration

### Setup / Versions
Betaflight / SPRACINGF3 (SRF3) 4.0.6 Sep  1 2019 / 01:05:57 (2a64051a2) MSP API: 1.41 / FEATURE CUT LEVEL 6

etracer65, 2021-08-21:
LED strip is not supported on F3 flight controllers.

closed 2021-08-21 by etracer65

etracer65, 2021-08-21:
@antoniosap F3 flight controllers are no longer supported by Betaflight. Further there is no point continuing to make bug reports for old firmware versions as they also are no longer supported and will not be updated.