betaflight/betaflight #10915: `udev` documentation seems wrong, at least for ArchLinux

andrejpodzimek, 2021-08-23:
The udev rule [mentioned here](https://github.com/betaflight/betaflight/blob/master/docs/USB%20Flashing.md) does _nothing_ (on current ArchLinux). This has been an issue for at least one geological period, but only now did I find the time to take a closer look (instead of blindly adding my user into `uucp` instead of `plugdev` and giving up). There are two reasons why the recommended rule won’t match:

0. `/usr/lib/udev/rules.d/50-udev-default.rules` (owned by `systemd`) takes precedence and sets group `uucp` (instead of `plugdev`) on this line:
   ```
   KERNEL=="tty[A-Z]*[0-9]|ttymxc[0-9]*|pppox[0-9]*|ircomm[0-9]*|noz[0-9]*|rfcomm[0-9]*", GROUP="uucp"
   ```
   So the name of the custom rule file (in `/etc/udev/rules.d`) needs to be (at least) **`51-stdfu-permissions.rules`**; `45` is “not enough”.

1. The subsystem involved is actually `tty`, not `usb`. I suspect that this may have chnaged at some point in `systemd-udevd`, but nonetheless, that’s how it is. There _might_ be something USB-related to it, but it won’t work without matching `tty`.

In summary, here’s my: **`/etc/udev/rules.d/51-stdfu-permissions.rules`**:

```
ACTION=="add", SUBSYSTEM=="usb|tty", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="0664", GROUP="plugdev"
ACTION=="add", SUBSYSTEM=="usb|tty", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE="0664", GROUP="plugdev"
```

This^^^ works for me and sets the expected group and permissions, especially `plugdev` instead of the default `uucp`.

andrejpodzimek, 2021-08-23:
I couldn’t care less, TBH. The only reason why I posted this is to (hopefully) help someone facing the same problem in the future. Meh.

DusKing1, 2021-08-23:
@andrejpodzimek hey don't mind that github action bot, that's just for preventing spam. 

Yeah raising an issue and hope that maybe someone who met the same issue with you can find a company is ok! I'm not on ArchLinux so alas I can't help, but maybe other ppl have a solution. So just don't upset yourself!

swilkens, 2021-08-24:
Given that this wasn't updated since 2015, it seems likely some changes to `systemd` weren't broguht forward here.

@andrejpodzimek The "Linux" section was probably written with Ubuntu in mind, which switched to `systemd` somewhere in 2015 as well I think. Any update you can provide is surely usefull, if you can verify the current state on Ubuntu as well.

PRs welcome, as they say!

haslinghuis, 2021-08-27:
Look at the documentation for [Fedora](https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Fedora.md) or [Ubuntu](https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Ubuntu.md) as they are updated.

Arch Linux docs has not been updated for a long time.

psobiech, 2021-09-27:
Hi,
I actually did not have this problem, since I already had `dfu-util` installed (so installing this package also solves the problem), it provides the following udev rules in: `/usr/lib/udev/rules.d/60-dfuse.rules`:

```
# STM32
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", TAG+="uaccess"

# GD32VF103
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="28e9", ATTRS{idProduct}=="0189", TAG+="uaccess"

# Spark Core
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="607f", TAG+="uaccess"

# On older systems, a user group like "plugdev" can be given access:
# ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="664", GROUP="plugdev"
```

closed 2021-12-06 by github-actions[bot]

andrejpodzimek, 2024-03-03:
To follow up on this (a couple of years later): On ArchLinux `dfu-util` has no effect, mainly because the `uaccess` tag in the `udev` configuration has no effect. (If it were to work, one would see POSIX extended ACLs on the `/dev` node, which is not the case; the node has just the usual u/g/o permissions.)

TL;DR: You have to be in `uucp`. There’s no other way.

haslinghuis, 2024-03-03:
Do you mean replacing `plugdev` with `uucp` to set GROUP="uucp" in udev rules?

Documention for arch linux has not been updated for quite some time:

http://betaflight.com/docs/development/building/Building-in-Arch-Linux

andrejpodzimek, 2024-03-03:
No, there is in fact no `uucp` in `/usr/lib/udev/rules.d/60-dfuse.rules`; `uupc` seems to be just `udev`s default for the device type.

What I meant was to literally add one’s user account to `uucp`, i.e.:

```bash
homectl update -G uucp,<other>,<groups> <username>
```

(The pre-existing “other groups” can be found e.g. using `homectl inspect <username>`.)

The `TAG+="uaccess"` is known to _not_ work, mostly, because it requires (?) correct assignment of USB ports / hubs to “seats”, which doesn’t exist in most distros. IIRC, only Fedora used to get this right, because it has (or had, at some point) good support for multi-seat configurations.

The ideal solution `TAG+="uaccess"` is supposed to provide would add additional POSIX ACLs to STM32’s `/dev` nodes, making each device available (only) to the user logged in at the _seat_ physically controlling the USB port where the STM32 is connected. This fine-grained access would work without a `uucp` membership.

Unfortunately, this↑ (almost) never works or “works” at random, when USB ports happen to be matched against seats and user sessions correctly. So I need a `uucp` membership. (Or I could add `GROUP="plugdev"` instead and get `plugdev` membership.)

haslinghuis, 2024-03-03:
If you like to make a contribution you can add updated instructions here: https://github.com/betaflight/betaflight.com