betaflight/betaflight #10917: Error in drivers/barometer/barometer_qmp6988.c

WilsonChang, 2021-08-24:
### Describe the bug

In:
[https://github.com/betaflight/betaflight/blob/master/src/main/drivers/barometer/barometer_qmp6988.c](https://github.com/betaflight/betaflight/blob/master/src/main/drivers/barometer/barometer_qmp6988.c)
At the end of the file, in:
[image]
`    if (pr)
    *pressure = (int32_t)(pr);
    if (tr)
    *temperature = (int32_t)tr/256;`
These checkings may be mistakes.
### To Reproduce

always
### Expected behavior

should be 
`if (pressure) *pressure = (int32_t)(pr); if (temperature) *temperature = (int32_t)tr/256;`

The purpose is to preven writting to 0 pointers, which are transferred from caller function.
in the meanwhile, local variables " float tr,pr; " will very unlikely to be 0
### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

etracer65, 2021-08-24:
You are misunderstanding the code. There is no need to check for valid pointers for the passed `pressure` and `temperature` variables. This function cannot be called in any way where invalid pointers could happen and it's unnecessary for it to validate these. The code is checking to see if valid (non-zero) values are calculated for pressure and temperature before returning the result under the assumption that these values should never be zero and if they are it's a failure in reading the sensor.

WilsonChang, 2021-08-24:
Maybe I misunderstood it. There are same functions in other barometer drivers code, BMP085, BMP280, MS5611 at least,  all of them check the pointers, but not the values. 
[image]
[image]
[image]

0 may not be invalid value for pressure or temperature logically. If these codes are published and shared with other projects, I would suggest checking 0 pointers which may cause system failure but not values.

etracer65, 2021-08-24:
Ah, I see. Checking for valid pointers is unnecessary for all the functions as they can only be called from one place:

https://github.com/betaflight/betaflight/blob/add05864bfd8aa9c5e40126328beeb840d14119d/src/main/sensors/barometer.c#L434

And the variables used are statics to the module so a invalid pointer is impossible. The zero checks should be removed from all the baro driver calculate functions as they're not needed and would have avoided this confusion. If other projects wish to use this code it's incumbent on them to use it properly and we can't be expected to account for their implementation mistakes with unnecessary extra code in this project.

WilsonChang, 2021-08-25:
@etracer65 
OK, then we can close this issue.

closed 2021-08-25 by WilsonChang