betaflight/betaflight #10922: [Feature Request] ledstrip race color MSP command

JyeSmith, 2021-08-29:
### Is your feature request related to a problem? Please describe.

With multiple RC links now being able to auto configure a quads VTx on power up, which is particularly useful at race events, it would be great to have a similar feature for LEDs.

### Describe the solution you'd like

An MSP packet for configuring ledstrip race color.  Users can then set it up in lua and quads are auto configured on connection.

### Describe alternatives you've considered

I believe this can currently only be done via the configurator.

Discussed on Slack with @haslinghuis

haslinghuis, 2023-02-20:
@ASDosjani this might be something for you

ASDosjani, 2023-02-20:
> @ASDosjani this might be something for you

Uhh, I've never used Lua.

Prajwal-Koirala, 2023-12-27:
**I understand the feature request and its potential benefits for streamlining LED configuration at race events.**

**Here's a visual breakdown of the request:**

**Problem:**

- Manual configuration of LED race colors via configurator is time-consuming and inconvenient at events.
- Lack of automation for LED settings alongside VTx configuration.

**Solution:**

- Introduce an MSP packet specifically for setting LED race colors.
- Enable configuration through Lua scripts for automation.
- Allow for auto-configuration of LEDs upon RC link connection.

**Implementation steps (visualized):**

1. **MSP Packet Development:**
   - Design a new MSP packet structure to handle LED race color settings.
   - Integrate packet handling logic into Betaflight firmware.
   - ****

2. **Lua Script Integration:**
   - Enhance Lua scripting capabilities to send the new MSP packet.
   - Provide examples for users to create scripts for automation.
   - ****

3. **RC Link Interaction:**
   - Trigger LED configuration upon RC link connection events.
   - ****

**Benefits:**

- Streamlined LED configuration at race events.
- Enhanced consistency and compliance with race regulations.
- Reduced setup time and potential errors.
- Improved overall user experience.

ASDosjani, 2023-12-27:
@Prajwal-Koirala please check https://github.com/betaflight/betaflight/pull/13096

closed 2026-09-26 by blckmn