betaflight/betaflight #10925: Custom order of colors for WS2811/WS2812 RGB LED strips

mmabln, 2021-08-29:
### Is your feature request related to a problem? Please describe.

I've made a custom RGB "Racewire" strip connected to my BF powered FC. However, due to layout reasons, the color order of the RGB LEDs is not the usual "R-G-B" but "B-G-R". Unfortunately BF only offer RGB and GRB orders by using

set ledstrip_grb_rgb = RGB

Due to this limitation, my LEDs have now the wrong colors.

### Describe the solution you'd like

I would like to have a more generic approach for the LED orders:

In the config this should be accepted:

set ledstrip_rgb_order = BGR | BRG | GBR | GRB | RBG | RGB

In light_ws2811strip.c the following switch case would need some extension to support the additional orders:

 switch (ledFormat) {
        case LED_RGB: // WS2811 drivers use RGB format
            packed_colour = (color->rgb.r << 16) | (color->rgb.g << 8) | (color->rgb.b);
            break;
        case LED_GRB: // WS2812 drivers use GRB format
        default:
            packed_colour = (color->rgb.g << 16) | (color->rgb.r << 8) | (color->rgb.b);
        break;

### Describe alternatives you've considered

Unfortunately I cannot change my LED PCB to support either RGB or GRB.

etracer65, 2021-08-29:
Seems a bit unreasonable to expect Betaflight to add support for your hardware design mistake when no other hardware would ever need to use this.

mmabln, 2021-08-29:
Why limit the functionality to only switchable between rgb and grb, when it is easy to have it generic? Why was GRB implemented at all, when all "correct" LED strips should have RGB? When I become a LED racewire vendor, and sell my strips with BGR, how many do I have to sell before it is "reasonable"?

I mean Betaflight is a highly configurable software in so many ways, why keep this limitation?

etracer65, 2021-08-29:
> Why was GRB implemented at all, when all "correct" LED strips should have RGB?

Because WS2812 chips (which are widely used) use the GRB color ordering. We're not in the habit of supporting and propagating hardware design mistakes. Since you're not selling your products yet now is the time to fix the design problem.

closed 2021-08-29 by mmabln