betaflight/betaflight #10936: Stats is not being saved when Rate profile adjustments are set (latest Master)

limonspb, 2021-09-01:
### Describe the bug
If Adjustments are set to change rate profiles on a switch, then stats (stats_total_flights) is not being saved on disarm and goes back to the previous value after the power cycle (4.3 branch)

### To Reproduce
1. Make sure you can arm the quad, props off.
2. set stats_min_armed_time_s = 5 
    set stats_total_flights = 1
    save
3. set adjustments, for example like that, save:
[image]
3. unplug USB
4. Arm your quad, keep it armed for 5+ seconds, disarm
5. OSD flights counter will do +1 - good.
6. Unplug the battery, plug it back.
7. Check stats_total_flights via OSD or CLI - it will show the old value, which is 0 in this case.

### Expected behavior
stats_total_flights must be = 1

### Flight controller configuration

```
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Aug 16 2021 / 12:59:06 (add05864b) MSP API: 1.44
# config: manufacturer_id: HENA, board_name: TALONF7V2, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name TALONF7V2
manufacturer_id HENA

# name: LIMON

# resources
resource MOTOR 2 B08
resource MOTOR 3 C08
resource MOTOR 4 B07
resource RX_BIND 1 NONE
resource PINIO 1 A14

# feature
feature TELEMETRY

# serial
serial 0 1 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 64 115200 57600 0 115200

# mixer
mixer QUADX1234

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1925 2075 0 0
aux 1 35 3 1925 2075 0 0
aux 2 39 1 1300 2075 0 0
aux 3 39 1 900 1300 0 0

# adjrange
adjrange 0 0 1 1375 1625 12 1 0 0
adjrange 1 0 1 1875 2100 12 1 0 0

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set gyro_notch1_hz = 242
set gyro_notch1_cutoff = 215
set dyn_notch_q = 150
set dyn_notch_min_hz = 80
set dyn_notch_max_hz = 300
set dyn_lpf_gyro_min_hz = 0
set acc_hardware = NONE
set acc_calibration = 48,-2,-71,1
set max_check = 1950
set serialrx_provider = CRSF
set dshot_bidir = ON
set align_board_pitch = 180
set force_battery_cell_count = 6
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set thrust_linear = 30
set osd_tim2 = 1
set osd_vbat_pos = 2432
set osd_rssi_pos = 26
set osd_link_quality_pos = 2103
set osd_throttle_pos = 2072
set osd_vtx_channel_pos = 1
set osd_current_pos = 391
set osd_mah_drawn_pos = 385
set osd_craft_name_pos = 2081
set osd_warnings_pos = 14634
set osd_avg_cell_voltage_pos = 2400
set debug_mode = GYRO_SCALED
set pinio_box = 39,0,0,0
set gyro_rpm_notch_q = 700
set stats_min_armed_time_s = 5
set stats_total_flights = 10
set stats_total_time_s = 55
set name = LIMON

profile 0

# profile 0
set anti_gravity_gain = 5000
set d_pitch = 30
set f_pitch = 133
set d_roll = 28
set f_roll = 126
set f_yaw = 126
set d_min_roll = 18
set d_min_pitch = 20
set simplified_dterm_filter = ON

rateprofile 1

# rateprofile 1
set rates_type = BETAFLIGHT
set roll_rc_rate = 116
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: PINIO 1
A15: OSD_CS
B00: LED 1
B01: FREE
B02: FREE
B03: CAMERA_CONTROL
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 4
B08: MOTOR 2
B09: FREE
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 3
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: Talon F7 V2, but the bug exists also with Pyrodrone F7 and supposedly all other FCs
 
 - Other components: crossfire receiver, talon ESC.
 
All wired according to the diagram and performs/flies as expected except the stats not being saved.

adarkthinline, 2021-09-01:
I can concure that it is an issue. Talon F7 V2 and Pyro F7 both FC will not keep the counter unless the BF adjustments is turned off.

haslinghuis, 2021-09-01:
Please check `stats_min_armed_time_s` - unless >= 0 it doesn't update `stats_total_flights`

haslinghuis, 2021-09-01:
We should check `setConfigDirty()` in `fc/ rc_adjustments.c`
What was the last version that did work?

limonspb, 2021-09-01:
I tried Master from a month ago - the same result.
It's kind of a corner case using rates on a switch AND stats, i guess.

adarkthinline, 2021-09-01:
***@***.***

Thank you for your inquiry.
I am not sure what you need to know, but performing “get” for suggested CLI phrases this is what appears.

haslinghuis, 2021-09-01:
Please someone check on releases. Start with 4.2. If it works check 4.2.5. If it work's go up, if it doesn't go down.

limonspb, 2021-09-01:
@haslinghuis  i think you are right... Is it kind of a safety feature that prevents saving stats?
Should we remove it?
[image]

haslinghuis, 2021-09-01:
Agree. But it's an expensive operation. So we just have to make sure the dirty flag is set.

haslinghuis, 2021-09-01:
In that case it would be safe to remove the condition here

limonspb, 2021-09-01:
@haslinghuis  as far as I understand this condition serves for 1 reason here. This scenario. Imagine no stats is set and:

1. plug your quad
2. Change PIDs/rates via OSD menu of via switches
3. Don't save - exit no save
3. Fly, land, disarm (with new PIDs/Rates)

At this point a user has an option - he can go ahead and save changes via OSD
or he can unplug if he does not want these changes to be saved.

Now if we introduce stats without checking that condition, it will automatically save everything on disarm.

adarkthinline, 2021-09-03:
I have used this adjustments setup (see image below) for my 3-way position switch to change between rate profiles. It has been setup the same way for these versions of BetaFlight. BF 3.5.7, - BF 4.2.2 and now in development BF 4.3. Currently in BF 4.3 I added the pack counter in CLI and OSD. I tested it (plug in battery, arm, wait 5+ sec, disarm, check stats, counter value increased by 1, re-arm the quad --- repeat). The count value would increase by 1 each test cycle "IF" the battery remained plugged into my quad. "IF" you disconnect the battery and plug in again the pack counter will reset to zero.

When the adjustments are removed the counter works properly.

[image]

----------------------------------------------------------------------------------------------------------------------------------------------

I found this video by Joshua Bardwell...
"Betaflight 4.1 Rate Profile with Jumper T16 6-position switch" https://www.youtube.com/watch?v=F6xTZkAfpR0 

In it he states to setup adjustments like so... (see image below). He also states something along the line that this is the NEW way to setup your adjustments to a switch at least moving forward from 4.1. I did set it up as seen below and tested again as stated above.

The pack counter now works properly.

[image]

----------------------------------------------------------------------------------------------------------------------------------------------
 
I am only speculating, but it seems to me that if there was a change in the code, at some point where the code recognized the switch as a 3-position or 6-position, then it designated the function instead of it needing to be programmed by the user into their radio. If this is the case then by attempted to designate an adjustment for each switch position it is causing an internal conflict.

mikeller, 2021-09-06:
@limonspb: This is a known limitation, and as such I'd classify this as 'by design', and as a 'fix' I think an update of the documentation around statistics is the right approach.
A couple of considerations here:
- assuming that most users fly their craft a lot more than they tune it, the occurrence of this is most likely and edge case;
- the negative impact of having an unwanted configuration saved probably outweighs the negative impact of having inaccurate stats recorded - other events such as crashing / loss of power without disarming first will have the same effect;
- stats are meant to convey a rough idea, and not pinpoint accuracy.

limonspb, 2021-09-06:
> @limonspb: This is a known limitation, and as such I'd classify this as 'by design', and as a 'fix' I think an update of the documentation around statistics is the right approach.
> A couple of [...]

I agree with your statements, @mikeller .
However, @adarkthinline managed to make it work with the different settings for the adjustments... THat one I can't understand :) He described that in his last long message.

AlexandreBonneau, 2022-04-15:
I just got hit with this bug. And as an fpv noob, I think this case must happen pretty often;
I got a new quad, and instead of trying some rates, landing it, unplugging everything, plugging it in betaflight-configurator, change the rate, re-plug it, rinse and repeat, I used the rate profile switch to test 3 different ones at once.

Well, the stats are not saved indeed.

What's weird to me is that on my other quads I could run `set stats = ON`, but on this Mobula7 1S, the `stats` variable does not exist (I flashed that version on it : `# Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Apr  6 2022 / 07:51:10 (2fd2e8eca) MSP API: 1.44`).

It seems to me (still as an fpv newbee), that stats are not settings, they are just...stats. So saving them should not mean saving a configuration, isn't? Perhaps stats should be saved as 'stats data' separated from the configuration settings?

haslinghuis, 2023-12-17:
@limonspb can we close this?

limonspb, 2023-12-19:
> @limonspb can we close this?

pretty sure yes.

closed 2023-12-19 by haslinghuis