betaflight/betaflight #10946: STM32G47X FLASH write not working with blackbox_mode=NORMAL

SteveCEvans, 2021-09-05:
See https://github.com/betaflight/betaflight/pull/10909 for background.

Using an AIRFLEETG4 V11 FC with NERC-HYBRIDG4 config logging to onboard FLASH works using `blackbox_mode=MOTOR_TEST` however with `blackbox_mode=NORMAL`, arming via MSP, only the log header is written as per the attached log. Motor output set to OneShot125 works, so this appears to be an issue specific to DSHOT.

[STM32G47X_DSHOT.bbl.zip](https://github.com/betaflight/betaflight/files/7111621/STM32G47X_DSHOT.bbl.zip)

DusKing1, 2021-09-05:
@github-action fk off

SteveCEvans, 2022-05-29:
From some brief testing it would appear that black box recording stops randomly unless `blackbox_sample_rate` is set to `1/1`. This will require some investigation.

SteveCEvans, 2022-05-30:
When debugging I'm seeing that the logging stops due to the following being called in msp.c. The question is why is the configurator disarming if I change tabs? @limonspb ?

[image]

limonspb, 2022-05-31:
> When debugging I'm seeing that the logging stops due to the following being called in msp.c. The question is why is the configurator disarming if I change tabs? @limonspb ?
> 
> [image]

Could you check is that with any receiver? Or "MSP receiver"? If its with any receiver I can check in a bit when i get home.

SteveCEvans, 2022-05-31:
I’ll add an RX this evening and check with that.

SteveCEvans, 2022-05-31:
@limonspb I've wired up an ELRS RX and switching between tabs is fine until I select the Motors tab at which point the FC disarms. Otherwise I can't reproduce this issue.

haslinghuis, 2022-05-31:
Is that with motor testing enabled in motors tab?

SteveCEvans, 2022-05-31:
Yes

closed 2026-09-26 by blckmn