betaflight/betaflight #10951: Onboard flash not found with IFLIGHT_F411_PRO firmware update

tzheng777, 2021-09-07:
### Describe the bug
Updating the firmware on an iFlight SucceX Micro F4 V2.1 FC disables the 8mb onboard flash for blackbox logging, i.e. the configurator shows the message "No dataflash chip found". 

Out of the box, the FC comes with BTFL 4.2.9, target: IFRC/IFLIGHT_F411_PRO(STM32F411) and the onboard flash definitely works with the configurator showing 8mb available. However, the CLI kept stating that no custom defaults were loaded and so I proceeded to update the firmware to fix this. Upon updating, this issue became apparent. This behaviour has been confirmed on two FCs. 

### To Reproduce
1. With a new untouched FC, plug in via USB to configurator and check that the onboard flash is recognised to rule out hardware fault. 
2. Enter the "update firmware" mode and update firmware with the target IFLIGHT_F411_PRO, version 4.2.9
3. Apply custom defaults when prompted after finishing the update. 
3. Once finished updating, reconnect in configurator in normal USB mode and observe that the error "No dataflash chip found" now appears.  

### Expected behavior
Onboard flash should be still recognised on the configurator, thus allowing blackbox logging to it. 

### Flight controller configuration
```
diff all

# version
# Betaflight / STM32F411 (S411) 4.2.9 Apr 27 2021 / 19:33:23 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F411_PRO, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_F411_PRO
manufacturer_id IFRC
mcu_id 002200433236510239313331
signature 

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 - Flight controller:  iFlight SucceX Micro F4 V2.1 
 - Other components: iFlight SucceX Micro VTX, iFlight SucceX Micro 4in1 15A ESC
 - How are the different components wired up: All three components are connected together via headers, as they are specifically designed to be used together as a full stack. 

I have 5 of these stacks in total, with the bug confirmed by performing the firmware update on two of the boards. I still have 3 boards with working onboard flash that have not been updated. If requested I can upload the diff and resource dumps from the untouched FCs.

tzheng777, 2021-09-07:
Comparing the resource lists between the working and nonworking FCs, I found that pin A15 is FLASH_CS on the working FC. 

On the faulty updated FC, pin A15 is PREINIT 2 instead. Attempting to manually set the pin to FLASH_CS using `resource FLASH_CS A15` does not seem to do anything. 

If FLASH_CS is unrelated to the onboard flash then please disregard.

etracer65, 2021-09-07:
Post a `diff` and `resource show all` from a working config.

mikeller, 2021-09-07:
@tzheng777: Can you check if `dma ADC 1 0` and then `save` in CLI will make it work?

tzheng777, 2021-09-08:
@etracer65: This didn't seem to work unfortunately. It does save the dma assignment, but unfortunately pin A15 is still listed as PREINIT 2 when using `resource show all`

@mikeller: As requested, from an FC straight out of the box:

```
diff all
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F411 (S411) 4.2.9 May  5 2021 / 16:39:36 (norevision) MSP API: 1.43
###ERROR: diff: NO CONFIG FOUND###
# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_F411_PRO
manufacturer_id IFRC
mcu_id 006300483236510339313331
signature 

# resources
resource BEEPER 1 B02
resource MOTOR 1 A00
resource MOTOR 2 B10
resource MOTOR 3 B06
resource MOTOR 4 B07
resource PPM 1 A03
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C13
resource LED 2 C14
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 B00
resource ADC_CURR 1 B01
resource FLASH_CS 1 A15
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A01
resource GYRO_CS 1 A04
resource USB_DETECT 1 C15

# timer
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer A00 AF1
# pin A00: TIM2 CH1 (AF1)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin A00 0
# pin A00: DMA1 Stream 5 Channel 3
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature OSD

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# aux
aux 0 0 0 900 1300 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 BAND_D   D CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set acc_calibration = 30,-22,197,1
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 1
set baro_hardware = NONE
set serialrx_provider = SBUS
set blackbox_p_ratio = 16
set blackbox_device = SPIFLASH
set dshot_burst = AUTO
set dshot_bitbang = OFF
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 100
set beeper_inversion = ON
set beeper_od = OFF
set system_hse_mhz = 8
set debug_mode = GYRO_SCALED
set max7456_spi_bus = 2
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: GYRO_EXTI
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: MOTOR 2
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: MOTOR 1
    CH3: MOTOR 2
TIM3: FREE
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

mikeller, 2021-09-08:
@tzheng777: Thanks for the information.

```
# Betaflight / STM32F411 (S411) 4.2.9 May  5 2021 / 16:39:36 (norevision) MSP API: 1.43
```

As the `norevision` shows, this version of Betaflight is not an officially released version. As such there is no way that we can support you with this as we lack the required information.
But in order to comply with the license conditions of Betaflight the manufacturer or your supplier is required to provide you with the source code of this version so that you can build your own updated versions. Please contact them.

tzheng777, 2021-09-08:
@mikeller: Thanks for the heads up, I will contact the manufacturer for support then. Which settings are you referring to that may cause injury/ are illegal?

tzheng777, 2021-09-15:
The manufacturer has gotten back to me with a custom firmware build that allows the onboard flash to be recognised again. This was their statement on what they think is the issue:
> During the chip shortage we were forced to use a different flash chip for a short period. This seems not to affect BF 4.3.0 but 4.2.9.

daleckystepan, 2022-01-11:
@tzheng777 Can you check what flash chips has been used on your board? Thank yu

hsuenaga, 2022-01-28:
@daleckystepan I have same board. My own board has Winbond W25Q64JVXGIM. A picture on manufacturer's site uses W25Q64JVXGIQ. These chips are almost compatible, but device id is different. There is also difference around Quad SPI supporting, 'M' variant is more flexible but need additional configuration to use Quad SPI.

hsuenaga, 2022-01-30:
device id of W25Q64JVXGIM has added by commit 2b21e2ff706b5455523d8e09e3dc57adb58c72cc.
I applied following patch on 4.2-maintain branch and it worked fine.
It's not a kind of BUG, just manufacturer's fault.

```
diff --git a/src/main/drivers/flash_m25p16.c b/src/main/drivers/flash_m25p16.c
index 8462a999d..3f7a70d87 100644
--- a/src/main/drivers/flash_m25p16.c
+++ b/src/main/drivers/flash_m25p16.c
@@ -63,6 +63,7 @@
 #define JEDEC_ID_WINBOND_W25Q16        0xEF4015
 #define JEDEC_ID_WINBOND_W25Q32        0xEF4016
 #define JEDEC_ID_WINBOND_W25Q64        0xEF4017
+#define JEDEC_ID_WINBOND_W25Q64M       0xEF7017
 #define JEDEC_ID_WINBOND_W25Q128       0xEF4018
 #define JEDEC_ID_WINBOND_W25Q128_DTR   0xEF7018
 #define JEDEC_ID_CYPRESS_S25FL128L     0x016018
@@ -196,6 +197,7 @@ bool m25p16_detect(flashDevice_t *fdevice, uint32_t chipID)
         break;
     case JEDEC_ID_MICRON_N25Q064:
     case JEDEC_ID_WINBOND_W25Q64:
+    case JEDEC_ID_WINBOND_W25Q64M:
     case JEDEC_ID_MACRONIX_MX25L6406E:
         fdevice->geometry.sectors = 128;
         fdevice->geometry.pagesPerSector = 256;
```

daleckystepan, 2022-02-07:
It is already fixed in BF 4.3 https://github.com/betaflight/betaflight/pull/10698