betaflight/betaflight #10961: betaflight_4.3.0_STM32H743-2520-a2bbd31c9 fiirmware can't set sdcard_mode to SPI

jamming, 2021-09-10:
I flash a STM32H743 unified_targets firmware 'betaflight_4.3.0_STM32H743-2520-a2bbd31c9.hex' into a STM32H743 board. It can connect to Betaflight Configurator at the begin. But once I type in the command "set sdcard_mode = SPI" and "save" in CLI. The STM32H743 board will lock-up  in the next reboot and No serial port was identified again. I have to re flash the firmware. 

It seems there is a conflict with the SDIO.

vincentpoont2, 2021-09-14:
Hi @Asizon @mikeller , would you please assist with this issue? Thank you!

DusKing1, 2021-09-15:
This can be reproduced on MATEKH743.

jamming, 2021-09-16:
I build a KakuteH7 target use 'legacy' format target definition in src/main/target/.
The features in Target.mk were set to "FEATURES       += VCP SDCARD_SPI "h
The board work well with this firmware.