betaflight/betaflight #10971: Battery not detected on TMVELOXF7/STM32F7X2

DocBraaa, 2021-09-19:
### Describe the bug
I got a T-Motor Velox F7 FC and i can't see the voltage. My configuration is OK but it still not working on multiple version of betaflight for TMVELOXF7. I've teste to install an other firmware (TMOTORF7) and the voltage of the battery is showing à

### To Reproduce
Got this FC and plus battery..

### Expected behavior
Have the voltage showing
diff
set dump
### Flight controller configuration

```
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.7 May 28 2020 / 15:06:20 (9ba02a587) MSP API: 1.42
# manufacturer_id: TMTR   board_name: TMVELOXF7   custom defaults: YES

# start the command batch
batch start

board_name TMVELOXF7
manufacturer_id TMTR

# name: ZEUBEE

# feature
feature -LED_STRIP

# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 1 900 1300 0 0
aux 2 35 1 1700 2100 0 0
aux 3 36 2 1700 2100 0 0

# master
set rc_smoothing_derivative_type = 
###ERROR: CORRUPTED CONFIG: rc_smoothing_derivative_type = 3###
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = 
###ERROR: CORRUPTED CONFIG: gps_sbas_mode = 5###
set name = ZEUBEE

profile 0

# profile 0
set p_yaw = 45

rateprofile 0

# rateprofile 0
set tpa_breakpoint = 1350

# end the command batch
batch end```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: MOTOR 4
B00: PINIO 1
B01: MOTOR 1
B02: PREINIT 2
B03: MOTOR 3
B04: MOTOR 2
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: CAMERA_CONTROL
B09: PINIO 2
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_CURR
C01: ADC_BATT
C02: FREE
C03: GYRO_EXTI
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FLASH_CS
C14: BEEPER
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: MOTOR 4
    CH2: MOTOR 3
TIM3:
    CH1: MOTOR 2
    CH4: MOTOR 1
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#```

### Setup / Versions
 
 - Flight controller: T-Motor - Velox F7 + V45A V2 4in1 Stack
 
 - Other components:
 - TBS Sixty9
 -  JHE42B_S Buzzer
 - Caddx Ratel 2
 
 - How are the different components wired up:
Ratel 2 is wired to the FC
Sixty9 too, Buzzer too

As i said, i see the volt (and its the real one) if the firmware installed is TMOTORF7 so its not a hardware issu..

PS: Sorry for my english :)

apeelme, 2021-09-19:
Same problem here! :(

kafie1980, 2021-09-20:
We had someone report this on our forum: https://intofpv.com/t-velox-f7-fc-doesn-t-show-voltage?pid=153751#pid153751

 The target config for the board is missing the correct "dma ADC 1" entry : https://github.com/betaflight/unified-ta...XF7.config

Try setting via CLI the following to see if this works as it worked for the person who asked that question on our forum:

dma ADC 1 1
save

Spitfire4466, 2021-09-20:
> Try setting via CLI the following to see if this works as it worked for the person who asked that question on our forum:
> 
> ```
> dma ADC 1 1
> save
> ```
Thank you sir, worked for me too!

DocBraaa, 2021-09-20:
Worked for me, thank you so much, i love open-source <3

closed 2021-09-20 by DocBraaa