betaflight/betaflight #10975: Automatic throttle down after given angle

gbbulciolu, 2021-09-19:
Feature Request: 
Allow to set an angle limit (example +- 90deg.) on selected axes after which the flight controller sets the throttle down to 1000 (or a predetermined value) still keeping air mode or other flight aids active as normally, and sets back the throttle to what it was before once the angle returns to a "safe" value. This would certainly help beginners not to destroy their new build (me) and pros for more and maybe new fancy maneuvers.
Cheers

p.s. sorry if this was already requested in the past.

etracer65, 2021-09-19:
This is a better solution for beginners: https://github.com/betaflight/betaflight/wiki/Acro-Trainer

closed 2021-09-19 by haslinghuis

KarateBrot, 2021-09-19:
> this feature would help free style beginners with misuse of throttle when performing maneuvers and pros for a more relaxed throttle management.

If you're good at flying you already have a relaxed throttle management in your fingers. No one half decent at flying would use this because it's taking throttle control away from the user.