betaflight/betaflight #10980: Buzzer after arm

terry-dactle, 2021-09-22:
### Is your feature request related to a problem? Please describe.
The buzzer is incredibly important in case you lose the drone, however I find myself and everyone else finds it incredibly annoying to test drone features without the transmitter present and on as it is constantly beeping loud, issue can be found even when turning on the drone before transmitter

### Describe the solution you'd like
Would be nice if we could have the option to only have the RX_LOST only active after arming.

### Describe alternatives you've considered
If GPS is present only allow RX_LOST if the drone has moved more than 20m as this is a very searchable area.

dkustec, 2021-09-22:
Check this feature: #10225

etracer65, 2021-09-22:
> Would be nice if we could have the option to only have the RX_LOST only active after arming.

This is already the way it works. The RX_LOST beeper will only sound if the quad is currently or has been already armed during the current battery.

You probably just have a poor configuration for your RX_SET beeper in the modes where you have it activated at 1500 which is the same as the aux channels get set to when there's no radio signal. Set your BEEPER mode to something other than the middle of the range. You can see what's happening by looking at the modes tab and turning your radio on/off.

closed 2021-09-22 by haslinghuis