betaflight/betaflight #10981: 30.5 Iflight Beast logging/diversity rx issues in 4.3

bad5life, 2021-09-22:
### Describe the bug

Logging not working with the Iflight beast 30.5, also 2x tbs diversity rx (6.09)were giving badrx connection/jumpy reading 1-2x a second laggy inputs in the receiver menu (betaflight 4.3) The diversitys work fine on other 4.3 FC's so im guessing its the beasts BF4.3 build? I soldered a regular nano rx to the beast and it was working fine again... weird issue but 2x diversity's did the same thing and the nano rx is using the same wiring

There is usually data in the BB log, but it is corrupted.... usually a smaller BB file than expected also 
### To Reproduce

Try to log with beast 30.5 or use tbs diversity reciever in 4.3
### Expected behavior

BB Logs /diversity working

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F745 (S745) 4.3.0 Sep  9 2021 / 15:18:20 (a2bbd31c9) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F745_AIO, version: 67512b06, date: 2021-02-08T08:35:22Z

# start the command batch
batch start

board_name IFLIGHT_F745_AIO
manufacturer_id IFRC

# feature
feature -AIRMODE
feature GPS

# serial
serial 0 0 115200 57600 0 115200
serial 2 1 115200 57600 0 115200

# beacon
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 13 4 1700 2100 0 0
aux 3 28 3 900 1300 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# rxfail
rxfail 11 s 1000

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 60
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set acc_calibration = 39,22,-53,1
set rssi_channel = 12
set rc_smoothing_setpoint_cutoff = 45
set serialrx_provider = CRSF
set min_throttle = 1066
set dshot_idle_value = 400
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1400
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter = ON
set osd_rssi_pos = 2432
set osd_current_pos = 2423
set osd_mah_drawn_pos = 2393
set osd_gps_speed_pos = 2368
set osd_gps_lon_pos = 2537
set osd_gps_lat_pos = 2533
set osd_gps_sats_pos = 2400
set osd_home_dir_pos = 2061
set osd_home_dist_pos = 2062
set osd_altitude_pos = 2336
set osd_avg_cell_voltage_pos = 2464
set osd_battery_usage_pos = 2065
set gyro_rpm_notch_min = 50

profile 0

# profile 0
set p_pitch = 70
set i_pitch = 60
set d_pitch = 70
set f_pitch = 40
set p_roll = 60
set i_roll = 60
set d_roll = 55
set f_roll = 40
set p_yaw = 85
set i_yaw = 75
set f_yaw = 40
set d_min_roll = 0
set d_min_pitch = 0
set simplified_dterm_filter = ON

rateprofile 0

# rateprofile 0
set roll_rc_rate = 14
set pitch_rc_rate = 14
set yaw_rc_rate = 18
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 50
set pitch_srate = 55
set yaw_srate = 45
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

# end the command batch
batch end

```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: CAMERA_CONTROL
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: LED 1
C14: FREE
C15: FREE
D00: GYRO_EXTI
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: FREE
E02: SPI_SCK 4
E03: FREE
E04: OSD_CS
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: SERIAL_RX 7
E08: SERIAL_TX 7
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1 : LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : CAMERA_CONTROL
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: DSHOT_BITBANG 5
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: Iflight Beast 30.5 IFLIGHT_F745_AIO on 4.3 2520 build
 
 - Other components: Caddx Vista Polar, TBS Diversity Rx (6.09)
 
 - How are the different components wired up:
With wires and solder

haslinghuis, 2021-09-22:
Did you have V1 or V2? Please try: https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFLIGHT_F745_AIO_V2.config

bad5life, 2021-09-22:
This is 30.5 version of beast that i thought only used v1 version firmware? could be wrong tho.... After further diversity testing on another quad/FC on 4.3 im still getting supper laggy like 5hz inputs so im thinking it might end up being a 6.09 diversity bug... hmm

closed 2021-10-30 by github-actions[bot]