betaflight/betaflight #10982: Increase Ampere Sense Limit

jasc76, 2021-09-22:
Hello

The new configurator warns about the max voltage of 25.5 and an Ampere limit of 63.5
There are these two pull requests: 
https://github.com/betaflight/betaflight-configurator/pull/2144
https://github.com/betaflight/betaflight/pull/10025

I noticed even before on my bigger builds with Tekko32 4in1 Metal with 65A per Channel that the Ampere Meter is capped. Of course for some time now 4in1 dont have individual shunt's anymore and rely on one shunt with the ADC current pin to report the amperes used and notby blheli telemetry. 

Is there any way to enable at least 4*65A = 260A measuring capability ?

etracer65, 2021-09-22:
It's just a display limit **in the configurator only**. They are not limited in Betaflight or in the OSD display. So this is only a problem if you're flying while connected to the configurator and looking at the battery tab - apparently with a really long USB cable. So in other words, don't worry about it as it's not a problem.

closed 2021-09-22 by haslinghuis

jasc76, 2021-09-27:
Ok, thx
but it seems in BB logs there is a cap... So there is also no cap there?