betaflight/betaflight #10994: Kakute F7 HDV can't see ESC on nightly from 30.09.21

maxbl4, 2021-10-01:
## If you believe there is an issue with the firmware itself please follow these steps:
1. Flashed latest 4.3 nightly (30.09.2021) and Kakute F7 HDV fails to communicate with mamba F50 ESC. ESC only gives two beeps, rpm telemetry shows 100% errors, motors don't spin from configuration both with RPM telemetry and without. This looks like a regression, since flashing back to 4.2.9 everything works and flashing older nightly from 04.08.2021 also works fine. I tried flashing back and forth several times, of course settings motor settings, it always work in 4.2.9 and older 4.3, but does not work on latest 4.3
2. Flash latest 4.3 nightly (30.09.2021)

Martivip, 2021-10-03:
I confirm the issue I have the same issue, precisly speaking it stops working since BF 4.3 build no. #2525 with earlier builds is OK.
My target is : HBRO/KAKUTEF7(STM32F745)

SteveCEvans, 2021-10-03:
Interesting use of SPI busses. SD card is on SPI bus 1, gyro on SPI bus 4. It can be seen from the `dma show` that `SPI MISO 4` is not being allocated. Bus 1 is normally used for the gyro. Having two busses on DMA controller 2 is limiting options for ADC and DSHOT channels. That said the two DSHOT bitbang channels on timer 8 are being allocated.

Could you please try `dshot_bitbang=OFF` and also disable the ADC on the battery tab. Does that assign four motor channels with `dma show`?

Martivip, 2021-10-03:
ok I disabled dshot_bitbang=OFF and also  ADC on the battery tab.

DMA location looks like this, and still no communication with ESC

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: MOTOR 4
DMA2 Stream 3: MOTOR 3
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

SteveCEvans, 2021-10-03:
Thanks. I've just realised that the motor allocation is the same as an IFLIGHT_F745_AIO, which I have to facilitate debug. I'll take a look now.

SteveCEvans, 2021-10-03:
Just tested the same build of master on the IFLIGHT_F745_AIO using BLHeli-M 16.9 and the Motors tab reports zero errors. Motors are on the same pins as you report above.

```
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
```

The significant difference is that a different bus is being used for the gyro and you have the SD card connected using SPI bus 1.

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: DSHOT_BITBANG 5
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

Can I suggest you go back to your prior config, but disable the SD card using `set sdcard_spi_bus = 0`. Your config should then be very similar to what I'm using which will make ongoing debug a bit easier.

Martivip, 2021-10-03:
ok after that operation set sdcard_spi_bus = 0 I can not establish USB connection with FC and I has been forced to reflash FC using manual DFU mode

maxbl4, 2021-10-03:
@SteveCEvans but we found particular commit that introduces the issue, did you try to look into changes?

SteveCEvans, 2021-10-03:
I'm trying to reproduce the issue here on the only F745 hardware I have, looking for differences between that and your FC. Setting the SPI bus for a device to zero should simply take it out of the equation. Looks like there some side effect for SD.

Martivip, 2021-10-04:
I've notice one weird behavior, sometimes after fresh FC flash communication with ESC works till first repower of the FC.
And of course on earlier nightly builds (before 2525) there is no such a issue.

maxbl4, 2021-10-04:
@Martivip yes, found same thing

Martivip, 2021-10-06:
@SteveCEvans did you successfully reproduce the issue ? if I could help I'm here :)

Martivip, 2021-10-06:
@SteveCEvans I think that issue is somehow connected with this issue https://github.com/betaflight/betaflight/issues/11003 here we have the same issue, when CALIB flag is active then there is no communication with ESC and usually this CALIB flag become active after repower of the FC

Martivip, 2021-10-07:
@SteveCEvans  any advise how to start those investigations and isolate the issue  ? I will be really apreciated for any clues, alter fresh FC flash before I apply below CLI settings the CPU utilization is stable.

those my CLI settings
```
# 

# diff all

# version
# Betaflight / STM32F745 (S745) 4.3.0 Oct  3 2021 / 04:50:54 (d2613e2e6) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7
manufacturer_id HBRO
mcu_id 003700263538510a38333932
signature 

# resources
resource MOTOR 5 NONE
resource CAMERA_CONTROL 1 B03
resource PINIO 1 C09

# timer
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)

# dma
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 0 2048 115200 57600 0 115200
serial 1 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# led
led 0 4,11::COW:2
led 1 5,11::COW:2
led 2 6,11::COW:10
led 3 7,11::COW:10
led 4 8,11::COW:10

# aux
aux 0 0 0 2050 2100 0 0
aux 1 1 2 1475 1525 0 0
aux 2 46 3 1200 1650 0 0
aux 3 13 3 1900 2000 0 0
aux 4 13 3 1675 1750 0 0
aux 5 13 3 1425 1525 0 0
aux 6 13 3 1200 1275 0 0
aux 7 15 3 2025 2075 0 0
aux 8 15 3 1900 1950 0 0
aux 9 15 3 1775 1825 0 0
aux 10 15 3 1650 1700 0 0
aux 11 15 3 1525 1575 0 0
aux 12 15 3 1425 1475 0 0
aux 13 15 3 1300 1350 0 0
aux 14 15 3 1175 1225 0 0
aux 15 35 2 2050 2100 0 0
aux 16 40 3 1650 1875 0 0
aux 17 40 3 1175 1400 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK I FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6C    X CUSTOM  5658 5695 5760 5800 5880 5917    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 3 h

# master
set gyro_lpf2_static_hz = 1000
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 500
set gyro_lpf1_dyn_max_hz = 1000
set acc_trim_roll = -7
set acc_calibration = -63,7,70,1
set align_mag = CW90FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 900
set baro_hardware = AUTO
set fpv_mix_degrees = 25
set serialrx_provider = GHST
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 3
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1300
set vbat_warning_cell_voltage = 340
set ibata_scale = 148
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 35
set gps_rescue_descent_dist = 100
set gps_rescue_throttle_min = 1000
set gps_rescue_throttle_max = 1400
set gps_rescue_throttle_hover = 1255
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_dth = 80
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter_multiplier = 200
set osd_warn_batt_not_full = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_alarm = 30
set osd_link_quality_alarm = 30
set osd_rssi_dbm_alarm = -96
set osd_cap_alarm = 1150
set osd_alt_alarm = 120
set osd_tim1 = 0
set osd_tim2 = 1
set osd_vbat_pos = 2168
set osd_link_quality_pos = 2233
set osd_rssi_dbm_pos = 2265
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2497
set osd_anti_gravity_pos = 2145
set osd_throttle_pos = 2113
set osd_vtx_channel_pos = 2101
set osd_current_pos = 2092
set osd_mah_drawn_pos = 2200
set osd_gps_sats_pos = 2088
set osd_home_dir_pos = 2127
set osd_home_dist_pos = 2129
set osd_flight_dist_pos = 2121
set osd_altitude_pos = 2241
set osd_warnings_pos = 2377
set osd_avg_cell_voltage_pos = 2136
set osd_battery_usage_pos = 2505
set osd_disarmed_pos = 2475
set osd_flip_arrow_pos = 2502
set osd_rate_profile_name_pos = 2081
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5917
set vcd_video_system = PAL
set camera_control_internal_resistance = 570
set pinio_box = 40,41,42,43
set gyro_1_align_yaw = 2700

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 150
set dterm_lpf1_dyn_max_hz = 300
set dterm_lpf2_static_hz = 300
set anti_gravity_gain = 4500
set p_pitch = 70
set i_pitch = 125
set f_pitch = 163
set p_roll = 67
set i_roll = 120
set d_roll = 39
set f_roll = 156
set p_yaw = 67
set i_yaw = 120
set f_yaw = 156
set d_min_roll = 39
set d_min_pitch = 46
set simplified_d_gain = 130
set simplified_pi_gain = 150
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 130
set simplified_pitch_d_gain = 120
set simplified_dterm_filter_multiplier = 200

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 30
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 70
set tpa_breakpoint = 1750

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

Martivip, 2021-10-08:
@maxbl4 please upload to your FC enclosed files via configurator and Load Firmware Local button, first load your target config file then firmware .hex file and then Flash Firmware

Download:
[Test HEX.zip](https://github.com/betaflight/betaflight/files/7312944/Test.HEX.zip)

And if you would like to know where you can find artifacts files please look in to that issue https://github.com/betaflight/betaflight/issues/11003 SteveCEvans precisely explaining here how to download artifacts files, and unified targets you will find here: https://github.com/betaflight/unified-targets/tree/master/configs/default

maxbl4, 2021-10-08:
Loaded binary from #11006 work fine, motors work, CPU load without RPM filter is 2%, with RPM filter 8%. But tasks shows gyro loop can't keep up. With RPM filter:
`Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       2       0    0.5%    0.0%         0      0     50       0
01 - (         SYSTEM)    982     328       1   32.7%    0.5%         5      1   4728       1
02 - (           GYRO)   7874      16       8   13.0%    6.7%       243      0  37875       0
03 - (         FILTER)   7874      26      15   20.9%   12.3%       566      0  37875       0
04 - (            PID)   7874      53      38   42.2%   30.4%      1469      0  37875       0
05 - (            ACC)    982      20       9    2.4%    1.3%        43      1   4728       9
06 - (       ATTITUDE)     99      11       7    0.6%    0.5%         3      0    478       7
07 - (             RX)     14      21       7    0.5%    0.5%         2     10    292       7
08 - (         SERIAL)     99  363962       2 3603.7%    0.5%       666     19    477       2
09 - (       DISPATCH)    982       3       1    0.7%    0.5%         2      0   4728       0
10 - (BATTERY_VOLTAGE)     49       4       1    0.5%    0.5%         0      0    241       1
11 - (BATTERY_CURRENT)     49       3       0    0.5%    0.0%         0      0    241       1
12 - ( BATTERY_ALERTS)      7       2       0    0.5%    0.0%         0      0     26       0
13 - (         BEEPER)     99       4       0    0.5%    0.0%         0      0    476       1
16 - (           BARO)     38      30      28    0.6%    0.6%        11     25    574      28
18 - (       ALTITUDE)     38       8       6    0.5%    0.5%         0      0    192       6
20 - (      TELEMETRY)    246       8       1    0.6%    0.5%         1      0   1183       1
23 - (            OSD)     41      13       6    0.5%    0.5%         0      1    287       6
25 - (            CMS)     18       3       1    0.5%    0.5%         0      0     97       1
26 - (        VTXCTRL)      7       1       0    0.5%    0.0%         0      0     26       0
27 - (        CAMCTRL)      7       1       0    0.5%    0.0%         0      0     26       0
29 - (    ADCINTERNAL)      5       4       0    0.5%    0.0%         0      0      6       0
RX Check Function                   2       0                         0
Total (excluding SERIAL)                       119.7%   55.8%`

maxbl4, 2021-10-08:
Tasks with RPM filter from 4.2.9
```
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms
00 - (         SYSTEM)     10       2       0    0.5%    0.0%         0
01 - (         SYSTEM)    998       4       1    0.8%    0.5%        77
02 - (           GYRO)   8000      19       8   15.7%    6.9%      4942
03 - (         FILTER)   8000      15      11   12.5%    9.3%      6099
04 - (            PID)   8000      47      36   38.1%   29.3%     20508
05 - (            ACC)    999      12       9    1.6%    1.3%       644
06 - (       ATTITUDE)    100       9       7    0.5%    0.5%        46
07 - (             RX)     33      32      31    0.6%    0.6%        71
08 - (         SERIAL)    100    1062       2   11.1%    0.5%       722
09 - (       DISPATCH)    997       3       0    0.7%    0.0%        32
10 - (BATTERY_VOLTAGE)     50       3       1    0.5%    0.5%         4
11 - (BATTERY_CURRENT)     50       3       1    0.5%    0.5%         2
12 - ( BATTERY_ALERTS)      5       2       1    0.5%    0.5%         0
13 - (         BEEPER)    100       3       1    0.5%    0.5%         6
16 - (           BARO)    123      26      16    0.8%    0.6%       137
18 - (       ALTITUDE)     40      10       6    0.5%    0.5%        18
20 - (      TELEMETRY)    250       3       1    0.5%    0.5%        22
23 - (            OSD)     60      12       3    0.5%    0.5%        11
25 - (            CMS)     20       3       1    0.5%    0.5%         1
26 - (        VTXCTRL)      5       1       1    0.5%    0.5%         0
27 - (        CAMCTRL)      5       2       1    0.5%    0.5%         0
29 - (    ADCINTERNAL)      1       3       2    0.5%    0.5%         0
30 - (       PINIOBOX)     20       3       1    0.5%    0.5%         0
RX Check Function                   2       1                         2
Total (excluding SERIAL)                        77.8%   55.0%
```

maxbl4, 2021-10-08:
Btw, I have DJI Air Unit, and possibly it's MSP requests also contribute to the CPU load

Martivip, 2021-10-08:
@SteveCEvans ok we know what setting causes this high CPU load, this is GPS functionality. Have you been able to take a closer look at this issue ? should I create another new Issue for this problem ?

Martivip, 2021-10-08:
Ok the new Issue for this problem has been created https://github.com/betaflight/betaflight/issues/11010

SteveCEvans, 2021-10-08:
@maxbl4 The gyro loop is keeping up just fine. The scheduler is locked to the gyro interrupt so won't necessarily be reporting 8kHz any more. Please try this image (still building under https://github.com/betaflight/betaflight/pull/11007).

[betaflight_4.3.0_STM32F745_38ca006.hex.zip](https://github.com/betaflight/betaflight/files/7314634/betaflight_4.3.0_STM32F745_38ca006.hex.zip)

The please post the output of `status`, `tasks` (capture the second innovation) and `dma show` with your motors enabled.

maxbl4, 2021-10-09:
Flashed the binary above. With GPS enabled, observe the same load spikes up to 100% at times
```
# status
MCU F745 Clock=216MHz, Vref=3.29V, Core temp=60degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4095, max available: 32768
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MSP
System Uptime: 88 seconds, Current Time: 2021-10-09T06:40:38.639+00:00
CPU:10%, cycle time: 128, GYRO rate: 7812, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: None inserted
Arming disable flags: RXLOSS CLI MSP
```
```
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       3       0    0.5%    0.0%         0      0   1355       0
01 - (         SYSTEM)    982     328       1   32.7%    0.5%       157      1 133100       1
02 - (           GYRO)   7874      19       9   15.4%    7.5%      9363      0 1065641       0
03 - (         FILTER)   7874      34      18   27.2%   14.6%     19649      0 1065641       0
04 - (            PID)   7874      71      52   56.4%   41.4%     55629      0 1065641       0
05 - (            ACC)    982      18       9    2.2%    1.3%      1196      6 133017       9
06 - (       ATTITUDE)    491      17       7    1.3%    0.8%       474      5  66561       7
07 - (             RX)     15      25      11    0.5%    0.5%        77     60   8120      11
08 - (         SERIAL)     99  373527       1 3698.4%    0.5%       755    523  13340       1
09 - (       DISPATCH)    982       7       1    1.1%    0.5%        58      0 133108       1
10 - (BATTERY_VOLTAGE)     50       6       2    0.5%    0.5%         9      0   6749       1
11 - (BATTERY_CURRENT)     50       3       1    0.5%    0.5%         4      0   6749       1
12 - ( BATTERY_ALERTS)      5       4       1    0.5%    0.5%         0      0    678       1
13 - (         BEEPER)     99       7       1    0.5%    0.5%        13      1  13339       1
14 - (            GPS)     99      57      22    1.0%    0.7%       109     98  13404      22
16 - (           BARO)     40      32      27    0.6%    0.6%       307     57  16150      27
18 - (       ALTITUDE)     40      13       7    0.5%    0.5%        36      0   5399       7
20 - (      TELEMETRY)    491      20       0    1.4%    0.0%        72     56  66352       0
23 - (            OSD)     59      90      30    1.0%    0.6%       109   1480   7761      29
25 - (            CMS)     20       4       1    0.5%    0.5%         3      0   2695       1
26 - (        VTXCTRL)      5       3       0    0.5%    0.0%         0      0    678       1
27 - (        CAMCTRL)      5       3       1    0.5%    0.5%         0      0    678       0
29 - (    ADCINTERNAL)      1       5       2    0.5%    0.5%         0      0    136       2
31 - (SPEED_NEGOTIATION)     99       6       1    0.5%    0.5%         7      0  13188       0
RX Check Function                   5       1                         6
Total (excluding SERIAL)                       146.3%   73.5%
```
```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 5
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
```
# diff

# version
# Betaflight / STM32F745 (S745) 4.3.0 Oct  9 2021 / 00:34:00 (38ca006) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF7HDV, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7HDV
manufacturer_id HBRO

# dma
dma ADC 1 NONE

# feature
feature GPS

# serial
serial 1 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# aux
aux 0 0 0 900 1300 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 3 1900 2100 0 0
aux 4 13 2 1900 2100 0 0
aux 5 35 2 1300 1700 0 0

# master
set acc_calibration = 82,115,-205,1
set rssi_channel = 15
set serialrx_provider = CRSF
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 71
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 70
set gps_rescue_descent_dist = 100
set gps_rescue_ground_speed = 1800
set gps_rescue_throttle_max = 1700
set gps_rescue_ascend_rate = 300
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = FIXED_ALT
set deadband = 2
set yaw_deadband = 2
set osd_warn_batt_not_full = OFF
set osd_cap_alarm = 1300
set osd_vbat_pos = 484
set osd_rssi_pos = 2491
set osd_link_quality_pos = 58
set osd_tim_2_pos = 166
set osd_flymode_pos = 0
set osd_anti_gravity_pos = 385
set osd_current_pos = 2539
set osd_mah_drawn_pos = 2543
set osd_craft_name_pos = 32
set osd_gps_speed_pos = 2112
set osd_gps_lon_pos = 2063
set osd_gps_lat_pos = 2053
set osd_gps_sats_pos = 2048
set osd_home_dir_pos = 6274
set osd_home_dist_pos = 2144
set osd_flight_dist_pos = 46
set osd_altitude_pos = 2080
set osd_pid_yaw_pos = 392
set osd_warnings_pos = 14721
set osd_avg_cell_voltage_pos = 2546
set osd_flip_arrow_pos = 412
set osd_stat_total_time = ON
set debug_mode = GYRO_SCALED
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end
```

SteveCEvans, 2021-10-09:
I see from your `dma show` dump that `SPI_MISO 4` is missing. Please enter the following:

```
dma SPI_MOSI 1 1
# dma SPI_MOSI 1: changed from NONE to 1
dma SPI_MISO 1 1
# dma SPI_MISO 1: changed from NONE to 1
save
```

The `status` command should then report `Gyros detected: gyro 1 locked dma` at which point I believe this issue should be resolved.

maxbl4, 2021-10-09:
Did that, status shows gyro 1 locked dma, now CPU spikes became much less. But still spikes to 50% at times

SteveCEvans, 2021-10-09:
Please set `debug_mode=TIMING_ACCURACY`.

```
// DEBUG_TIMING_ACCURACY, requires USE_LATE_TASK_STATISTICS to be defined
// 0 - % CPU busy
// 1 - Tasks late in last second
// 2 - Total lateness in last second in us
```

On the IFLIGHT_F745_AIO I see the following on the sensors tab. CPU usage sitting just above 45%. Looks to be what you see too? Zero tasks late per second means we have achieved perfect timing.

[image]

What lateness there is can be attributes to being connected to USB which is far from deterministic.

maxbl4, 2021-10-15:
This one is with battery connected, all config setup, GPS and air unit are running
[image]

SteveCEvans, 2021-10-15:
Looks good. Thanks.

closed 2021-11-22 by github-actions[bot]