betaflight/betaflight #10995: [4.2.10] Flycolor X-Tower stack, motors 1 and 2 not running

0crap, 2021-10-01:
I flashed v4.2.10 to my Flycolor X-Tower stack and motors 1 and 2 are not running.
http://www.flycolor.net/index.php?c=category&id=197
Tested with the radio and with the motor sliders in the Configurator.

Flashed with full erase and accepted custom config.
Second test I only configured DSHOT300 and used motor sliders, same result. 

On v4.2.9 all works fine.

```# version
# Betaflight / STM32F405 (S405) 4.2.10 Sep 30 2021 / 11:29:38 (e4b212699) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z```

haslinghuis, 2021-10-01:
Please attach full `diff` (not dump) and a `resource show all`.

0crap, 2021-10-02:
Sure.

4.2.9
```
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.9 Apr 27 2021 / 19:33:01 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4SD
manufacturer_id AIRB
mcu_id 00<snip>
signature 

# master
set acc_calibration = 13,89,-167,1
set motor_pwm_protocol = DSHOT300
set pid_process_denom = 4
set gyro_1_align_yaw = 2700

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 3 C09
resource INVERTER 6 C08
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource BARO_CS 1 B03
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 B07
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05
```

4.2.10
```
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.10 Sep 30 2021 / 11:29:38 (e4b212699) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4SD
manufacturer_id AIRB
mcu_id 00<snip>
signature 

# master
set acc_calibration = 11,90,-170,1
set motor_pwm_protocol = DSHOT300
set pid_process_denom = 4
set gyro_1_align_yaw = 2700

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 3 C09
resource INVERTER 6 C08
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource BARO_CS 1 B03
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 B07
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05
```

No difference that I can see.
Tested both with the motor sliders in the Configurator GUI.

If I try DSHOT600 it works on both!

0crap, 2021-10-02:
A resource show all on both version match if I do a compare on the output.

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: PREINIT
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

mikeller, 2021-10-31:
@0crap: Can I get you to test this firmware please (from https://github.com/betaflight/betaflight/pull/11053):

[betaflight_4.2.11_STM32F405_norevision.zip](https://github.com/betaflight/betaflight/files/7448962/betaflight_4.2.11_STM32F405_norevision.zip)

0crap, 2021-11-01:
> @0crap: Can I get you to test this firmware please (from #11053):
> 
> [betaflight_4.2.11_STM32F405_norevision.zip](https://github.com/betaflight/betaflight/files/7448962/betaflight_4.2.11_STM32F405_norevision.zip)

Bench tested and passes with flying colors. :-)
Thx!

PR #11053 (2021-11-03): Fixed burst Dshot.
Fixes #10995.
Fixes #11002.
Fixes #11036.

**Update 2 Nov 2021:** There was another fix, I've updated the targets below, please re-test.

Test targets:

[betaflight_4.3.0_STM32H743_60e035be0.zip](https://github.com/betaflight/betaflight/files/7455762/betaflight_4.3.0_STM32H743_60e035be0.zip)
[betaflight_4.3.0_STM32G47X_60e035be0.zip](https://github.com/betaflight/betaflight/files/7455763/betaflight_4.3.0_STM32G47X_60e035be0.zip)
[betaflight_4.3.0_STM32F745_60e035be0.zip](https://github.com/betaflight/betaflight/files/7455764/betaflight_4.3.0_STM32F745_60e035be0.zip)
[betafli [... cut]
closed 2021-11-03 by mikeller