betaflight/betaflight #11002: motor 4 of SpeedyBeeF7V2 did not work on BF4.2.10

azolyoung, 2021-10-06:
### Describe the bug
motor 4 of SpeedyBeeF7V2 did not work on BF4.2.10, and If I flash the FC back to the 4.2.9, all the motors are work fine.

### To Reproduce
1. Flash Betaflight 4.2.10 firmware in BetaFlight Configurator 10.8.0
2. Change ESC Protocol to DSHOT600 at Motors tab, then Save
3. Turn on motor test switch, and drag the Master slider to about 1250

### Expected behavior
1. All the motors spinning, and the speed of them is same.

### Flight controller configuration

```
# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.10 Sep 30 2021 / 11:31:52 (e4b212699) MSP API: 1.43
# config: manufacturer_id: SPBE, board_name: SPEEDYBEEF7V2, version: 55694498, date: 2021-05-15T13:56:51Z

# start the command batch
batch start

board_name SPEEDYBEEF7V2
manufacturer_id SPBE

# master
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 1
A09: MOTOR 2
A10: MOTOR 3
A11: USB
A12: USB
A13: SWD
A14: FLASH_CS
A15: PINIO 1
B00: LED_STRIP
B01: FREE
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
TIM2: FREE
TIM3:
    CH4: MOTOR 4
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: LED_STRIP
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: SpeedyBee F7 V2
 
 - Other components: SpeedyBee BLHeli32 45A 4 in 1 ESC
 
 - How are the different components wired up:
JST 8 Pin  cable

I have also connect to BLHeliSuite32xm 64 1.0.3.1 on Mac OS, and read the ESC info, it show a `Unknown` at the Input protocol of No.4 ESC, following is the screenshot:
[image]

SteveCEvans, 2021-10-06:
I have a SPEEDYBEEF7V2 on the bench and can investigate this.

peshaloo, 2021-10-07:
I have same issue with SPEEDYBEEF7V2 motor 4 not working on BF4.2.10

azolyoung, 2021-10-07:
@0crap Yes, seems we met same issue.

I have did some test\changes, and found if I change the timer of Motor4, then the Motor4 spin normally. using the following command in the CLI:
```
timer C09 AF3
timer B00 AF2
save
```

but just test on the desk, have not test in the air. I'll test it in comming 1 or 2 days.

0crap, 2021-10-07:
> @0crap Yes, seems we met same issue.
> 
> I have did some test\changes, and found if I change the timer of Motor4, then the Motor4 spin normally. using the following commands in the CLI:
> 
> ```
> timer C09 AF3
> timer B00 AF2
> save
> ```
> 
> but just test on the desk, have not test in the air. I'll test it in comming 1 or 2 days.

Well, thanks for sharing your findings, however it should not be needed to fiddle with timers on the CLI.
Never been needed.
I hope the devs can find/fix why some boards see this condition.

I noticed switching from DSHOT300 to 600 fixes the issue.
To me it looks like some timing issue is messing things up on 4.2.10 
¯\_(ツ)_/¯

SteveCEvans, 2021-10-09:
It was, please see the `diff all` above.

SteveCEvans, 2021-10-09:
I see the issue with the following values.

```
dshot_burst = ON
Allowed values: OFF, ON, AUTO

dshot_bidir = OFF
Allowed values: OFF, ON

dshot_bitbang = AUTO
Allowed values: OFF, ON, AUTO

dshot_bitbang_timer = AUTO
Allowed values: AUTO, TIM1, TIM8
```

Setting `dshot_burst = OFF` fixes it.

For the purposes of note taking MOTOR 4 is on a different timer. I'll check to see what relevant changes were made between 4.2.9 and 4.2.10.

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
TIM2: FREE
TIM3:
    CH4: MOTOR 4
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: LED_STRIP
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

0crap, 2021-10-09:
Thx, for the similar issue I reported I already narrowed it down.

> @azolyoung Can you test using Jenkins? For me build 82 is still OK, _[build 83](https://ci.betaflight.tech/job/Betaflight%20Maintenance%204.2/83/)_ breaks it. Thx.

SteveCEvans, 2021-10-09:
@mikeller Issue is with b6723d1f.

Output from git bisect:

```
b6723d1fe2b2ab304962f729f904d00fc49816c1 is the first bad commit
commit b6723d1fe2b2ab304962f729f904d00fc49816c1
Author: Michael Keller <github@ike.ch>
Date:   Wed Sep 29 01:04:53 2021 +1300

    Merge pull request #10895 from mikeller/add_dma_resource_allocation

:040000 040000 fc779d50812d86c5454fc38ad585705c8eddad16 dcccb504d2f5dd6963e855aed689c0025e7f39ad M	make
:040000 040000 260bfcc578d2859c97f3a325b9c6b89afb1bfb3a 6e69b8273b2962fba06191cce91d69a40207c28e M	src
```

I have repeatedly confirmed that the prior commit 4b88c37 is good.

SteveCEvans, 2021-10-09:
It's also worth noting that this issue also impacts current master (42f2fc2)

mikeller, 2021-10-31:
@SteveCEvans: Can I get you to please test with https://github.com/betaflight/betaflight/pull/11053?

SteveCEvans, 2021-11-02:
Tested a **SPEEDBYBEEF7V2** running 4.2.10 with a **HolybroTekko32 F3 Metal 65A 4in1 ESC** running 32.7 again with defaults applied and then DSHOT600 enabled. BLHeli32 configurator fails to talk to motor 4. With this fix applies that now works fine.

PR #11053 (2021-11-03): Fixed burst Dshot.
Fixes #10995.
Fixes #11002.
Fixes #11036.

**Update 2 Nov 2021:** There was another fix, I've updated the targets below, please re-test.

Test targets:

[betaflight_4.3.0_STM32H743_60e035be0.zip](https://github.com/betaflight/betaflight/files/7455762/betaflight_4.3.0_STM32H743_60e035be0.zip)
[betaflight_4.3.0_STM32G47X_60e035be0.zip](https://github.com/betaflight/betaflight/files/7455763/betaflight_4.3.0_STM32G47X_60e035be0.zip)
[betaflight_4.3.0_STM32F745_60e035be0.zip](https://github.com/betaflight/betaflight/files/7455764/betaflight_4.3.0_STM32F745_60e035be0.zip)
[betafli [... cut]
closed 2021-11-03 by mikeller