betaflight/betaflight #11015: Issues Flashing Eachine Wizard x220

JustinDoubs, 2021-10-10:
when flashing i get the error: 
STM32 - timed out, programming: FAILED
[image]
Quad is in boot mode and have tried with manual and automatic baud rate 
[image]
beta-flight version 10.7.0

SteveCEvans, 2021-10-10:
You're clearly not in boot mode with a good USB connection. If you were you'd see the serial port reflect that this was DFU mode, and also the "Exit DFU Mode" button would be enabled. When in DFU mode every FC looks the same. This is not specific to your FC. Looks like a connector or driver issue. Try a different FC.

JustinDoubs, 2021-10-11:
i noticed that when i connect the device (not in boot mode) it give some information and i believe that the FC is already running the latest firmware, (sorry i have just bought the quad and wanting to make sure it is fully updated), would that stop the firmware from being able to updated or is there still a remaining error?

[image]

ggmaniack, 2021-11-09:
@JustinDoubs I would recommend using the STM Flasher Demonstrator application for flashing this FC. That's the only reliable way of doing it that I've ever seen. Make sure you understand how to put it in bootloader mode. 
There is a guide on youtube called "FIX YOUR BRICKED SPRACING F3 BOARDS-Programming failed" which shows how to do it. You can generally follow it, just make sure to skip over the old firmware stuff and download new files on your own.
Otherwise it's as others have said, ancient components are a pain to deal with.

@SteveCEvans this is not a modern FC that flashes over USB DFU. The USB doesn't go directly to the STM chip, instead it goes into a USB-Serial converter, which is then hooked up to UART1. It will always show up as a COM port.

closed 2021-12-17 by github-actions[bot]