betaflight/betaflight #11036: MAMBAF722_I2C on 4.2.10 have a wrong motors behaviour

DusKing1, 2021-10-22:
### Describe the bug
Motor 3 slot will spin both motor3 and 4, whereas Motor 4 slot will only dithering the motor 4.

### To Reproduce
Flash 4.2.10 on MAMBAF722_I2C can reproduce this

### Expected behavior
Well you know, Motor 3 slot spins motor 3, Motor 4 slot spins motor 4.

### Flight controller configuration

```
# version

# Betaflight / STM32F7X2 (S7X2) 4.2.10 Sep 30 2021 / 11:31:52 (e4b212699) MSP API: 1.43

# config: manufacturer_id: DIAT, board_name: MAMBAF722_I2C, version: ec268d11, date: 2021-07-26T13:10:45Z

# start the command batch

batch start

# reset configuration to default settings

defaults nosave

board_name MAMBAF722_I2C

manufacturer_id DIAT

mcu_id 003f00173456500f2030324b

signature 

# master

set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

# restore original profile selection

profile 0

```
```
# dma show

Currently active DMA:

--------------------

DMA1 Stream 0: FREE

DMA1 Stream 1: FREE

DMA1 Stream 2: FREE

DMA1 Stream 3: FREE

DMA1 Stream 4: FREE

DMA1 Stream 5: FREE

DMA1 Stream 6: FREE

DMA1 Stream 7: FREE

DMA2 Stream 0: ADC

DMA2 Stream 1: TIMUP 8

DMA2 Stream 2: FREE

DMA2 Stream 3: FREE

DMA2 Stream 4: FREE

DMA2 Stream 5: TIMUP 1

DMA2 Stream 6: FREE

DMA2 Stream 7: FREE

```

### Short video to demonstrate

https://drive.google.com/file/d/1xg0vncu-8q7czGwpnHNuRYlnHFmXLVFF/view?usp=drive_web

filipslavik, 2021-10-30:
I can confirm the same issue on "Mamba-F722-Mini" with MAMBAF722_I2C on 4.2.10. I downgraded to 4.2.9 and everything works as expected.

DusKing1, 2021-11-01:
@Mode-1 @filipslavik there might be a fix for our problem, try firmware in the PR if you have time https://github.com/betaflight/betaflight/pull/11053

Mode-1, 2021-11-01:
I tried betaflight_4.2.11_STM32F7X2_5d3941edd.hex (allthough without full chip erase to keep my settings) from #11053
and still same faulty behavior. Went back to 4.2.9. (same without full chip erase) and voila ok again.

mikeller, 2021-11-01:
Thanks for testing @Mode-1. Looks like there's a different fix needed for this board - I'll investigate.

DusKing1, 2021-11-01:
@mikeller Thanks for taking care of this!

mikeller, 2021-11-01:
@Mode-1: Try now please with the new targets in #11053.

Mode-1, 2021-11-01:
@mikeller
"betaflight_4.2.11_STM32F7X2_6cb4612f0.hex" of #11053 works for me.
Flashed without full chip erase and did motor tests which worked normally.
Tried a short session of LOS in the garden, just 10m of range, happens to be OK.
I'm short before leaving into holidays (saturday), my question @mikeller:
Do you think 4.2.11 is ok to keep, or shall i return to 4.2.9 until 4.2.11 is released ?
Thanks for solving this issue so fast btw :-)

mikeller, 2021-11-01:
@Mode-1: 4.2.11 is 4.2.10 with a small number of bug fixes, so it should be fine.

PR #11053 (2021-11-03): Fixed burst Dshot.
Fixes #10995.
Fixes #11002.
Fixes #11036.

**Update 2 Nov 2021:** There was another fix, I've updated the targets below, please re-test.

Test targets:

[betaflight_4.3.0_STM32H743_60e035be0.zip](https://github.com/betaflight/betaflight/files/7455762/betaflight_4.3.0_STM32H743_60e035be0.zip)
[betaflight_4.3.0_STM32G47X_60e035be0.zip](https://github.com/betaflight/betaflight/files/7455763/betaflight_4.3.0_STM32G47X_60e035be0.zip)
[betaflight_4.3.0_STM32F745_60e035be0.zip](https://github.com/betaflight/betaflight/files/7455764/betaflight_4.3.0_STM32F745_60e035be0.zip)
[betafli [... cut]
closed 2021-11-03 by mikeller