betaflight/betaflight #11058: Mamba F722 flight controller DMA conflict

Michael-Gerlach, 2021-11-02:
### Describe the bug
I experienced DMA conflicts for the motor output. Motor 3 and 4 were assigned to the same DMA stream and channel:

dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0

This obviously led to problems with the flight control and resulted in a flyaway on the maiden. 

### To Reproduce
I tried various firmwares and applied the target specific custom defaults and the same DMA conflict occured every time. 

### Expected behavior
Not having to manually check DMAs and reassign them.

### Flight controller configuration

```
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.1 Jul 19 2020 / 06:19:20 (caa0d683c) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: ec268d11, date: 2021-07-26T13:10:45Z

# start the command batch
batch start

board_name MAMBAF722
manufacturer_id DIAT

# dma
dma pin A09 1
# pin A09: DMA2 Stream 2 Channel 6

# feature
feature GPS

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 0 19200 57600 0 115200
serial 4 2 115200 57600 0 115200

# aux
aux 0 0 1 1300 1700 0 0
aux 1 1 2 1300 2100 0 0
aux 2 13 0 1700 2100 0 0
aux 3 35 4 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set acc_calibration = -49,-19,103,1
set serialrx_provider = CRSF
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set osd_vbat_pos = 352
set osd_rssi_pos = 2072
set osd_link_quality_pos = 2103
set osd_throttle_pos = 2336
set osd_ah_pos = 2286
set osd_current_pos = 2368
set osd_gps_speed_pos = 2423
set osd_gps_lon_pos = 2449
set osd_gps_lat_pos = 2432
set osd_gps_sats_pos = 2445
set osd_home_dir_pos = 2327
set osd_home_dist_pos = 2328
set osd_flight_dist_pos = 2359
set osd_compass_bar_pos = 2090
set osd_altitude_pos = 2391
set osd_power_pos = 2400
set osd_warnings_pos = 14473
set osd_avg_cell_voltage_pos = 2304
set osd_pit_ang_pos = 33
set osd_rol_ang_pos = 65
set osd_battery_usage_pos = 10
set osd_disarmed_pos = 2250
set osd_nvario_pos = 375
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5725
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# rateprofile 0
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: CAMERA_CONTROL
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

Please note, that this is after I manually set: 'dma pin A09 1'

### Setup / Versions
 
 - Flight controller: Mamba F722
 
 - Other components: ESC F60 Pro, VTX: TBS unify Pro, RX: Crossfire nano diversity
 
 - How are the different components wired up:

This is a intofpv.com thread where the issue and the troubleshooting process are documented more thorougly: https://intofpv.com/t-flyaway

Best

JonnyPhenomenon, 2021-11-05:
to be clear, this is affecting all boards using the the DIAT-MAMBAF722* unified targets.  
The files in https://github.com/betaflight/unified-targets/tree/master/configs/default 
DIAT-MAMBAF722.config
DIAT-MAMBAF722_I2C.config
DIAT-MAMBAF722_X8.config 
- all have the same error. 
I believe the line "dma pin A09 0" needs to be changed to "dma pin A09 1".

DusKing1, 2021-11-06:
@JonnyPhenomenon @Michael-Gerlach Hey can you try the firmware in the PR? https://github.com/betaflight/betaflight/pull/11053 We have met the same problem and the firmware in this PR fixed my issue.

closed 2021-12-14 by github-actions[bot]