betaflight/betaflight #11063: HGLRCF722 GPS unstable

AndreaCCIE, 2021-11-09:
### Describe the bug
After having upgraded BF 4.3 from c23dd47f4 (2 Sept) to 5d41f9830 (4 Nov), the GPS is really unstable: it gets all the required satellites and after a few minutes (random value from 1 to x) all the GPS details on the OSD disappear. I get also a telemetry alert for GPS loss. After a while, without a power cycle, the GPS details come back live without having to search for the satellites again.
I have checked all the cables, no issue there.

### To Reproduce
Install BF 4.3 5d41f9830 (4 Nov)

### Expected behavior
GPS has to be consistent over time

### Flight controller configuration
```
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Nov  4 2021 / 19:20:36 (5d41f9830) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: 836acaf9, date: 2021-10-31T19:48:38Z

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR

# name: Rekon 5

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature OSD

# serial
serial 1 2 115200 115200 0 115200
serial 3 1 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# led
led 0 4,7::C:4
led 1 5,7::C:4
led 2 6,7::C:4
led 3 7,7::C:4
led 4 8,7::C:4
led 5 9,7::C:4
led 6 10,7::C:4
led 7 11,7::C:4

# aux
aux 0 0 0 1300 1675 0 0
aux 1 1 3 900 1300 0 0
aux 2 46 4 1700 2100 0 0
aux 3 13 3 1700 2100 0 0
aux 4 26 1 1300 1700 0 0
aux 5 31 1 1700 2100 0 0
aux 6 36 5 1700 2100 0 0

# adjrange
adjrange 0 0 2 900 1200 29 2 0 0
adjrange 1 0 2 1350 1650 29 2 0 0
adjrange 2 0 2 1800 2100 29 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 36
vtxtable powerlabels 25 100 400 MAX

# master
set gyro_lpf1_static_hz = 800
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = 0,36,60,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_idle_value = 800
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set failsafe_delay = 3
set failsafe_procedure = GPS-RESCUE
set failsafe_stick_threshold = 15
set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 290
set vbat_scale = 111
set ibata_scale = 609
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 40
set gps_rescue_descent_dist = 150
set gps_rescue_ground_speed = 1000
set gps_rescue_ascend_rate = 300
set gps_rescue_throttle_hover = 1300
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set deadband = 5
set yaw_deadband = 5
set thrust_linear = 20
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set osd_warn_launch_control = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_cap_alarm = 2500
set osd_tim1 = 2049
set osd_tim2 = 1
set osd_vbat_pos = 97
set osd_rssi_pos = 420
set osd_link_quality_pos = 14391
set osd_link_tx_power_pos = 14433
set osd_tim_1_pos = 14369
set osd_tim_2_pos = 215
set osd_remaining_time_estimate_pos = 193
set osd_flymode_pos = 432
set osd_throttle_pos = 12375
set osd_vtx_channel_pos = 84
set osd_current_pos = 8577
set osd_mah_drawn_pos = 12705
set osd_craft_name_pos = 364
set osd_gps_speed_pos = 14550
set osd_gps_lon_pos = 8623
set osd_gps_lat_pos = 8591
set osd_gps_sats_pos = 14785
set osd_home_dir_pos = 14414
set osd_home_dist_pos = 14799
set osd_flight_dist_pos = 12758
set osd_compass_bar_pos = 14378
set osd_altitude_pos = 14518
set osd_pid_roll_pos = 2407
set osd_pid_pitch_pos = 2439
set osd_pid_yaw_pos = 423
set osd_warnings_pos = 14505
set osd_avg_cell_voltage_pos = 14401
set osd_disarmed_pos = 12523
set osd_esc_tmp_pos = 428
set osd_log_status_pos = 440
set osd_rate_profile_name_pos = 151
set osd_pid_profile_name_pos = 2401
set osd_efficiency_pos = 12742
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_flight_dist = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_freq = 5917
set name = Rekon 5

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_expo = 0
set dterm_lpf1_type = PT3
set dterm_lpf1_static_hz = 100
set dterm_lpf2_static_hz = 0
set throttle_boost = 2
set throttle_boost_cutoff = 10
set feedforward_transition = 80
set feedforward_averaging = 4_POINT
set dyn_idle_min_rpm = 26
set simplified_dterm_filter = OFF

rateprofile 1

# rateprofile 1
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 40
set tpa_breakpoint = 1550

# end the command batch
batch end

# 
```

```
Entering CLI Mode,

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 - Flight controller: 
HGLRCF722
 - GPS:
BN180

WinniNhm, 2021-11-16:
Hi @AndreaCCIE ,

I don’t think that the GPS code got recently changed (just fixing a typo, see [https://github.com/betaflight/betaflight/pull/10904](url)

Looking into your config, I would recommend to change 

`set gps_sbas_mode = none` 

in the first place since sbas (“ground assistance” in the configurator) can cause dropouts. 
You can read some background info here:

[https://github.com/betaflight/betaflight/issues/9194](url)
[https://github.com/betaflight/betaflight/issues/9194#issuecomment-578496079](url)

As a consequence of this trouble shouting the sbas mode “none” was introduced in BF 4.2 and became the default setting.
 
Next I would switch off `gps_auto_baud` and set the baud rate manually.
“Galileo” can safely switched off since it brings no value.

Finally these modules are sensitive to electrical and RF noise (some more, some less). 
May be your tune got a bit more aggressive, there is a bit more electrical noise in the system, and your module drops out…

WinniNhm, 2021-11-16:
Hi @AndreaCCIE ,

If GPS disappears then it means you are losing communication with your module when arming.
I would try to identify the build number by tail and error when it start failing.
Builds 2527 and 2544 would be candidates for me.

AndreaCCIE, 2021-11-20:
Hi @WinniNhm,
sorry but the test is taking a bit, a lot of firmware to test, and the issue doesn't always appear as soon as I turn on the engine (but normally within the first few minutes).
So far, I can see 2527 seems ok: I flew it 10 minutes without any problem.
On 2541, instead, GPS failed in a minute.

AndreaCCIE, 2021-11-20:
Hi @WinniNhm,
I had the same issue with 2526: I'm now downgrating starting from that one.
Am I the only crazy guy testing the 4.3 on a long range build? :-)

WinniNhm, 2021-11-20:
Hi @AndreaCCIE ,

Check this issue:
[https://github.com/betaflight/betaflight/issues/11010](url)

WinniNhm, 2021-11-21:
This is the correct link, @AndreaCCIE 
https://github.com/betaflight/betaflight/issues/11010
It seems you are not alone.

AndreaCCIE, 2021-11-22:
@WinniNhm this is getting too technical to me, I've done my best. I will wait for the next nightly release ( if any :-))

AndreaCCIE, 2021-12-18:
Not sure if this issue was taken into account, but yesterday's night build was still showing the problem.
Please find a video below of a test flight. I rolled back to the version dated 2nf of Sept, and everything was fine.

https://youtu.be/zcjRFJuCQTE

WinniNhm, 2021-12-19:
I can confirm this on Matek H743.
GPS works initially, gets 3D lock but fails after approx. 60 seconds.
It start failing after #10813 got merged.
@SteveCEvans , please kindly check.

PSi86, 2021-12-19:
Something is off in this discussion. This issue was opened on 4th of November so the origin of the issue can hardly be a PR that was merged mid December 🙄. But in the initial log there is dshot bitbang activated - wasn't there an issue related to that?

AndreaCCIE, 2021-12-19:
Agree with @PSi86, I had this problem for more than a month now.
Today I tested another quad, but this time with Mamba 722 FC to prove (even if it was clear to me) it wasn't related to faulty hardware. I had the same GPS problem.

SteveCEvans, 2021-12-19:
I'm suspicious that the issue only occurs when the quad is powered. I've been flying this code for a long time, most recently yesterday, and once I get a GPS lock it is maintained. This was with a JHEF7DUAL which has an F722.

SteveCEvans, 2021-12-19:
I've been using bit-banged DSHOT. Have you tried starting the motors with any other protocol such as OneShot125? That would help eliminate the DSHOT code as being implicated.

AndreaCCIE, 2021-12-19:
I can test it on Wednesday if no one can do it earlier.

AndreaCCIE, 2021-12-19:
@SteveCEvans I can test it on Wednesday in no one can do it earlier. However, I have also lost the GPS a couple of times while playing with the menu in the osd.

WinniNhm, 2021-12-19:
@AndreaCCIE see PR #11135

AndreaCCIE, 2021-12-19:
> @AndreaCCIE see PR #11135

Thanks @WinniNhm I will wait until it gets merged (I know how to test beta but not how to build specific releases myself)

limonspb, 2021-12-20:
@AndreaCCIE you don't have to wait to test actually.
You can download these hex files from that PR and **flash as unified tarted**:
https://artprodeau1.artifacts.visualstudio.com/A3a9037bc-5464-4b2e-9e8d-ed1a5aa29705/ce1d2eeb-ce65-4b95-a02a-7fee6c006c71/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL0JldGFmbGlnaHQvcHJvamVjdElkL2NlMWQyZWViLWNlNjUtNGI5NS1hMDJhLTdmZWU2YzAwNmM3MS9idWlsZElkLzUwMTMvYXJ0aWZhY3ROYW1lL2JldGFmbGlnaHQ1/content?format=zip

Could you please check?
Thanks!

AndreaCCIE, 2021-12-20:
9 minutes hovering in the backyard, and the GPS hasn't lost connectivity. Not sure if this "static test" is relevant, but it's the best I can do for today. I will go down to the field for a couple of packs on Wednesday.

closed 2021-12-21 by haslinghuis

limonspb, 2021-12-22:
> 9 minutes hovering in the backyard, and the GPS hasn't lost connectivity. Not sure if this "static test" is relevant, but it's the best I can do for today. I will go down to the field for a couple of packs on Wednesday.

thanks for reporting and thanks for testing!