betaflight/betaflight #11064: Start/Stop recording on bluetooth enabled camera (GoPro, Insta360 Go2...) when arming/disarming

ghouet, 2021-11-13:
### Is your feature request related to a problem? Please describe.
There use to be a function where you could automatically start/stop recording of your onboard cam when configuring your camera with the Runcam Camera protocol. While we don't have wired interface like that anymore, it looks like most of the action camera that are used nowadays have bluetooth. For FCs that have an integrated bluetooth chip, we could use it to interface with the camera (and even display its settings in the goggles.)

### Describe the solution you'd like
Each camera has its own protocol, GoPro, Insta360 Go2, etc.  It might even differ from Hero 6 to Hero 10, we would have to confirm.
The implementation should allow the user to configure what camera it should talk to, with which protocol.

To avoid RF interferences, the bluetooth could simply connect when arming, turn the camera on, then disconnect and do the opposite when disarming,

### Describe alternatives you've considered
Manually pushing the REC button on the camera... But it's 2021, we can do better than that.

etracer65, 2021-11-13:
I doubt this is possible with the existing hardware. The flight controllers that actually have bluetooth simply have standalone modules that connect via serial to a UART on the micro controller running Betaflight. They are simply wireless serial port extensions. Betaflight would have no way to access the actual bluetooth chipset or control it. Further the bluetooth chips on the flight controller are meant to be a slave device (something pairs to them, they can't pair to something else) and doubtful that they could initiate a connection to another device.

ghouet, 2021-11-13:
Interesting, so we would need an additional board like this at first: https://www.digikey.ca/catalog/en/partgroup/bluefruit-le-uart-friend-bluetooth-low-energy-ble/66179.
Would it be possible then?

closed 2021-12-21 by github-actions[bot]