betaflight/betaflight #11066: OSD VTx Power Incorrect Reading (25mW)

dezignhero, 2021-11-14:
### Describe the bug
On the VTx OSD element I can read band/channel fine but the power stays at 25mW on Smartaudio for Build 4.3.0#2549.  When I go to the actual features menu in the OSD, the power is correct at 200mW and saves what I set.  Exiting out, regardless of power level, always shows 25mW in OSD when it should be 200mW.  This is on HGLRCF722.  

### To Reproduce
Using a screen or google for feedback.  On HGLRCF722 board (may work on others), set the OSD to display the VTx element.  Connect a smartaudio device and set the power to any value that's not 25mw, (let's say 100mw).  Save your changes, and exit back out to the OSD and you'll see the band and channel are correct but power is stuck at 25mw.

### Expected behavior
In the VTX OSD element which displays band/channel/power, the power value should be refletive of the saved value in the feature > Smartaudio > power value in the OSD.

### Flight controller configuration

```
# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 MAX
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5843
set vcd_video_system = NTSC
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: FREE
B01: MOTOR 1
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: LED 1
C14: BEEPER
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH2 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1
```

### Setup / Versions

 - Flight controller: HGLRCF722 30x30 Betaflight 4.3.0 #2549
 - Other components: TBS Unify Race
 - How are the different components wired up: Smartaudio cable running from VTx to Uart on FC.  Able to change band/channels no issues.

closed 2021-11-14 by github-actions[bot]

limonspb, 2021-11-15:
[image]

dezignhero, 2021-11-15:
Thanks!  I saw it in the Slack and it worked.  Didn't follow up on the ticket since it was auto-closed.