betaflight/betaflight #11071: Build 2549 - looptimes consistent, but wrong

packocrayons, 2021-11-16:
### Describe the bug

The looptimes (shown in the "tasks" output) are very consistent, however they are not 8k/4k/2k etc. They are consistently ~10% low (3953 (but config shows 250us) 4k, 127us on 8k) even when CPU usage is low (2% on 4k, 5% on 8k). 

### To Reproduce

Flash a holybro kakute v2 F745 with build 2549. All settings default.

### Expected behavior

looptimes should be exactly as set (they are in 4.2.11)

### Flight controller configuration

```
# version
# Betaflight / STM32F745 (S745) 4.3.0 Nov  9 2021 / 20:26:28 (6ecd0d232) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# master
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: Holybro Kakute F7 V2
 
 - Other components: Happymodel EP1 ELRS 1.2.0 (UART6), flywoo goku 506s (UART7) , eachine TX805 (UART 4)
 
 - How are the different components wired up: RX on UART6, ESC on UART, vtx on UART4. VTX using vbat (passed through FC), rx using 5V from FC

Disabling acc/baro do not do anything useful, cpu usage is reduced but looptimes stay exactly the same

SteveCEvans, 2021-11-17:
Previously the gyro task timing was controlled, and measured by, the processor's internal clock. Now, if gyro interrupts are detected, the scheduler counts every 25000 interrupts and measures the elapsed time. See https://github.com/betaflight/betaflight/blob/6ecd0d232fd388dd7ce09a2432a25e0f507284e3/src/main/scheduler/scheduler.c#L466. This is then used to determine the gyro loop time and locks the scheduler's gyro task loop to the gyro's sample rate. As the gyro's internal clock differs from the processor clock we see rates either side of 8kHz. If DMA is being used the value reported is typically very slightly over 8kHz, but when DMA is not being used a value slightly under 8kHz is reported. It may be that the time reading the data from the gyro slows it's internal refresh rate although this is not fully understood. Rest assured that the reported gyro loop rate is correct and reflects the native loop rate of the gyro itself.

packocrayons, 2021-11-18:
@SteveCEvans How does that affect filtering? My understanding was that RPM filtering was heavily dependant on a precise looptime.
This quad has severe flyaway issues on 4.3 that are not there on 4.2, I suspected due to this filtering issue.

ctzsnooze, 2021-12-11:
If it has flyaways, it's something else, not loop time.

packocrayons, 2021-12-11:
Something has changed because this build flies perfectly fine with 4.2 and has flyaway issues on 4.3 #2549

SteveCEvans, 2021-12-30:
Closing this issue as the loop times deviating from 8k nominal have been explained.

closed 2021-12-30 by SteveCEvans