betaflight/betaflight #11072: Matekf411 4.3 Build#2549 Motor Temps

DivingFalconFPV, 2021-11-17:
### Describe the bug
Motors are getting extremely scolding hot

### To Reproduce
Default tune hover or fly around

### Expected behavior
Default tune on 2" whoop to run motors cool. Motors in a short hover get extremely hot. I didnt try tuning to lower D since it can run higher PIDs on 4.2.11. 
On 4.2 had higher PIDs and less filtering it flew nice and cool. I even went back to 4.2.11 and matched 4.3 PID and filtering and 4.2.11 ran cool motor temps. 

### Flight controller configuration

```
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Nov  9 2021 / 20:25:08 (6ecd0d232) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF411, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411
manufacturer_id MTKS
mcu_id 003d000b3239510e34393435
signature 

# name: BlueJay

# feature
feature -TELEMETRY
feature -AIRMODE

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 36
vtxtable powerlabels 25 100 400 MAX

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -170,53,-248,1
set rssi_channel = 16
set dshot_bidir = ON
set motor_poles = 12
set small_angle = 180
set thrust_linear = 25
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_rssi_alarm = 50
set osd_link_quality_alarm = 35
set osd_cap_alarm = 450
set osd_vbat_pos = 6531
set osd_rssi_pos = 6198
set osd_link_quality_pos = 50
set osd_tim_2_pos = 2420
set osd_remaining_time_estimate_pos = 343
set osd_flymode_pos = 6166
set osd_throttle_pos = 6540
set osd_crosshairs_pos = 6413
set osd_current_pos = 2402
set osd_mah_drawn_pos = 2411
set osd_craft_name_pos = 6154
set osd_compass_bar_pos = 9
set osd_pid_roll_pos = 97
set osd_pid_pitch_pos = 65
set osd_pidrate_profile_pos = 2049
set osd_avg_cell_voltage_pos = 6548
set osd_esc_rpm_pos = 0
set osd_esc_rpm_freq_pos = 0
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_total_time = ON
set vtx_band = 2
set vtx_channel = 3
set vtx_power = 1
set vtx_freq = 5771
set gyro_1_align_yaw = 1800
set name = BlueJay

profile 0

# profile 0
set iterm_relax_cutoff = 10

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
Betafpv old f411 whoop board with matekf411 target
 - Flight controller: 
XM receiver, TBS pro32 VTX, Baby Ratel camera. HGLRC AEOLUS 1202.5 5600kV motors plugged in. 
 - Other components:
Betafpv 85x frame. 
 - How are the different components wired up:
all wired to FC. Motors plugged in.

packocrayons, 2021-11-19:
I'm seeing similar issues - see #11071

DivingFalconFPV, 2021-11-19:
> I'm seeing similar issues - see #11071

So you think turning off rpm filter would help?  
Not that there's a point it flies ok in 4.2.11 just was hoping 4.3 help it fly better

packocrayons, 2021-11-20:
I imagine it might, however then you'll have less filtering.

closed 2021-12-28 by github-actions[bot]