betaflight/betaflight #11073: SD Card Fatal Error - H7 Board on 4.2.11

vincentpoont2, 2021-11-18:
### Describe the bug
Shows "Fatal Error Reboot to Retry" on Blackbox Onboard SD Card on 4.2.11

### To Reproduce
Tested with using two KakuteH7 

- 4GB SanDisk EDGE (No issue)
- 8GB SanDisk EDGE (Fatal Error)
- 8GB Toshiba (Fatal Error)
- 32GB SanDisk (Fatal Error)
- 64GB SanDisk Extreme (Fatal Error)

[image]

### Flight controller configuration
Tested on 10.7.1 & 10.7.0 configurator.
Firmware 4.2.11
Target: HBRO/KAKUTEH7/(STM32H743)

Also Tested on BF 4.3, same Issue.

diff
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32H743 (SH74) 4.2.11 Nov  9 2021 / 20:29:57 (948ba6339) MSP API: 1.43
###ERROR: diff: NO CONFIG FOUND###
# start the command batch
batch start

board_name KAKUTEH7
manufacturer_id HBRO

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B03
resource MOTOR 4 B10
resource MOTOR 5 A00
resource MOTOR 6 A02
resource MOTOR 7 C08
resource MOTOR 8 C09
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 D00
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource LED 1 C02
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 4 E02
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 4 E05
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 4 E06
resource CAMERA_CONTROL 1 E09
resource ADC_BATT 1 C00
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C01
resource SDCARD_CS 1 A04
resource SDCARD_DETECT 1 A03
resource PINIO 1 E13
resource OSD_CS 1 B12
resource GYRO_EXTI 1 E01
resource GYRO_CS 1 E04
resource USB_DETECT 1 A08

# timer
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer D12 AF2
# pin D12: TIM4 CH1 (AF2)
timer E09 AF1
# pin E09: TIM1 CH1 (AF1)

# dma
dma SPI_TX 1 13
# SPI_TX 1: DMA2 Stream 5 Request 38
dma ADC 1 8
# ADC 1: DMA2 Stream 0 Request 9
dma ADC 3 9
# ADC 3: DMA2 Stream 1 Request 115
dma TIMUP 1 0
# TIMUP 1: DMA1 Stream 0 Request 15
dma TIMUP 2 0
# TIMUP 2: DMA1 Stream 0 Request 22
dma TIMUP 3 2
# TIMUP 3: DMA1 Stream 2 Request 27
dma TIMUP 4 0
# TIMUP 4: DMA1 Stream 0 Request 32
dma TIMUP 5 0
# TIMUP 5: DMA1 Stream 0 Request 59
dma TIMUP 8 1
# TIMUP 8: DMA1 Stream 1 Request 51
dma pin B00 0
# pin B00: DMA1 Stream 0 Request 25
dma pin B01 1
# pin B01: DMA1 Stream 1 Request 26
dma pin B03 2
# pin B03: DMA1 Stream 2 Request 19
dma pin B10 3
# pin B10: DMA1 Stream 3 Request 20
dma pin A00 4
# pin A00: DMA1 Stream 4 Request 55
dma pin A02 5
# pin A02: DMA1 Stream 5 Request 57
dma pin C08 6
# pin C08: DMA1 Stream 6 Request 49
dma pin C09 7
# pin C09: DMA1 Stream 7 Request 50
dma pin D12 14
# pin D12: DMA2 Stream 6 Request 29
dma pin E09 12
# pin E09: DMA2 Stream 4 Request 11

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature OSD

# serial
serial 0 1 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# master
set mag_bustype = I2C
set mag_i2c_device = 1
set baro_bustype = I2C
set baro_i2c_device = 1
set serialrx_provider = SBUS
set blackbox_device = SDCARD
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 109
set ibata_scale = 168
set beeper_inversion = ON
set beeper_od = OFF
set sdcard_detect_inverted = ON
set sdcard_mode = SPI
set sdcard_spi_bus = 1
set max7456_spi_bus = 2
set dashboard_i2c_bus = 1
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
set gyro_1_bustype = SPI
set gyro_1_spibus = 4
set gyro_1_sensor_align = CW270
set gyro_1_align_yaw = 2700
set gyro_2_bustype = NONE

profile 0

rateprofile 0

# end the command batch
batch end

resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SDCARD_DETECT
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: LED 1
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: CAMERA_CONTROL
E10: FREE
E11: FREE
E12: FREE
E13: PINIO 1
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: CAMERA_CONTROL
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 

### Setup / Versions
Holybro Kakute H7 
Target: HBRO/KAKUTEH7/(STM32H743)

vincentpoont2, 2021-11-18:
@mikeller are you familiar with this issue?

vincentpoont2, 2021-12-27:
Anyone has this issue on other H7 Board?

haslinghuis, 2022-01-06:
@vincentpoont2 See also https://github.com/betaflight/betaflight/issues/11093.

Can you please try with RC1? (This also requires configurator 10.8.0RC1).
RC2 is around the corner. You can already test some of it with https://github.com/betaflight/betaflight/pull/11198

SteveCEvans, 2022-01-06:
Works perfectly for me using a Sandisk Ultra 32G.
[image]

```
# version
# Betaflight / MATEKH743 (M743) 4.3.0 Jan  6 2022 / 20:01:37 (d492257) MSP API: 1.44

# status
MCU H743 (Rev.V) Clock=480MHz, Vref=3.30V, Core temp=47degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 3931, max available: 131072
Devices detected: SPI:2, I2C:1
Gyros detected: gyro 1, gyro 2 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=DPS310
OSD: MAX7456
System Uptime: 39 seconds, Current Time: 2022-01-06T20:06:34.974+00:00
CPU:19%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 5
SD card: Manufacturer 0x3, 31166976kB, 05/2005, v8.0, 'SC32G'
Filesystem: Ready
Arming disable flags: RXLOSS CLI MSP RPMFILTER
```

vincentpoont2, 2022-01-10:
> @vincentpoont2 See also #11093.
> 
> Can you please try with RC1? (This also requires configurator 10.8.0RC1). RC2 is around the corner. You can already test some of it with #11198

I tried 4.3RC1 with 10.8.0RC1 on the Holybro Kakute H7 and got similar result as before. It worked with some SD card and give error on some.
[image]
I

vincentpoont2, 2022-01-10:
> Works perfectly for me using a Sandisk Ultra 32G. [image]
>
> ```
> # version
> # Betaflight / MATEKH743 (M743) 4.3.0 Jan  6 2022 / 20:01:37 (d492257) MSP API: 1.44
>
> # status
> MCU H743 (Rev.V) Clock=480MHz, [...]

@SteveCEvans Would you have other cards to try? Any chance you can try it on the Holybro Kakute H7 board also? 
I am not sure what limitation this is.

vincentpoont2, 2022-02-07:
Any Idea on what can be causing some card to show fatal error?

haslinghuis, 2022-02-09:
Works with Holybro Kakute H743 on 4.3RC3 with 10.8.0RC3

[image]

haslinghuis, 2022-03-14:
Also recognized on H7FR on latest release

diogo-f, 2022-04-16:
I'm having this problem and tried messing with partition file systems and solve it for now I think. The board I'm using is the  HBRO/KAKUTEF7(STM32F745).

I had to format with this tool [MiniTool Partition Wizard Free](https://www.partitionwizard.com/free-partition-manager.html), somehow windows format tool settings breaks compatibility with betaflight. Might be the allocation size but I choose default on both tools so might be something else. I might check later with [GParted](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjT2efR8pj3AhVRXhoKHetfCAYQFnoECCMQAQ&url=https%3A%2F%2Fgparted.org%2F&usg=AOvVaw1XI7Wv48JPIEE_QkcMm6Sz) on linux what info it can get between windows formatted vs this minitool wizard formatted partition to compare.

I also tried to format the sd card as FAT16 with this tool and it also worked fine with betaflight.

Hope this helps!

closed 2022-04-17 by haslinghuis