betaflight/betaflight #11074: Display Motor Output Limit as OSD Element

TehllamaFPV, 2021-11-19:
This is a request in part to fulfill the goal of having 'flash this build and get spec racer from any setup', but as an interim capability to have motor output limit displayed in the OSD (for later DVR review) while retaining the Throttle% as scaled motor output.

Just enabling an OSD element that can be displayed would be what I'm after for this.

The ideal long-term answer is the more complex RPM Bidirectional DShot telemetry driven answer that can make performance more identical, but as a starting point to make 'run-what-you-brung' racing more attainable, this would help.

etracer65, 2021-11-19:
I don't really understand how having the motor limit percent displayed in the OSD helps make a "spec" racer. The type of motor (kv, stator size, etc), props, battery cells, even ESC settings all affect the amount of thrust generated between two quads even when set at the same motor output limit. So unless you're mandating identical hardware (a true "spec series") then I don't see how this could work. Or maybe you're not fully explaining the goals or purpose.

And trying to enforce compliance by mandating everybody run some specific "build" won't work. In 5 minutes I could create a custom cheating version that falsely displayed the motor output limit (with no limit actually enabled) but yet still appeared to be the "build" you mandated.

TehllamaFPV, 2021-11-19:
Great feedback - yes, I've definitely thought of that.

This in isolation wouldn't achieve that - it would be at best a stopgap to fill in for other easily inspectable items (props, battery), and at least provided motors are outputting something close to the nominal KV on the bell, this provides a quick sanity check that the selected motor output limit was actually applied (and that pilots aren't inadvertently switching profiles to one without a motor output limit, or with an incorrect one).
To make it like spec, it would still require minimum weights, matching propellers, and something that keeps maximum thrust at or near parity.
This is more intended to be a quick option to aid with post-tech inspected one for sportsman level racing where the actual performance isn't critical, but having the same realm of performance is key for making events fair enough to be fun, instead of requiring pilots to bring very specific hardware when the majority of event entrants aren't actually holding it wide open for long enough to make it actually relevant.  Hence, "run what you brung", this is mostly a check that the applied configuration limit was present (or not).

There are some other more implementation-heavy options that could achieve forced parity despite motors selections - there are some definite limitations with this, particularly when it comes to stator size and that somebody could either work some OSD/build magic to bypass, or rewind/rebell a motor to make it over-deliver with a given battery combination... but this is more to have an easy option where pilots can take a setup that is already working and tuned well, use a tool such as the TexasSpec calculator ( http://texasdronechamps.com/spec/ ), and apply that motor output and be close enough.
With top level racers where spending time at wide open throttle matters, running spec motors and spec batteries becomes necessary, that is correct.

closed 2021-12-28 by github-actions[bot]