betaflight/betaflight #11076: New target stm32F427IIH6

mathieu-justin, 2021-11-20:
### Is your feature request related to a problem? Please describe.
I am in a team in my university working on a drone project with a DJI flight controller that uses a stm32F427IIH6 MCU. The closest supported targets are the stm32F405 or stm32F411. Could it be possible to add this new target by developers or my team could attempt to add it with your help.

### Describe the solution you'd like
The solution that I would like is to either obtain assistance in creating the new target for the stm32F427IIH6.

### Describe alternatives you've considered
We have considered using tools such as stm32cubeMx or stm32cubeIDE to help with the debugging.

DusKing1, 2021-11-21:
What's the advantage of this chip? Easy to get? Cheap than others? Higher frequency? Good for future? You see we are not an academic organization, we do things for purposes.

mathieu-justin, 2021-11-21:
This chip is used in the RoboMaster Development Board type A and is a highly flexible controller board designed to be used in a wide range of robotics projects. It uses an STM32F427IIH6 as its main controller chip and features multiple extensions and communication interfaces. It features various ports, CAN port, UART port, variable-voltage PWM port, on board resources including IMU+E-compass, LED, etc. It is compatible with various devices such as servo motor, single axis gyro, ESC, PC, Bluetooth, and external Arduino devices. It has proven stability, power and reduced costs for an embedded system.

DusKing1, 2021-11-21:
@mathieu-justin this is more like words from an ad, we want you to point out explicitly the real advantage of it in clear terms.

Personally I have zero interest on those non-fpv DJI related product, especially for these education hype.

mathieu-justin, 2021-11-21:
Just to be clear, I wrote this post in hopes of getting advice on how developers make new targets and was not looking for betaflight to do the work for me. Therefore, I understand that you don't want to get involved in the details. However, I would appreciate it if you could point me to a good reference source for creating a new target/configuration.

DusKing1, 2021-11-21:
src/main/targets, src/main/drivers, and FYI from now one I'll put a little foucus on this education hype race and don't let me found anyone violated the GPL.

mathieu-justin, 2021-11-21:
Thanks, my team and I are going to read up on the GPL license to avoid violating it.

closed 2021-12-30 by github-actions[bot]