betaflight/betaflight #11077: Superior Filtering

rindoze, 2021-11-21:
### Is your feature request related to a problem? Please describe.

### Describe the solution you'd like

### Describe alternatives you've considered

daleckystepan, 2021-11-21:
Can you specify what you expect or what exactly we need to improve and how?

I don't understand how we can help you.

rindoze, 2021-11-21:
I'm saying the approach to filtering the gyro signal is inefficient. Basically what we want is inverse filtering. The path for data from gyro will change from:

- gyro_signal + filter(s) -> output_signal 

- gyro_signal + filter - gyro_signal -> output

We only need 1 filter that is targeting the frequency band we use (~sub 40hz) once we have that we can either amplify or attenuate it, then use simple math to get what we want.

KarateBrot, 2021-11-21:
@rindoze What exactly do you want to do? Sounds like you just want to have a single lowpass filter with a cutoff frequency of 40Hz?

rindoze, 2021-11-21:
That is basically what we are aiming for with the filtered gyro signal isn't it? I had heard that low hz lowpass added unacceptable delay.

KarateBrot, 2021-11-21:
Yeah, the phase delay for low cutoff frequencies is quite big.

rindoze, 2021-11-21:
This approach will also net complete destructive interference on all signals outside the filter band(s), instead of just highly attenuated.

KarateBrot, 2021-11-21:
How does the approach work? I don't see how you could filter data without introducing phase lag.

rindoze, 2021-11-21:
You could use a filter that has less delay/ is not affected by frequency like the low pass. So for example a second order notch filter targeting the low frequency we want.
We would in this case have:

-  Raw gyro signal input, saved to variable _gyro_
- Add notch filter described above to _gyro_ save as something else _gyroFiltered_
- Subtract _gyroFiltered from _gyro_ and pass it on

Result would be a signal with just the notch's band data with the delay of just a single notch filter

KarateBrot, 2021-11-21:
A notch filter has a phase shift of 180° at its center frequency. You cannot use it to filter low frequencies of you will have an unstable PID loop.

rindoze, 2021-11-21:
The 180 phase shift is what I am seeking. Just making sure we are on the same page, what we have now is:

gyro - filter1 -filter2 -filter3 -filter4..... = Something that gives pretty much the 0-40hz band

So a more flexible simple solution is to filter out what we want to keep, and subtract that from the main signal:

gyro - (gyro + filter) = filter

Filter in this case being designed to encompass the frequency band we want.

etracer65, 2021-11-21:
> So a more flexible simple solution is to filter out what we want to keep, and subtract that from the main signal:
>
> gyro - (gyro - filter) = filter
>
> Filter in this case being designed to encompass the frequency band we want.

The problem is that the filtered part that you "want to keep" has been phase shifted and delayed by the filtering. So you can't just subtract it from the original gyro signal.

rindoze, 2021-11-22:
I haven't seen how the gyro data stream looks, or how it is transformed currently, but these are all digital filters correct? In the digital realm its simple to just dump the original signal in a queue to be subtracted post filter. Phase shift should be predictable and as such easy to account for as well.

KarateBrot, 2021-11-22:
@rindoze phase shift is dependent on frequency. The signal won't get delayed as a whole. I think you imagine this to be way simpler than it actually is.

ctzsnooze, 2021-12-11:
A high-pass is just input-lowpass.  So your approach would be:
```
output = input - highpass
output = input - (input - lowpass), or
output = lowpass
```
The result of `input - highpass` would  be mathematically identical to `lowpass`.

closed 2021-12-11 by ctzsnooze

rindoze, 2021-12-17:
> A high-pass is just input-lowpass. So your approach would be:
> 
> ```
> output = input - highpass
> output = input - (input - lowpass), or
> output = lowpass
> ```
> 
> The result of `input - highpass` would be mathematically identical to `lowpass`.

This is the entire point, that there is now flexibility. Instead of being restricted to a low pass, you can use a high pass, or literally any filter. This allows the choosing of whatever filter gives the best performance. I am still confused as to why the current approach is dozens of notches and other filters on the entire spectrum, when we only care about a tiny frequency band.

KarateBrot, 2021-12-18:
> This is the entire point, that there is now flexibility. Instead of being restricted to a low pass, you can use a high pass, or literally any filter. This allows the choosing of whatever filter [...]

ctzsnooze showed you that there is no difference between using a lowpass or an inverted highpass. It's the same thing. Using a lot of notches gives us a very low filter latency. It's a lot more efficient than just filtering out everything above a threshold.