betaflight/betaflight #11079: Motor output limit on potentiometer

ATKCA, 2021-11-22:
### Is your feature request related to a problem? Please describe.

My suggestion is to have an option for motor output limit on a potentiometer so there is fine grain control during flight. The feature could include user configurable minimum maximum values for example 25% to 100%. 

### Describe the solution you'd like

Adding of the support for mapping of a potentiometer to motor output limit along with configurable range.
The addition of another bar on the modes screen where you can select and configure the pot.

### Describe alternatives you've considered

I have tried using a Taranis mixer mod to limit throttle but it does not limit motor output only throttle.

haslinghuis, 2021-11-23:
@ATKCA what's wrong with previous issue. Please don't create duplicates as this creates extra work.

Just looking as this. You could use expo on the pot?

https://www.youtube.com/watch?v=H4aWOA4-Wv0

ATKCA, 2021-11-25:
> @ATKCA what's wrong with previous issue. Please don't create duplicates as this creates extra work.
> 
> Just looking at this. You could use expo on the pot?
> 
> https://www.youtube.com/watch?v=H4aWOA4-Wv0

As explained modifying the throttle from the controller side only limits the throttle and does not limit the motor output. For example the PID will still have maximum throttle authority regardless of input limits. I would like full motor output limiting.

closed 2022-01-01 by github-actions[bot]