betaflight/betaflight #11080: Mamba D Shot 600 bad input frames

Kiffi007, 2021-11-22:
### Describe the bug

MAMBA F722 MK3 
T-Motor 1950 kv
Mamba F60 BL32 

Problem

ich habe alles zusammengebaut alle Teile sind neu 
Betaflight neuste Version geflasht 4.2.10. 
neuste Version BL heli32 32.8

bei 8 k 8k D Shot 600 wurden mir Errors angezeigt
bei 4 k 4k D Shot 300 war alles oki 0 Errors
   
das gleiche im Bl heli 32 Configurator
 
2. danach habe ich denn ESC gewegselt wieder ein Mamba F60 BL32 
   neuer ESC nur die Errors in Motor Tap  waren immer noch da

3. nach 3 nachten Googlen hab ich das bei Facebook gefunden 

  Beim Mamba f722 hat diatone die Timer beim bidirectional dshot bitbang falsch gesetzt. Versuche bidirectional dshot auszuschalten   und auf dshot 300 runter zu gehen. Dann sollte das Problem weg sein.
  Wenn es weg ist kannst du Übergangsweise folgendes versuchen:

  set dshot_bitbang=off
  dma pin a09 1
  dma pin a08 1
  dma pin c09 0
  dma pin c08 1
  set pid_process_denom=1
  set scheduler_optimize_rate=on
  set dshot_burst=off
  Anschließend sollte DShot 600 bidirectional laufen. Ein richtiger Fix kommt mit BF4.3.
  
4. habe ich so gemacht danach waren die Errors fast weg von ca 1 % auf 0,001 aber ich bin jetzt mit D Shot 300 fliegen gegangen,
   ich habe andere BF Versionen ausprobiert andere BL Heli Versionen das hat alles nichts gebraucht  ... 

### To Reproduce

### Expected behavior

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```# name: Kiffi FPV

# feature
feature GPS
feature LED_STRIP

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 4 2 115200 57600 0 115200

# led
led 0 0,0::AT:0
led 1 1,0::AT:0
led 2 2,0::AT:0
led 3 3,0::AT:0

# mode_color
mode_color 6 1 2
mode_color 7 0 10

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 4 900 1300 0 0
aux 2 2 4 1300 1700 0 0
aux 3 13 5 1700 2100 0 0
aux 4 26 1 1700 2100 0 0
aux 5 28 2 1700 2100 0 0

# master
set gyro_lowpass2_hz = 325
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 0,32,69,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set vbat_warning_cell_voltage = 340
set ibata_scale = 80
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set pid_process_denom = 2
set osd_rssi_pos = 2139
set osd_flymode_pos = 2074
set osd_crosshairs_pos = 269
set osd_current_pos = 2081
set osd_mah_drawn_pos = 2049
set osd_craft_name_pos = 463
set osd_display_name_pos = 394
set osd_gps_speed_pos = 2106
set osd_gps_sats_pos = 2112
set osd_home_dir_pos = 2062
set osd_core_temp_pos = 385
set debug_mode = GYRO_SCALED
set gyro_1_align_yaw = 1800
set name = Kiffi FPV
set display_name = Kiffi FPV

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set f_pitch = 114
set f_roll = 108
set f_yaw = 108

rateprofile 0

# rateprofile 0
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 125
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 82
set pitch_srate = 82
set tpa_breakpoint = 1750
```
PASTE THE OUTPUT OF 'resource show all' HERE
```# Betaflight / STM32F7X2 (S7X2) 4.0.1 1. Mai 2019 / 22:19:19 (25f4a5ffe) MSP-API: 1.41

board_name MAMBAF722_I2C
Hersteller_ID DIAT

# E/A
Ressourcen-LED 1 C15
Ressourcen-LED 2 C14
Ressource BEEPER 1 B02
set beeper_inversion = ON
set beeper_od = OFF

# ACC/GYRO
Ressource GYRO_EXTI 1 C04
Ressource GYRO_CS 1 A04
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW180

# SPI
Ressource SPI_SCK 1 A05
Ressource SPI_MISO 1 A06
Ressource SPI_MOSI 1 A07

Ressource SPI_SCK 2 B13
Ressource SPI_MISO 2 B14
Ressource SPI_MOSI 2 B15

Ressource SPI_SCK 3 C10
Ressource SPI_MISO 3 C11
Ressource SPI_MOSI 3 B05

# BLITZ
Ressource FLASH_CS 1 A15
setze flash_spi_bus = 3

# OSD
Ressource OSD_CS 1 B12
setze max7456_spi_bus = 2

# ADC
Ressource ADC_BATT 1 C01
Ressource ADC_RSSI 1 C02
Ressource ADC_CURR 1 C03
setze adc_device = 3

# SERIELLE PORTS
Ressource SERIAL_TX 1 B06
Ressource SERIAL_RX 1 B07

Ressource SERIAL_TX 2 A02
Ressource SERIAL_RX 2 A03

Ressource SERIAL_TX 3 B10
Ressource SERIAL_RX 3 B11

Ressource SERIAL_TX 4 A00
Ressource SERIAL_RX 4 A01

Ressource SERIAL_TX 5 C12
Ressource SERIAL_RX 5 D02

Ressource SERIAL_TX 6 C06
Ressource SERIAL_RX 6 C07

#I2C
Ressource I2C_SCL 1 B08
Ressource I2C_SDA 1 B09

# TIMER
Zeitschaltuhr B09 AF3
# Pin B09: TIM11 CH1 (AF3)
Timer C08 AF3
# Pin C08: TIM8 CH3 (AF3)
Timer C09 AF3
# Pin C09: TIM8 CH4 (AF3)
Zeitschaltuhr A08 AF1
# Pin A08: TIM1 CH1 (AF1)
Zeitschaltuhr A09 AF1
# Pin A09: TIM1 CH2 (AF1)
Timer B08 AF2
# Pin B08: TIM4 CH3 (AF2)
Zeitschaltuhr B03 AF1
# Pin B03: TIM2 CH2 (AF1)
Ressource MOTOR 1 C08
Ressource MOTOR 2 C09
Ressource MOTOR 3 A08
Ressource MOTOR 4 A09
Ressource LED_STRIP 1 B03

# dma
dma-ADC 3 0
# ADC 3: DMA2-Stream 0 Kanal 2
dma-Pin C08 0
# Pin C08: DMA2 Stream 2 Kanal 0
dma-Pin C09 0
# Pin C09: DMA2 Stream 7 Kanal 7
dma-Pin A08 0
# Pin A08: DMA2 Stream 6 Kanal 0
dma-Pin A09 0
# Pin A09: DMA2 Stream 6 Kanal 0
dma-Pin B08 0
# Pin B08: DMA1 Stream 7 Kanal 2
dma-Pin B03 0
# Pin B03: DMA1 Stream 6 Kanal 3

# seriell
seriell 0 64 115200 57600 0 115200
Serie 3 1 115200 57600 0 115200

# EXTRAS
Ressource PINIO 1 B00

# Besonderheit
Funktion -RX_PARALLEL_PWM
Funktion RX_SERIAL
Funktion TELEMETRIE
Funktion OSD

# Meister
set baro_bustype = I2C
setze baro_i2c_device = 1
set mag_bustype = I2C
setze mag_i2c_device = 1
setze serialrx_provider = SBUS
set blackbox_device = SPIFLASH
setze dshot_burst = ON
set motor_pwm_protocol = DSHOT600
setze pinio_box = 0,255,255,255
setze pinio_config = 129,1,1,1
setze current_meter = ADC
set battery_meter = ADC

### Setup / Versions
 <!-- Specify your flight controller model (what type is it, where was it bought from, ...) -->
 - Flight controller: 
 <!-- Specify other components attached to the flight controller (RX, VTX, brand / model for all of them, firmware version where applicable...) -->
 - Other components:
 <!-- Details about how all is wired -->
 - How are the different components wired up:

<!-- Add any other context about the problem that you think might be relevant here. -->

haslinghuis, 2021-11-23:
Please edit your post and use international english. I have 25% german blood in me but we al use a common language here.

DusKing1, 2021-11-23:
_the latest betaflight release_ is 4.2.11 and we have fixed this issue in that release. you should change to that.

Kiffi007, 2021-11-23:
I also tried BF 4.2.11 problem is still there

DusKing1, 2021-11-23:
So what's your reporduce step?

Kiffi007, 2021-11-23:
[image]

Kiffi007, 2021-11-23:
yes Full chip erease

D Shot 600 Bad Frames
D Shot 300 everything ok 0 Errors

DusKing1, 2021-11-23:
I can not reproduce it

haslinghuis, 2021-11-23:
Maybe have to add a low ESR capacitor to the battery pads.

UglyBabyGaming, 2021-12-19:
I just got the 20x20mm mamba f722 mini mk3 with the pro 32bit esc, I also get the error randomly on random motors over 5% at full power brand new board left at 8k 8k but changed to dshot300 no errors... low ESR capacitor fitted to board

trbarton, 2022-01-04:
Experiencing the same issue and behaviour on my new build. Exact same stack both in BF4.2 and BF4.3 RC 1
Also notice one motor spins up with a delay as mentioned above

haslinghuis, 2022-01-04:
Please also check ESC firmware.

alienwaredream, 2022-01-06:
> yes Full chip erease
> 
> D Shot 600 Bad Frames D Shot 300 everything ok 0 Errors

Same situation as you. My motor is brand new T-motor P2207
I use HGLRC Zeus F722(4.2.11,4.3RC1) FC and MAMBA F50_BL32(32.8,32.8.3) ESC
at 8k 8k bidirectional DShot 600  got 1-5% errors
but at 8k 4k bidirectional DShot 300  works ok   no error

after this, I also tested  HGLRC Zeus F722(4.2.11,4.3RC1) FC and  hobbywing XRotor  60A (32.8.3) ESC 
at 8k 8k bidirectional DShot 600  no error

so I think it might be a ESC problem? Or MAMBA F50_BL32 donot support bidirectional DShot 600 at all?

alienwaredream, 2022-01-09:
UPUP!
Still error on BF4.3RC2

haslinghuis, 2022-01-09:
Can you try BLHeli 32.7?

trbarton, 2022-01-09:
Same issue on BLHeli 32.7 I swapped to a Tmotorf7 fc and the problem is still present in both 4.2.11 and 4.3.0 RC1 and RC2 so I presume this is an ESC problem

filipslavik, 2022-01-11:
I have the exact same issue. On BF 4.2.11 with BLHeli 32.8.

UglyBabyGaming, 2022-01-13:
wondering if we try rolling back the bl-heli32 to older firmware??? I had the issue on BF 4.2.11 and 4.3 or are we looking at a manufacturing issue at the Diatone factory?

haslinghuis, 2022-01-13:
So it works on one combination but not the other. I guess this is reason to report this on BLHeli.
> I use HGLRC Zeus F722(4.2.11,4.3RC1) FC and MAMBA F50_BL32(32.8,32.8.3) ESC
at 8k 8k bidirectional DShot 600 got 1-5% errors
but at 8k 4k bidirectional DShot 300 works ok no error

> after this, I also tested HGLRC Zeus F722(4.2.11,4.3RC1) FC and hobbywing XRotor 60A (32.8.3) ESC
at 8k 8k bidirectional DShot 600 no error
Look at tasks command in cli (execute two times to get stable readings)

Attached to https://github.com/bitdump/BLHeli/issues/582

UglyBabyGaming, 2022-02-06:
06/Feb/2022 still a problem on 4.3 RC3 running mamba f722 mini 20x20mm mk3 on 40A mamba pro with low ESR cap that is fully insulated. latest Bl32, 4s 850mah Tattu r-line, flywoo 2104 3000kv.. getting around 5% error random on random motors when using D-shot 600... no issue with D-shot 300

closed 2022-03-21 by github-actions[bot]