betaflight/betaflight #11087: G47x USB broken

DusKing1, 2021-11-30:
### Describe the bug
The USB was broken on G47x target since build#2538 and later version.

### To Reproduce
@SteveCEvans you can grab the AirFleet G474 dev board and flash the HYBRIDG4 target to reproduce this problem, build#2537 was ok, build#2538 can't recognize the USB device.

### Expected behavior
Flash the latest G47x target and all works fine

### Setup / Versions
 
 - Flight controller: AirFleet G474 dev board with HYBRIDG4 target

DusKing1, 2021-11-30:
@github-actions fck F

SteveCEvans, 2021-11-30:
Commit 305bfda3cc10a3edf19878cc164c451d3a2d2b31 is OK, the following commit ea3e64d49ae1c66ab11e72c10518def64d976ef4 results in a hard fault.

```
commit ea3e64d49ae1c66ab11e72c10518def64d976ef4
Merge: 305bfda 6a6930e
Author: Michael Keller <github@ike.ch>
Date:   Sun Oct 10 16:12:56 2021 +1300

    Merge pull request #10987 from mikeller/add_simplified_tuning_on_reset

commit 305bfda3cc10a3edf19878cc164c451d3a2d2b31
Merge: c32a770 e42c3f2
Author: Michael Keller <github@ike.ch>
Date:   Sun Oct 10 16:12:30 2021 +1300

    Merge pull request #11001 from JoeriCapens/cms-simplified-tuning-autoapply
```

SteveCEvans, 2021-11-30:
@DusKing1 Please try the attached image
[betaflight_4.3.0_STM32G47X_e8ce157.hex.zip](https://github.com/betaflight/betaflight/files/7628586/betaflight_4.3.0_STM32G47X_e8ce157.hex.zip)

DusKing1, 2021-12-01:
Thank you for your fix for this!

SteveCEvans, 2021-12-01:
@DusKing1 Please confirm that this image works for you. This fixes the underlying cause rather than plastering over the cracks
[betaflight_4.3.0_STM32G47X_71f8eb8.hex.zip](https://github.com/betaflight/betaflight/files/7637537/betaflight_4.3.0_STM32G47X_71f8eb8.hex.zip)
.

DusKing1, 2021-12-02:
@SteveCEvans Tested, `71f8eb8` can work both on BMI270 and LSM6DSO dev G4 board

closed 2021-12-12 by blckmn