betaflight/betaflight #11089: ADC current sensor value has too much jitter

cmanley, 2021-11-30:
### Describe the bug
I have a GEPRC F722 35A AIO flight controller. While running the motors on the bench at a constant speed using the Motors tab in Configurator or when flying horizontally with sticks neutral, the current value displayed in the OSD and/or Configurator has what looks to me like too much jitter (roughly +- 40%). Measuring the current using a clamp meter does not show this jitter. My guess is that a filter function is being applied to the current sensor values, but that it does not smooth the result enough, at least not for this board. I've tried using both 8khz and 4khz loop times but that didn't make a difference.

### To Reproduce
Do what I did above. I'm not sure if it's specific to my board or to all (F722) boards.

### Expected behavior
A smooth current value in undisturbed sticks-neutral horizontal flight.

### Flight controller configuration
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.11 Nov 19 2021 / 15:35:27 (norevision) MSP API: 1.43
# config: manufacturer_id: GEPR, board_name: GEPRC_F722_AIO, version: d3484d96, date: 2021-04-28T07:05:45Z

# start the command batch
batch start

board_name GEPRC_F722_AIO
manufacturer_id GEPR

# dma
dma pin B06 NONE
dma pin B07 NONE
dma pin B01 NONE
dma pin B00 NONE

# feature
feature -SOFTSERIAL
feature -LED_STRIP
feature -AIRMODE
feature GPS
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# serial
serial 0 1 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 4 0 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1800 2100 0 0
aux 2 2 1 1325 1650 0 0
aux 3 46 2 1800 2100 0 0
aux 4 13 2 1100 1375 0 0
aux 5 28 1 900 1800 0 0
aux 6 35 2 1375 1625 0 0

# rxfail
rxfail 7 s 1000

# master
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -42,-1,-80,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_poles = 12
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 90
set bat_capacity = 3000
set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 310
set vbat_scale = 111
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 40
set gps_rescue_descent_dist = 40
set gps_rescue_ground_speed = 1500
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set deadband = 3
set yaw_deadband = 10
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_warn_rssi = ON
set osd_warn_dji = ON
set osd_rssi_alarm = 80
set osd_rssi_dbm_alarm = -99
set osd_cap_alarm = 850
set osd_vbat_pos = 1
set osd_rssi_pos = 14471
set osd_rssi_dbm_pos = 140
set osd_tim_1_pos = 198
set osd_tim_2_pos = 205
set osd_flymode_pos = 6541
set osd_vtx_channel_pos = 371
set osd_current_pos = 14694
set osd_mah_drawn_pos = 14726
set osd_craft_name_pos = 14607
set osd_display_name_pos = 394
set osd_gps_speed_pos = 2262
set osd_gps_lon_pos = 2416
set osd_gps_lat_pos = 2384
set osd_gps_sats_pos = 2197
set osd_home_dir_pos = 2190
set osd_home_dist_pos = 2191
set osd_altitude_pos = 2324
set osd_pid_roll_pos = 353
set osd_pid_pitch_pos = 34
set osd_power_pos = 358
set osd_pidrate_profile_pos = 2447
set osd_warnings_pos = 14573
set osd_avg_cell_voltage_pos = 14662
set osd_pit_ang_pos = 226
set osd_battery_usage_pos = 457
set osd_disarmed_pos = 2252
set osd_flip_arrow_pos = 310
set osd_rate_profile_name_pos = 23
set osd_pid_profile_name_pos = 2
set osd_stat_max_spd = OFF
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set debug_mode = GYRO_SCALED
set name = RSSI LOW

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set anti_gravity_gain = 5000
set feedforward_transition = 30
set iterm_relax_type = GYRO
set p_pitch = 56
set i_pitch = 65
set d_pitch = 70
set f_pitch = 75
set p_roll = 52
set i_roll = 60
set d_roll = 62
set f_roll = 75
set p_yaw = 90
set i_yaw = 70
set f_yaw = 80
set d_min_roll = 39
set d_min_pitch = 22

rateprofile 0

# rateprofile 0
set thr_mid = 32
set thr_expo = 50
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
set tpa_rate = 75
set tpa_breakpoint = 1250

# end the command batch
batch end
```
```
esource show all
###ERROR: input: UNKNOWN COMMAND, TRY 'HELP'###
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: FLASH_CS
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: LED 1
C05: FREE
C06: MOTOR 4
C07: MOTOR 3
C08: MOTOR 2
C09: MOTOR 1
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: GEPRC F722 35A AIO from banggood
 https://geprc.com/product/gep-f722-35a-aio-f722-fc-35a-2-6s-8bits-bls-esc-26-5mm-m2/
 
 - Other components: GPS + DJI Caddx + TBS receiver + beeper, all on separate hardware UARTs.
 
 - How are the different components wired up: According to Crocodile Baby 4 HD (latest version with F722) as sold.

Could it have something to do with src/main/sensors/battery.c and .ibatLpfPeriod = 10 ?
If the smoothing function can be controlled in theory using CLI commands, then that'll be a nice feature to resolve such issues.

KarateBrot, 2021-11-30:
Are you sure this is a problem of the filtering and not a problem of the electrical circuit that maybe has too much noise to begin with?

cmanley, 2021-11-30:
No I'm not sure. I've got a 330uF capacitor on the input if that helps. The jitter also occurs at low current draws which is when electrical noise normally isn't too bad afaik.

klutvott123, 2021-11-30:
You can try increasing ```ibat_lpf_period``` in cli.
+-40% variation sure sounds like an electrical issue. Try a smaller ceramic cap from the current wire to ground.

cmanley, 2021-11-30:
Thank you. I didn't realize ibat_lpf_period was a CLI parameter. I'll give that a try and get back after some tests (weather/work permitting).

cmanley, 2021-11-30:
Update based on Configurator with motors at 3500 rpm: the result of setting ibat_lpf_period to 50 looks promising already. 
I just don't know for sure how much lag ibat_lpf_period will introduce into the current sensing result or if high values have any other downsides as I can't find any documentation about it anywhere. Thanks again :)

klutvott123, 2021-11-30:
The period is in seconds x 10 so 50 is a period of 5 seconds and a cutoff of 0.2 Hz. It results in a significant delay, but if you don't care about transients during short throttle blips and such it should be fine. 🙂

closed 2021-11-30 by klutvott123