betaflight/betaflight #11092: Add generic pin/swap functionality disabled ESC telemetry on #2518 and subsequent

hawkeyemedia, 2021-12-01:
### Describe the bug
Using a Holybro Kakute F7 board on Betaflight 4.3 #2518, there is no ESC telemetry. Using version #2517 the ESC telemetry works. 

It looks like there was a change made for #2518 for the UART - Add generic pin/swap functionality.
If RX pin resource is assigned to a TX hardware pin, or vice versa, then
pin-swap is enabled.
 (commit: b5a61f6)

This change seems to have disable the ESC telemetry. 

The Kakute F7 is using UART 7 for the ESC and when I load version 4.3 #2518 or subsequent, this functionality is disabled.

### To Reproduce
Load firmware #2517, set UART 7 Sensor input to ESC, set ESC protocol to DSHOT600, enable ESC_SENSOR. The data for esc temperatures and current are available and will display.
Load firmware #2518 or subsequent, set UART 7 Sensor input to ESC, set ESC protocol to DSHOT600, enable ESC_SENSOR. There is no data available, no temperatures, no current.

### Expected behavior
When setting UART 7 Sensor input to ESC, set ESC protocol to DSHOT600, enable ESC_SENSOR, I expect to be able to see the esc temperatures and amperage.

### Flight controller configuration
```
# diff

# version
# Betaflight / STM32F745 (S745) 4.3.0 Sep  9 2021 / 14:23:19 (df5716bbb) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# feature
feature ESC_SENSOR

# serial
serial 6 1024 115200 57600 0 115200

# master
set acc_calibration = 92,15,55,1
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set current_meter = ESC
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: SERIAL_RX 7
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM2: FREE
TIM3:
    CH3 : MOTOR 1
    CH4 : MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: MOTOR 4
DMA2 Stream 3: MOTOR 3
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Setup / Versions
 
 - Flight controller: 
 - Holybro Kakute F7 V1.0 purchased from Race Day Quads
 
 - Other components:
 - Joshua Bardwell 4 in 1 ESC
 - FrSky RXSR receiver
 - Caddx Polar Vista VTX with DJI xmtr
 
 - How are the different components wired up:
esc is plugged into the FC. Radio receiver is on UART 6

hawkeyemedia, 2022-01-02:
I looked through the code for Betaflight 4.3 RC1 and the new code is there. I'm not sure why enabling the ability to pin swap disables the ESC telemetry on my FC. The ESC telemetry uses UART 7, is it possible the new code is swapping the pins and is there a way I can use a CLI command to swap them back? The ESC I'm using doesn't have a single current sensor so the only way to monitor current is using the ESC telemetry. I'm almost at the point of learning how to disable this code and compile my own firmware. Or upgrade to a new ESC. 

No one has any comment on this? Hydra?

SteveCEvans, 2022-01-03:
@hydra Can you please take a look at this. I'm looking now, but I'd more than welcome you beating me to a solution. This bus was introduced by https://github.com/betaflight/betaflight/pull/10940

SteveCEvans, 2022-01-03:
@hydra To aid your investigation. On a `TMOTORF7` using `UART4` for the `ESC SENSOR`, we have the following resources.

```
resource SERIAL_TX 4 A00
resource SERIAL_RX 4 A01
```

In contrast, on a `SPEEDYBEEF7V2`, also using UART4 which doesn't work, we have.

```
resource SERIAL_TX 4 NONE
resource SERIAL_RX 4 A01
resource CAMERA_CONTROL 1 A00
```

Defining `resource SERIAL_TX 4 A00` fixes the `SPEEDYBEEF7V2`.

mituritsyn, 2022-01-04:
Is it duplicate of #11054  ?

hawkeyemedia, 2022-01-04:
I switched back to BF 4.3 #2517 and verified the ESC telemetry worked using UART R7, installed 4.3 RC1 and no ESC Telemetry. My solution was to cut the wire going into RX7 and move it to RX3, an unused UART. ESC telemetry works, so my problem is solved. 

I tried setting SERIAL_RX 7 to none, save, reboot. Then reset it to E07, but it didn't help. I tried swapping the pin used for RX6 to RX7, didn't work. In the end, swapping the input signal to RX3 solved my problem.
Mark

closed 2022-01-06 by haslinghuis

hydra, 2022-01-09:
Hey all, I'm somewhat behind in checking BF related emails that aren't directly related to the H7RF right now.  I see the fix has been found and merged and I was glad I'm not the only experienced BF dev/reviewer that didn't spot the issue in the original PR. 😌