betaflight/betaflight #11093: Matek H743 Slim stuck forever when having a sd card inserted

dragnea, 2021-12-01:
### Describe the bug

Powering up a Matek H743-Slim with a sd card inserted, it will hang forever, Betaflight 4.3, of course.
I have tried 4 sd cards, from 1GB to 32GB, with several nightly builds.
Tried to format them FAT16 or FAT32... only one 1GB FAT16 worked partially, as it has missing parts in the log, 2khz or 1khz, same thing.

Note: All the sd cards work correctly on all the F722-SE's I have, with 2Khz / GYRO_RAW settings, on Betaflight 4.2.x.

### To Reproduce

Take one SD Card 8GB for example, format to FAT32, using windows or https://www.sdcard.org/downloads/formatter/, put it in the fc slot, power up the fc, the OSD don't appear, esc's will not output the initialization sound, nothing happens. Remove the power, eject the sd card, plug in the power again, all good.

### Expected behavior

### Flight controller configuration

```
# version
# Betaflight / MATEKH743 (M743) 4.3.0 Nov  9 2021 / 20:17:57 (6ecd0d232) MSP API: 1.44

# start the command batch
batch start

board_name MATEKH743

# name: FR7

# feature
feature GPS

# serial
serial 1 64 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# mixer
mixer CUSTOM
mmix 0  1.000 -0.809  0.659 -1.000
mmix 1  1.000 -1.000 -0.659  1.000
mmix 2  1.000  0.809  0.659  1.000
mmix 3  1.000  1.000 -0.659 -1.000

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 46 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905 5925    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 23 27 29
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 0 h
rxfail 1 h
rxfail 2 h

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 500
set acc_calibration = -73,-18,267,1
set mag_hardware = NONE
set max_check = 1950
set serialrx_provider = CRSF
set dshot_idle_value = 600
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 210
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_ground_speed = 1600
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set deadband = 4
set yaw_deadband = 4
set osd_alt_alarm = 1000
set osd_vbat_pos = 33
set osd_link_quality_pos = 2072
set osd_link_tx_power_pos = 2049
set osd_rssi_dbm_pos = 2061
set osd_flymode_pos = 2106
set osd_throttle_pos = 2274
set osd_current_pos = 2112
set osd_mah_drawn_pos = 2282
set osd_gps_speed_pos = 2307
set osd_gps_lon_pos = 2509
set osd_gps_lat_pos = 2496
set osd_gps_sats_pos = 2522
set osd_home_dir_pos = 2127
set osd_home_dist_pos = 2093
set osd_altitude_pos = 18710
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2081
set osd_stat_max_dist = ON
set osd_stat_min_rssi = OFF
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_min_rssi_dbm = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set rpm_filter_min_hz = 90
set name = FR7

profile 0

# profile 0
set iterm_relax_cutoff = 7
set p_pitch = 74
set i_pitch = 132
set d_pitch = 82
set f_pitch = 0
set p_roll = 67
set i_roll = 120
set d_roll = 66
set f_roll = 0
set p_yaw = 67
set i_yaw = 120
set f_yaw = 0
set d_min_roll = 49
set d_min_pitch = 61
set feedforward_averaging = 3_POINT
set simplified_master_multiplier = 150
set simplified_d_gain = 110
set simplified_feedforward_gain = 0
set simplified_pitch_d_gain = 125
set simplified_pitch_pi_gain = 110

rateprofile 0

# rateprofile 0
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 60

# end the command batch
batch end
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller:  Matek H743-Slim
 
 - Other components:
 
 - How are the different components wired up:
...

SteveCEvans, 2021-12-01:
Working fine with a Sandisk Ultra 32Gbyte card.

```
# status
MCU H743 (Rev.V) Clock=480MHz, Vref=3.30V, Core temp=48degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 3927, max available: 131072
Devices detected: SPI:2, I2C:1
Gyros detected: gyro 1, gyro 2 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=DPS310
OSD: MAX7456
System Uptime: 20 seconds, Current Time: 2021-12-01T23:20:02.601+00:00
CPU:22%, cycle time: 104, GYRO rate: 9615, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 5
SD card: Manufacturer 0x3, 31166976kB, 05/2005, v8.0, 'SC32G'
Filesystem: Ready
Arming disable flags: RXLOSS CLI MSP RPMFILTER
```

dragnea, 2021-12-02:
Ok, but have you tried to arm your quad to see if there is something logged? I get something like "Initializing..." or "Ready", but it doens't log anything.
I will upload my statuses tonight, when I come back from work.
Thanks!

SteveCEvans, 2021-12-02:
I have setup with MSP rx and successfully armed and logged using build `6ecd0d232` and current master.

dragnea, 2021-12-02:
This is the SD card removed recently from Matek F722-SE - BF 4.2

```
# status
MCU H743 (Rev.V) Clock=480MHz, Vref=3.31V, Core temp=41degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 3928, max available: 131072
Devices detected: SPI:2, I2C:1
Gyros detected: gyro 1, gyro 2 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=DPS310
OSD: MAX7456
System Uptime: 35 seconds, Current Time: 2021-12-02T20:19:50.630+00:00
CPU:5%, cycle time: 129, GYRO rate: 7751, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 1
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP RPMFILTER
```

[image]

[image]

closed 2022-01-12 by github-actions[bot]

haslinghuis, 2022-01-12:
@dragnea Did you try RC2 or last nightly dev build?