betaflight/betaflight #11109: HGLRC Zeus F722 no Barometer

Ahrimaan, 2021-12-06:
### Describe the bug
Zeus F722 with built-in Barometer, but in Betaflight no Baro is shown.
Checkbox is ticked.

### To Reproduce

### Expected behavior
Expected to have a Barometer

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Dec  3 2021 / 00:23:40 (95589e0fc) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: 836acaf9, date: 2021-10-31T19:48:38Z

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR

# feature
feature OSD
feature ESC_SENSOR

# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# master
set acc_calibration = 115,-78,-4120,1
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set current_meter = ESC
set battery_meter = ESC

profile 0

rateprofile 0

# end the command batch
batch end

# 
```
PASTE THE OUTPUT OF 'resource show all' HERE
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: PWM 1
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
    CH3 : MOTOR 3
    CH4 : MOTOR 4
TIM4: FREE
TIM5:
    CH1 : PWM 3
    CH2 : PWM 2
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9:
    CH1 : PWM 1
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# ```

### Setup / Versions
 <!-- Specify your flight controller model (what type is it, where was it bought from, ...) -->
 - Flight controller: 
 HGLRC Zeus F722
https://www.hglrc.com/products/hglrc-zeus-f722-3-6s-f722-flight-controller
 <!-- Specify other components attached to the flight controller (RX, VTX, brand / model for all of them, firmware version where applicable...) -->
 - Other components:
 -TBS Nano Receiver
 <!-- Details about how all is wired -->
 - How are the different components wired up:
 TBS Nano on RX1/TX1

alienwaredream, 2021-12-22:
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0

https://github.com/betaflight/unified-targets/pull/524

alienwaredream, 2021-12-23:
under 4.3.x
https://github.com/betaflight/unified-targets/blob/master/configs/default/HGLR-HGLRCF722.config

haraldcool, 2021-12-29:
Hglrc is setting the baro to i2c in the config file... But the baro is on spi 1 together with the gyro. Since Bf 4.3 it does not work anymore setting the baro back to spi 1. When you do, the controller gets stuck a few seconds after startup.

alienwaredream, 2021-12-30:
> Hglrc is setting the baro to i2c in the config file... But the baro is on spi 1 together with the gyro. Since Bf 4.3 it does not work anymore setting the baro back to spi 1. When you do, the controller gets stuck a few seconds after startup.

Is there any way to fix it?

shahmirthesquid, 2022-01-01:
I'm having this same exact issue with my HGLRC F722 as well. I tried several different betaflight versions but none of them worked. Flashing inav firmware finally got my barometer working. I was worried that I had a defective unit for a bit haha.

I hope this issue can get resolved quickly, I was very excited to try out betaflight 4.3 so I don't want to have to switch to new firmware just to get my components working

AFintheair, 2022-01-02:
the same problem with me. Zeuss F722 mini (20x20). the Baro works with BF4.2 (The altitude is displayed correctly in the DJI OSD, but I do not know if it is also used correctly for GPS Rescue.  But I assume so.) , but not with BF4.3RC1.
I hope that the problem is easy to solve, as I have equipped all my LR copters with this FC and do not want to do without the Baro. But I'm hot on BF 4.3;-). 
Many thanks to the developers. And sorry for my bad english. 
The diff is from a complete new setup, I did yesterday. maybe not complete. It was not in the air already.
[BTFL_cli_TransformerWhite_20220101_205941_Start.txt](https://github.com/betaflight/betaflight/files/7798718/BTFL_cli_TransformerWhite_20220101_205941_Start.txt)
.

SteveCEvans, 2022-01-03:
This may be fixed by https://github.com/betaflight/betaflight/pull/11215. Could somebody please test using the images in the PR. Thanks.

haraldcool, 2022-01-03:
Wonderful, it doesn't get stuck anymore!! Thank you!!