betaflight/betaflight #11119: CLRacingF7-unable to access log files when activating mass storage device with 4.3 development builds

Loadntz, 2021-12-12:
### Describe the bug
No log files visible in the drive when activating mass storage device mode on 4.3 development builds for my given target. Log files are on the onboard flash, just not accessible. 

### To Reproduce
Create a black-box flight log then enter betaflight configurator. Go into the blackbox tab and "activate mass storage device mode"

### Expected behavior
The storage device drive opens up as normal, however the drive contains no log files. It contains an autorun file, a betaflight logo and a "padding" text document which it says is 256 MB.

### Flight controller configuration
```# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Dec  9 2021 / 23:13:55 (ca25215f4) MSP API: 1.44
# config: manufacturer_id: CLRA, board_name: CLRACINGF7, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name CLRACINGF7
manufacturer_id CLRA

# name: Badger

# resources
resource MOTOR 1 B09
resource MOTOR 2 B08
resource MOTOR 3 B07
resource MOTOR 4 B06

# feature
feature GPS
feature TELEMETRY

# serial
serial 2 2048 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 2100 0 0
aux 2 27 1 1700 2100 0 0
aux 3 46 1 2075 2100 0 0
aux 4 13 3 1775 1825 0 0
aux 5 13 3 1875 1925 0 0
aux 6 26 0 1700 2100 0 0
aux 7 35 3 1375 1425 0 0
aux 8 35 3 1775 1825 0 0
aux 9 36 2 1700 2100 0 0
aux 10 40 0 900 1000 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 26 29 36
vtxtable powerlabels 25 400 800 1W6

# master
set gyro_lpf1_static_hz = 0
set gyro_to_use = FIRST
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_max_hz = 350
set acc_calibration = 3,-1,-160,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rc_smoothing_setpoint_cutoff = 30
set rc_smoothing_feedforward_cutoff = 30
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_idle_value = 500
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set failsafe_switch_mode = STAGE2
set vbat_scale = 128
set vbat_divider = 8
set ibata_scale = 164
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_set_home_point_once = ON
set gps_rescue_angle = 35
set gps_rescue_descent_dist = 100
set gps_rescue_yaw_p = 100
set gps_rescue_ascend_rate = 1000
set gps_rescue_throttle_hover = 1250
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = CURRENT_ALT
set gps_rescue_use_mag = OFF
set deadband = 2
set yaw_deadband = 2
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 35
set osd_cap_alarm = 1300
set osd_tim1 = 2561
set osd_vbat_pos = 2529
set osd_rssi_pos = 474
set osd_link_quality_pos = 2519
set osd_rssi_dbm_pos = 465
set osd_tim_1_pos = 2551
set osd_tim_2_pos = 408
set osd_anti_gravity_pos = 302
set osd_throttle_pos = 2497
set osd_vtx_channel_pos = 1
set osd_crosshairs_pos = 269
set osd_ah_pos = 110
set osd_current_pos = 2536
set osd_mah_drawn_pos = 2544
set osd_craft_name_pos = 10
set osd_gps_speed_pos = 2082
set osd_gps_sats_pos = 2104
set osd_home_dir_pos = 2092
set osd_home_dist_pos = 2097
set osd_debug_pos = 1
set osd_warnings_pos = 2505
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_total_dist = ON
set debug_mode = GYRO_SCALED
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5740
set rpm_filter_fade_range_hz = 100
set name = Badger

profile 0

# profile 0
set dterm_lpf1_dyn_expo = 8
set dterm_lpf1_static_hz = 0
set anti_gravity_gain = 4000
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 0
set d_pitch = 50
set d_roll = 44
set d_min_roll = 33
set d_min_pitch = 37
set feedforward_jitter_reduction = 12
set simplified_d_gain = 110

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 152
set pitch_rc_rate = 152
set yaw_rc_rate = 108
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 64
set pitch_srate = 64
set yaw_srate = 64
set tpa_breakpoint = 1250

# end the command batch
batch end```

```Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: PINIO 1
A15: OSD_CS
B00: LED 1
B01: FREE
B02: FREE
B03: CAMERA_CONTROL
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 4
B07: MOTOR 3
B08: MOTOR 2
B09: MOTOR 1
B10: SERIAL_TX 3
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: MCO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: PREINIT 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE```

### Setup / Versions
 - Flight controller: CLRacing F7 MPU6000 v2.1 purchased from PhaserFPV

 - Other components: Hobbywing XRotor 60A 4in1 ESC, RushTank Solo VTX, TBS Crossfire RX, Beltian 220 GPS

 - How are the different components wired up: VTX powered via Vbat through FC and smartaudio on UART3, RX powered via 5V BEC on FC and using UART5, GPS powered via 5V BEC on FC and using UART6.

The issue appears to be localized to the CLRACINGF7 target and has been apparent on all betaflight 4.3 development builds to date. If I switch back to the latest release of 4.2 the same blackbox logs can be accessed without any issue. Also worth noting that the logs can be accessed when using the "save flash to file" method, albeit very slowly.

SteveCEvans, 2021-12-12:
@Loadntz Thank you for reporting this. I've tracked this down to the following commit.
```
commit 8730a8efeae1581d20a28a6e11d06fa711ce8a1c
Author: Michael Keller <github@ike.ch>
Date:   Fri Oct 1 03:21:04 2021 +1300

    Fixed SPI SD card support on H7 targets with SDIO support enabled.
```

Can you please confirm that this image from the prior commit also works for you:
[betaflight_4.3.0_STM32F7X2_a1c6cda.hex.zip](https://github.com/betaflight/betaflight/files/7698740/betaflight_4.3.0_STM32F7X2_a1c6cda.hex.zip)

And that this one doesn't:
[betaflight_4.3.0_STM32F7X2_8730a8e.hex.zip](https://github.com/betaflight/betaflight/files/7698737/betaflight_4.3.0_STM32F7X2_8730a8e.hex.zip)

SteveCEvans, 2021-12-12:
Image from https://github.com/betaflight/betaflight/pull/11122 for test available at https://artprodeau1.artifacts.visualstudio.com/A3a9037bc-5464-4b2e-9e8d-ed1a5aa29705/ce1d2eeb-ce65-4b95-a02a-7fee6c006c71/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL0JldGFmbGlnaHQvcHJvamVjdElkL2NlMWQyZWViLWNlNjUtNGI5NS1hMDJhLTdmZWU2YzAwNmM3MS9idWlsZElkLzQ4NjYvYXJ0aWZhY3ROYW1lL2JldGFmbGlnaHQ1/content?format=zip

Loadntz, 2021-12-13:
@SteveCEvans  That is amazing. 

I can confirm that the first hex file fixed the issue and the second hex file doesn't as you indicated.

Much appreciated. Thank you.

closed 2021-12-15 by haslinghuis