betaflight/betaflight #11124: XRACER SPI servo not moving after firmware upgrade

1205g, 2021-12-12:
### Describe the bug
Got an x-racer spi the other week, i configured it as "airplane" and i was instantly able to move all the servos.
i then decided to upgrade firmware to the latest one 2.5 (cant remember which firmware was on before) so i flash erased and uploaded the new firmware.
configured again as "airplane", but this time servos are not moving at all, i even checked with an oscilloscope that no pwm signal is being output in any channel except for motor 1.
as i move the controller i can see the bar moving in Motors>servos.

using CLI "dump" command, i noticed that every servo was "NONE"

resource SERVO 6 NONE

meanwhile the motor was connected to an output channel

resource MOTOR 6 B09

so i did free every motor and connected the relative pin to the servo but nothing happened, using resource SERVO 6 B09

finally i would like to point out that even after saving and rebooting, the "resource" command tell me that:
resource SERVO 6 B09.
but using the "resource list", the B09 pin is "free":
B09: FREE
### To Reproduce

-erase and flash the latest firmware (v2.5) for xracerspi
-configure it as "airplane"
-servos is not moving, so i enter CLI and write the following commands:
resource MOTOR 6 NONE
resource SERVO 6 B09
save
-servo still not moving
-"resource" tell the servo as connected to b09
 "resource list" tell that b09 is free
### Expected behavior
expecting the servo signal to be output as before the upgrade

### Flight controller configuration

```
# diff
# Cleanflight / X_RACERSPI (XRC3) 2.5.0 Oct  1 2018 / 04:38:57 (08348b705) MSP API: 1.40

board_name X_RACERSPI
manufacturer_id 
resource MOTOR 6 NONE
resource SERVO 6 B09
mixer AIRPLANE

profile 0

rateprofile 0

```
```
PASTE THE OUTPUT OF 'resource show all' HERE

# resource show all
###ERROR### Invalid

# resource
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 7 A02
resource MOTOR 8 A03
resource SERVO 6 B09
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B00
resource PWM 6 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 11 B00
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource LED 1 C14
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 A00
resource ADC_BATT 1 A04
resource ADC_RSSI 1 B02
resource ADC_CURR 1 A05
resource FLASH_CS 1 B12
resource SPI_PREINIT_IPU 1 A15
resource SPI_PREINIT_IPU 2 B12

# resource list
IO
A00: PPM
A01: FREE
A02: FREE
A03: FREE
A04: ADC_BATT
A05: FREE
A06: MOTOR 1
A07: FREE
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: MPU_CS
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C14: LED 1
C15: BEEPER
F00: FREE
F01: FREE
F03: FREE
F04: FREE
```

### Setup / Versions

 - x-racer spi f303 v3.1
 - cleanflight 2.4
 
 
 - How are the different components wired up:
 usb in the flight controller and regulated 5v to the 5v pin

1205g, 2021-12-12:
edited the message

etracer65, 2021-12-12:
This is the Betaflight GitHub, not Cleanflight. Try https://github.com/cleanflight/cleanflight

closed 2021-12-12 by etracer65