betaflight/betaflight #11125: SPI reciever FrSky D8 Mode will not bind in Betaflight 4.3

ipetepete, 2021-12-13:
### Describe the bug

Trying to bind in D8 mode will not work unless previously bound in an earlier firmware build. 
D16 binds just fine.

### To Reproduce

Flash board with SPI reciever. Try to bind in D8 mode.

### Expected behavior

Reciever should bind

### Flight controller configuration

```
# version
# Betaflight / STM32F411 (S411) 4.3.0 Dec 12 2021 / 02:27:17 (adcc36888) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# name: Moblite7

# resources
resource MOTOR 1 B08
resource MOTOR 2 B07
resource MOTOR 3 B06
resource MOTOR 4 B10

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature TELEMETRY

# serial
serial 1 2048 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1775 2100 0 0
aux 1 1 1 900 1225 0 0
aux 2 2 1 1375 1775 0 0
aux 3 28 1 1375 2100 0 0
aux 4 35 2 1700 2100 0 0

# adjrange
adjrange 0 0 3 1150 1375 12 3 0 0
adjrange 1 0 3 1425 1575 12 3 0 0
adjrange 2 0 3 1675 1825 12 3 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# master
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 720
set acc_trim_pitch = -4
set acc_calibration = -77,68,-4,1
set mag_hardware = NONE
set baro_hardware = NONE
set fpv_mix_degrees = 25
set rx_spi_protocol = FRSKY_D
set dshot_idle_value = 1000
set motor_poles = 12
set align_board_yaw = 180
set vbat_max_cell_voltage = 431
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 6177
set osd_rssi_pos = 2105
set osd_link_quality_pos = 2101
set osd_tim_2_pos = 2113
set osd_flymode_pos = 2457
set osd_craft_name_pos = 396
set osd_warnings_pos = 2378
set osd_disarmed_pos = 2219
set osd_core_temp_pos = 33
set osd_rate_profile_name_pos = 2434
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = NTSC
set frsky_spi_tx_id = 27,46,0
set frsky_spi_offset = -26
set frsky_spi_bind_hop_data = 0,30,60,91,120,150,180,210,5,35,65,95,125,155,185,215,10,40,70,100,130,160,190,221,15,45,75,105,135,165,195,225,20,50,80,110,140,170,200,230,25,55,85,115,145,175,205,0,0,0
set gyro_1_align_yaw = 900
set name = Moblite7

profile 1

# profile 1
set anti_gravity_gain = 10000
set iterm_relax_cutoff = 10
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set throttle_boost = 20
set p_pitch = 66
set i_pitch = 117
set d_pitch = 99
set f_pitch = 176
set p_roll = 63
set i_roll = 112
set d_roll = 86
set f_roll = 168
set p_yaw = 63
set i_yaw = 112
set f_yaw = 168
set angle_level_strength = 63
set horizon_level_strength = 63
set horizon_transition = 90
set level_limit = 65
set horizon_tilt_effect = 80
set horizon_tilt_expert_mode = ON
set d_min_roll = 65
set d_min_pitch = 74
set level_race_mode = ON

rateprofile 0

# end the command batch
batch end
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 3
B07: MOTOR 2
B08: MOTOR 1
B09: LED
B10: MOTOR 4
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 4
TIM3: FREE
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 2
    CH3 : MOTOR 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: Crazeebeef4r
 
 - Other components: SPI FrSky 
 
 - How are the different components wired up:
Default AIO board

I know of one other person having the same issue. When binding to a new transmitter it will not bind, and if using the defaults in BF after flashing it will not bind either. Flashing 4.2 seems to remedy the issue.

haslinghuis, 2021-12-13:
ping @SteveCEvans

Asizon, 2021-12-13:
No ping needed, D8 protocol doesnt work with ISRM tx or 2.0 frsky version. @ipetepete wich is your transmitter?

ipetepete, 2021-12-13:
> No ping needed, D8 protocol doesnt work with ISRM tx or 2.0 frsky version. @ipetepete wich is your transmitter?

@Asizon I have both the RadioMaster Tx16s and a Jumper T-Lite. I found the issue with both TXs.

Gramps50, 2021-12-13:
I'm having the same issue using 2 different Matek411RX targets. In both cases D8 will no bind with my Jumper T12 Pro or my Jumper T16 both using the internal multiprotocol module. Using the nightly date 12/12/2021 of BF & the configuator. Tried frequency tuning which did not work. I can bind to D16 but get dropouts when less than 20' from the TX. Sometimes this will happen if I try BF 4.2.11 unless I go back to an earlier version 1st then come back to BF 4.2.11. Have never got a good bind in BF 4.3. I can furnish a dump or anything else that might help to resolve this issue.

krotow, 2021-12-13:
I can confirm that FrSky SPI receivers binding in Betaflight 4.3 is not possible in any mode. Two CrazyBee F4 Pro V2 FC's here (CRAZYBEEF4FR target). Does not bind neither in FrSky D / D8 nor FrSky X / D16 in any variation (16/8 channels, telemetry ON/OFF). Have original FrSky Taranis Q X7 ACCST, Jumper T-Lite JP4IN1 variety and TBS Tango 2 with multimodule radios. Both three does not bind when FC is flashed with any Betaflight 4.3 build. When FC is flashed with Betaflight 4.2 (tried 4.2.11), binding happen on first try with both 3 radios.

It is however possible to copy binding variables frsky_spi_tx_id, frsky_spi_offset and frsky_spi_bind_hop_data from Betaflight 4.2 with successful binding to Betaflight 4.3 to get bind with radio which is used during binding on Betaflight 4.2. Although that doesn't solve main problem.

SteveCEvans, 2021-12-13:
Only binding doesn’t work in D8. I believe you can bind in D16 mode and then switch to D8 if desired.

ipetepete, 2021-12-14:
> Only binding doesn’t work in D8. I believe you can bind in D16 mode and then switch to D8 if desired.

This was not my experience. I was able to bind in D16 but switching to D8 mode lost the binding, and would not bind.

Gramps50, 2021-12-14:
> > Only binding doesn’t work in D8. I believe you can bind in D16 mode and then switch to D8 if desired.
> 
> This was not my experience. I was able to bind in D16 but switching to D8 mode lost the binding, and would not bind.

I also found the same thing, binding D16 then going to D8 did not work in my case either

etracer65, 2021-12-14:
Broken by https://github.com/betaflight/betaflight/pull/9697

MJ666, 2021-12-15:
Fix is now available for testing.

krotow, 2021-12-15:
> Fix is now available for testing.

Thanks. Can't test them though - not in development downloads yet.

haslinghuis, 2021-12-15:
Yes you can:

https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=4930&view=artifacts&pathAsName=false&type=publishedArtifacts

Have to flash manual (first load your unified target config)

krotow, 2021-12-15:
> Yes you can:
..... 
> Have to flash manual (first load your unified target config)

I would, if particular ZIP archive would contain CRAZYBEEF4FR target. Can't find it here.

haslinghuis, 2021-12-15:
Look here: https://github.com/betaflight/unified-targets/tree/master/configs/default

Need to download the right target in RAW format.

krotow, 2021-12-16:
> Need to download the right target in RAW format. Select offline button in flashing tab, load this file. Then click again and load the STM32F411 file and click flash button

I see. Managed to flash it now. Indeed binding for CRAZYBEEF4FR configuration in FrSky D / D8 and FrSky X D16 modes now works fine. Although there is no indication in CLI, but only affirmative LED flash on FC. But that seems is SPI binding related thing. You can now release this fix.

MLutzAT, 2022-01-11:
Should be solved by now with #11269

closed 2022-01-13 by haslinghuis