betaflight/betaflight #11137: Dshot not working on latest master build on G4

DusKing1, 2021-12-20:
### Describe the bug
 Any Dshot settings (600/300, bidir\no-bidir) does not working on currently master build on G47x target.

### To Reproduce
@SteveCEvans grab that AirFleetV11 dev board, flash with latest master build, dshot just won't work.

### Expected behavior
Dshot300 can work both in bidir mode and no-bidir mode.

### Flight controller configuration
`NERC-HYBRIDG4.config`

DusKing1, 2021-12-20:
@github-actions fk off

DusKing1, 2021-12-21:
@SteveCEvans any spare time for this?

SteveCEvans, 2021-12-22:
Confirmed, connected to a [Tekko32 F3 Metal 4in1 ESC](http://www.holybro.com/manual/Holybro_Tekko32_F3_Metal_4in1_ESC(65A)_spec_v1.2.pdf). Also not getting any ESC sensor data. BLHeli configurator is able to communicate though.

SteveCEvans, 2021-12-22:
@DusKing1 Has this ever worked to your knowledge? If so please provide me with the hash of the build.

DusKing1, 2021-12-22:
@SteveCEvans Can't provide commit hash, ATM, but I remember build 2476 can work nicely.

SteveCEvans, 2021-12-22:
[`3d5a0e585cc6a3e9db50b3b1b50602e389511983`](https://ci.betaflight.tech/job/Betaflight/2476/)

Not working for me I'm afraid. @DusKing1 @Asizon please check.

```
# dshot_telemetry_info
Dshot reads: 318044
Dshot invalid pkts: 318044
Dshot directionChange cycles: 0, micros: 0

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0   NO DATA
    1         0        0       0   NO DATA
    2         0        0       0   NO DATA
    3         0        0       0   NO DATA

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

Asizon, 2021-12-22:
```
# dshot_telemetry_info
Dshot reads: 243967
Dshot invalid pkts: 243967
Dshot directionChange cycles: 0, micros: 0

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0   NO DATA
    1         0        0       0   NO DATA
    2         0        0       0   NO DATA
    3         0        0       0   NO DATA

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
```

Asizon, 2021-12-22:
Doesnt work for mine with this hash
```
# dshot_telemetry_info
Dshot reads: 0
Dshot invalid pkts: 0
Dshot directionChange cycles: 0, micros: 0

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0   NO DATA
    1         0        0       0   NO DATA
    2         0        0       0   NO DATA
    3         0        0       0   NO DATA

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

Asizon, 2021-12-22:
with last master i received dshot reads:
```
# dshot_telemetry_info
Dshot reads: 243967
Dshot invalid pkts: 243967
Dshot directionChange cycles: 0, micros: 0

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0   NO DATA
    1         0        0       0   NO DATA
    2         0        0       0   NO DATA
    3         0        0       0   NO DATA

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

Asizon, 2021-12-22:
Definitelly it works as expected with last master. Dshot bidir works!

[image]

Asizon, 2021-12-22:
Then @DusKing1 you should check this config DMA resources :)

DusKing1, 2021-12-23:
@Asizon no we are talking about a different target.

DusKing1, 2021-12-23:
@SteveCEvans I'm tracing back, will give you the hash later

Asizon, 2021-12-23:
@DusKing1 no hash meeded, master is working correctly, you should chech your target dma config or hardware for this board.

DusKing1, 2021-12-23:
@SteveCEvans how about this, the first version when G47x has the BMI270 driver code. ``148f9e9d90c69acf850eadc330e77df9cdb4eb79``

SteveCEvans, 2021-12-23:
@Dusking I've tested with both current master and the build you reference above, and neither work with bit banged dshot.

With `set dshot_bitbang = OFF` the motor RPM is reported as expected. With `set dshot_bitbang = ON` there is no motor output whatsoever (as seen on a logic analyzer connected to the MOTOR 1 pin). Investigating.

SteveCEvans, 2021-12-23:
Bitbanged DSHOT doesn't work irrespective of bidirectional being used. The only difference is that the line sits low with is disabled, high with it enabled.

SteveCEvans, 2021-12-24:
@haslinghuis I'm reopening this as this is broken functionality and the unified target change is no more than a work around.

There's some clearly bogus code [HERE](https://github.com/betaflight/betaflight/blob/5933d96bc3e0718cc4fcd54f23436241d6e1c3c3/src/main/drivers/dshot_bitbang_ll.c#L159) which I'm investigating.

daleckystepan, 2022-02-07:
Should be fixed. Can you retest?

DusKing1, 2022-02-10:
@daleckystepan I'm a bit outdated here and what exact test I should be do?

haslinghuis, 2022-04-14:
Please check with latest firmware / configurator

DusKing1, 2022-04-15:
ok I will try to arrange some tests for this

DusKing1, 2022-04-19:
Dshot works normally now.

But I found that OSD still have some problem.

closed 2022-04-19 by DusKing1