betaflight/betaflight #11144: F722 refuses to arm when powered by 4S battery but works fine on 3S

wernerterreb, 2021-12-21:
### Describe the bug
My quad won't arm when I use a 4S battery.   The  arming prevention code that gets displayed is "CALIB" (What for sensor calibration to complete.  The strange thing that I discovered (just by fluke) is that when I plug in a 3S battery that is at storage voltage, then the quad arms and flies without a problem.  However this is supposed to be a 4S quad and it does not have much power if I fly with a 3S battery

I have seen other cases where increasing the gyro_calib_noise_limit to something above 120 can resolve the problem.   However that did not work in my case even when I made it the maximum 200.

I also tried loading the latest nightly build of Betaflight 4.3 but that did not fix the issue either. So I reverted back to 4.2.11.

### To Reproduce
STM32F722 board with Betaflight 4.2.11  (Exact FC is Supra F722 Bluetooth Flight Control board)
Plug in a 4S battery and try to arm.   I beeps two long and three short beeps and displays the "CALIB" error flag in the OSD.
Swop out the battery for a 3S battery and try again.  The quad arms and flies perfectly, except that it does not have enough power.

### Expected behavior
When I arm the quad with a 3S or a  4S battery it should arm.  At the moment it only arms if I use a 3S battery.

### Flight controller configuration
```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.11 Nov  9 2021 / 20:29:32 (948ba6339) MSP API: 1.43
# config: manufacturer_id: JHEF, board_name: JHEF7DUAL, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

board_name JHEF7DUAL
manufacturer_id JHEF

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 5 2 115200 9600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 900 1125 0 0
aux 2 2 1 1375 1650 0 0
aux 3 13 2 1900 2075 0 0
aux 4 13 3 1875 2100 0 0
aux 5 28 1 1350 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 200 400 800 999

# master
set gyro_to_use = SECOND
set acc_calibration = -102,237,-169,1
set mag_hardware = NONE
set baro_hardware = NONE
[... 307 lines cut ...]
DMA2 Stream 1: ADC
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions

 - Flight controller: 
 Supra F722 Bluetooth Flight Control
 
 - Other components:
 ExpressLRS receiver
 Mamba VTX
  BN220 GPS
  PDB with 5V supply
 
 - How are the different components wired up:
 UART1 - VTX IRC Tramp
 UART2 - ExpressLRS  Rx
 UART6 - GPS
 5V supply for all pheripherals comes from PDB and not the FC

etracer65, 2021-12-21:
You have a hardware problem with your flight controller - specifically with the onboard voltage regulators. Voltage regulators have to work harder with a higher input voltage. It's likely that it's starting to fail when under more stress resulting in unstable and noisy voltage output. The gyro is highly sensitive to needing a very stable and clean voltage supply. So this instability in the 3.3v voltage supply is making the gyro output lots of additional noise resulting in the calibration failure. The problem is likely with the 5v regulator as it sees the battery voltage. The 3.3v regulator is run from the 5v supply so a noisy 5v would cascade into a problem in the 3.3v rail.

It's possible you've knocked some components off the board (like capacitors) and it could be fixed. But likely you'll have to replace the flight controller unless you have the tools to diagnose and replace the failing components.

Closing as this is not a Betaflight issue.

closed 2021-12-21 by etracer65

wernerterreb, 2021-12-24:
Hey @etracer65.   Just a quick thank you.   Your comment was my Christmas gift on the 24th of December.   After reading your comment I decided to try a put a 100uF capacitor on the 5V output of the voltage regulator.  And what do you know!   Problem gone!  :)     Save me some costs of a new flight controller and also a considerable amount of work to replace the flight control board.   

I hope my  comment will also help someone else in the future that might have experienced the same problem,